zsmartsystems/com.zsmartsystems.zigbee #583: Network re-initialization still sets the network state to OFFLINE

TomTravaglino, 2019-03-29:
This issue is related to #548 #554 #557 and to [this one in the binding](https://github.com/openhab/org.openhab.binding.zigbee/issues/398)

In the issues mentioned above we discussed the problem that the reconnect job in the binding becomes active when the initialization is in progress. A solution was provided in #554 which prevents the ZigBeeNetworkState from going OFFLINE during initialization.
Despite this solution I'm still experiencing the network state going OFFLINE during initialization. I investigated the issue and could reproduce how it happens.

- My app calls `ZigBeeNetworkManager.startup(true)`.
- `ZigBeeNetworkManager.startup` calls `transport.startup(reinitialize)`. In my case `transport` is an instance of `ZigBeeDongleEzsp`.
- `ZigBeeDongleEzsp.startup` calls `EmberNetworkInitialisation.formNetwork(...)`.
- `EmberNetworkInitialisation.formNetwork` calls `EmberNcp.leaveNetwork()`.
- `EmberNcp.leaveNetwork` sends a `EzspLeaveNetworkRequest`.
- By sending `EzspLeaveNetworkRequest` a `EzspStackStatusHandler [status=EMBER_NETWORK_DOWN]` is generated.
- `ZigBeeDongleEzsp.handlePacket(EzspFrame)` receives `EzspStackStatusHandler` and calls `ZigBeeDongleEzsp.handleLinkStateChange(false)`.
- `ZigBeeDongleEzsp.handleLinkStateChange` calls `ZigBeeNetworkManager.setTransportState(ZigBeeTransportState)` where the network state changes from ONLINE to OFFLINE and this makes the reconnect job run.

Maybe the network state should be set to INITIALISING at an earlier point (before forming the network) so that the state change to OFFLINE can be filtered as implemented in fix #554.

cdjackson, 2019-03-29:
Please define what version of the libraries you are using?

TomTravaglino, 2019-03-29:
We're testing zss 1.2.0 which contains the fix #554.

Regarding my statement
>Maybe the network state should be set to INITIALISING at an earlier point ...

Where do you think is the right place to set the network state to INITIALISING?

cdjackson, 2019-03-30:
> Where do you think is the right place to set the network state to INITIALISING?

When the network manager starts to be initialised - so when the ```ZigBeeNetworkManager.initialise()``` method is called.  This is pretty early in the initialisation - before this it is set to ```UNINITIALISED``` and the initialisation should not be able to start until the ```initialise``` method is called. This is set before the ```startup``` method is called - I don't think it can be set much earlier than this?

> Maybe the network state should be set to INITIALISING at an earlier point (before forming the network) so that the state change to OFFLINE can be filtered as implemented in fix #554.

It is - the state is set to ```INITIALISING``` during the ```initialize``` method which is before the ```startup``` method, and therefore before forming the network.

> ZigBeeDongleEzsp.handleLinkStateChange calls ZigBeeNetworkManager.setTransportState(ZigBeeTransportState) where the network state changes from ONLINE to OFFLINE and this makes the reconnect job run.

Where in your flow above was the network set to ONLINE?

TomTravaglino, 2019-04-02:
>Where in your flow above was the network set to ONLINE?

The network state was ONLINE from the start. I attached a log file which shows that the network state first changes to OFFLINE.

[logs-nw-restart-fails.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3032543/logs-nw-restart-fails.txt)

cdjackson, 2019-04-02:
Nearly the first line in the ```initialize``` method is to set the state to ```INITIALIZING``` so it should not start with ```ONLINE``` -:

```
    public ZigBeeStatus initialize() {
        synchronized (this) {
            if (networkState != ZigBeeTransportState.UNINITIALISED) {
                return ZigBeeStatus.INVALID_STATE;
            }
            setNetworkState(ZigBeeTransportState.INITIALISING);

            if (networkStateSerializer != null) {
                networkStateSerializer.deserialize(this);
            }
        }
```

In your log is that I see the polling right from the beginning of the log, so this log does not contain the initialisation - ie the first line is -:

```
2019-03-27_15:34:02:043	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP: EzspNetworkStateRequest []
```

> The network state was ONLINE from the start. I attached a log file which shows that the network state first changes to OFFLINE.

Your issue is that the state is ONLINE from the beginning, so please either try to debug this, or provide a log that shows the full initialisation. This log starts well after the initialisation starts, and I cannot confirm what happens before this so it doesn't show the issue you report.

triller-telekom, 2019-04-03:
I think from the log posted in https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/583#issuecomment-478860962 the issue is not clear without a little explanation:

I have abbreviated the log a bit so it now starts at the line where we call `ZigbeeNetworkManager.startup(true)`

[logs-nw-restart-fails-cut.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3038271/logs-nw-restart-fails-cut.txt)

Please not the line

```
2019-03-27_15:34:10:794	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Network state is updated to OFFLINE
```

which occurs before

```
2019-03-27_15:34:11:962	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Network state is updated to INITIALISING
```

And this is probably the issue.

cdjackson, 2019-04-03:
> I have abbreviated the log a bit so it now starts at the line where we call ZigbeeNetworkManager.startup(true)

Please can you provide the full log of the startup rather than starting after the state is set to ```INITIALISING```.

>And this is probably the issue.

At first glance, I don't see why. The call to set the state to initialising is done in the ```startup``` method, which is called BEFORE the ```initialize``` method. As this log starts at the ```startup``` then it should have already been set - shouldn't it? Clearly there is something not right as the state seems to be set to INITIALISING later, but that's not how it looks in the code, and as your log is not complete I'm hesitant to spend time on this without the full story.

Please can you provide the full log of the initialisation - or alternatively take a look through the code to see why the initial setting of the initialisation state does not seem to be working?

I'm happy to take a better look at this, but I'd like to have the full information if it's possible?

cdjackson, 2019-04-03:
From a quick look at the code, the only place that the state is set to ```INITIALISING``` is in the ```initialise``` method. I've not looked to see if these can get out of order through the notification service in the ```setNetworkState``` method - that might explain this if that is happening. Maybe you could take a look at that ?

triller-telekom, 2019-04-03:
I guess there is a misunderstanding here, because this IS the full story...

As @TomTravaglino says:

> My app calls ZigBeeNetworkManager.startup(true).

This **IS** the very first call we do. Question is: Are we supposed to call this directly, instead of `ZigBeeNetworkManager.initialise()`?

Background of this is: The zigbee network stack is currently running, i.e. it is in state ONLINE. We have a backupprovider which is trying to do the following code in this state:

```
networkManager.setZigBeePanId(Integer.parseInt(panID));
            networkManager.setZigBeeExtendedPanId(new ExtendedPanId(extendedPanID));
            networkManager.setZigBeeChannel(ZigBeeChannel.create(Integer.parseInt(channel)));
            networkManager.setZigBeeLinkKey(new ZigBeeKey(linkKey));

            ZigBeeKey key = new ZigBeeKey(networkKey);
            key.setOutgoingFrameCounter(frameCounter);
            networkManager.setZigBeeNetworkKey(key);

            networkManager.startup(restart);
```
If I wanted to call `ZigBeeNetworkManager.initialise()` instead of `startup(true)` there is a condition check which prevents us from doing so:

```
public ZigBeeStatus initialize() {
        synchronized (this) {
            if (networkState != ZigBeeNetworkState.UNINITIALISED) {
                return ZigBeeStatus.INVALID_STATE;
            }
...
```

So what we are trying to achieve is a restart of the zigbee stack with the new parameters which we got from the backup.

This worked fine before I implemented the reconnectJob in the Coordinatorhandler of OH, so the code was working. But now this job kicks in if the zigbee network state becomes OFFLINE and interferes with the startup call from the backupprovider.

Does this give you a better picture of what we are doing?

cdjackson, 2019-04-03:
> This IS the very first call we do. Question is: Are we supposed to call this directly, instead of ZigBeeNetworkManager.initialise()?

Ok, clearly I don't know what your appication is doing. It should call ZigBeeNEtworkManager.initialise as per the documentation in the top of ZigBeeNetworkManager -:

```
 * The ZigBeeNetworkManager lifecycle is as follows -:
 * <ul>
 * <li>Instantiate a {@link ZigBeeTransportTransmit} class
 * <li>Instantiate a {@link ZigBeeNetworkManager} class passing the previously created {@link ZigBeeTransportTransmit}
 * class
 * <li>Optionally set the {@link ZigBeeSerializer} and {@link ZigBeeDeserializer} using the {@link #setSerializer}
 * method
 * <li>Call the {@link #initialize} method to perform the initial initialization of the ZigBee network
 * <li>Set the network configuration (see below).
 * <li>Call the {@link #startup} method to start using the configured ZigBee network. Configuration methods may not be
 * used.
 * <li>Call the {@link shutdown} method to close the network
```

please follow this - that said, I'm pretty surprised that you can call startup without calling initialise as the first check in initialise checks the state -:

```
        if (networkState == ZigBeeNetworkState.UNINITIALISED) {
            return ZigBeeStatus.INVALID_STATE;
        }
```

> If I wanted to call ZigBeeNetworkManager.initialise() instead of startup(true) there is a condition check which prevents us from doing so:

No - this ensures that the state is UNINITIALISED when calling initialise - it then sets the state to INITIALISING, and then in the startup method it checks that the state is NOT UNINITIALISED, thus 
preventing you from calling startup before initialise.

All in all, I'm confused - sorry.

triller-telekom, 2019-04-03:
Let me rephrase what we are doing in other words:

Our BackupProvider has some network keys, PanID etc, inside a backup and wants to write these onto the hardware, in our case an Ember chip.

If I would call `ZigbeeNetworkManager.shutdown()` and then `ZigbeeNetworkManager.initialise()` the zigbee network state would become OFFLINE after the shutdown. This would trigger the reconnect job again, which I would like to avoid.

So what is the correct way if **restarting** the `ZigbeeNetworkManger` whereas I expect the network state as follows: ONLINE (as it is running) -> INITIALIZING -> ONFLINE/OFFLINE (depending on success or failure of init)?

We thought calling `ZigbeeNetworkManager.startup(true)`, but as you said this is probably wrong. So what is the right way of doing it instead?

cdjackson, 2019-04-03:
Ok, I didn’t know that this is what you were doing - it’s useful to understand this and I will look at this further in due course.

triller-telekom, 2019-04-04:
Thank you for looking into it. And next time we try to give you a better picture right from the start to avoid further confusions and misunderstandings.

triller-telekom, 2019-04-11:
@cdjackson may I get back to you on this issue?

Have you had a chance to look at a possibility to reinitialize the framework?

cdjackson, 2019-04-12:
@triller-telekom firstly, it is not possible to call ```shutdown``` other than to free resources - this method can only be called when the ```ZigBeeNetworkManager``` is finally closed to free resources etc. It can't be reinitialised after this.

What I think is needed is a new method in ```ZigBeeNetworkManager``` to take the system down temporarily.

[image]

I will take a look at this - hopefully it will solve your issue.

cdjackson, 2019-04-13:
I propose the following transitions. I think this should broadly achieve what you want.

[image]

This adds a new ```reinitialise``` method to the network manager to reinitialise and avoids the ```OFFLINE``` state. The only way to transition to ```OFFLINE``` is a failed ```startup``` or a notification from the transport layer.

The ```reinitialise``` method will take the transport layer offline, and effectively put the system back into the state it was in during the first initialisation.

Let me know what you think before I implement this.

triller-telekom, 2019-04-15:
> This adds a new reinitialise method to the network manager to reinitialise and avoids the OFFLINE state. The only way to transition to OFFLINE is a failed startup or a notification from the transport layer.

Sound good to be, but I have one remark regarding:

> The reinitialise method will take the transport layer offline, and effectively put the system back into the state it was in during the first initialisation.

Our client application the `ZigbeeCoordinatorHandler` from the openHAB binding is registered as a `ZigBeeNetworkStateListener`. So will these `listerners` be informed about the OFFLINE state of the transport layer? If so, then the described concept won't help unfortunately.

triller-telekom, 2019-04-15:
I just read your comment in the WIP PR:

> Note that while this method will take the {@link ZigBeeTransportState} to {@link ZigBeeTransportState#OFFLINE}
     * the {@link ZigBeeNetworkState} will be {@link ZigBeeNetworkState#INITIALISING} - the
     * {@link ZigBeeTransportState#OFFLINE} will not be reflected through the the application.

So yes, then it should work as you described it.

cdjackson, 2019-04-15:
> So will these listerners be informed about the OFFLINE state of the transport layer?

No - once in INITIALISATION state, the OFFLINE from the transport is not passed through to the application.

cdjackson, 2019-04-18:
Closed by #590

closed 2019-04-18 by cdjackson