zsmartsystems/com.zsmartsystems.zigbee #584: Timeout while reading from Xiaomi Aqara Temperature

DreyerAc, 2019-04-04:
While reading the temperature from an Xiaomi Aqara Temperature Sensor with the console application I receive a Timeout. 

The device join works without any problems. Also the output fro the nodes command seems to be fine:
> nodes
Network  Addr  IEEE Address       Logical Type   EP   Profile                    Device Type      Manufacturer     Model          
      0  0000  00124B001936758B   COORDINATOR      1  ZIGBEE_HOME_AUTOMATION     ON_OFF_SWITCH                                    
  56047  DAEF  00158D0002F81E0E   END_DEVICE       1  ZIGBEE_HOME_AUTOMATION     ON_OFF_SWITCH             

Also the output from the node and the endpoint command seems fine:
> node 56047
IEEE Address     : 00158D0002F81E0E
Network Address  : 56047
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=127, complexDescriptorAvailable=false, manufacturerCode=4151, logicalType=END_DEVICE, serverCapabilities=[], incomingTransferSize=100, outgoingTransferSize=100, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[REDUCED_FUNCTION_DEVICE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=0]
Power Descriptor : RECEIVER_ON_IDLE, [DISPOSABLE_BATTERY], DISPOSABLE_BATTERY, FULL
Associations     : []
Endpoints        : 
            1    : Profile     0104 ZIGBEE_HOME_AUTOMATION
                 : Device Type 0000 ON_OFF_SWITCH
                   -> BASIC
                   -> TEMPERATURE_MEASUREMENT
                   -> IDENTIFY
                   -> PRESSURE_MEASUREMENT
                   -> RELATIVE_HUMIDITY_MEASUREMENT
                   -> GENERAL
                   <- BASIC
                   <- GROUPS
                   <- GENERAL
Neighbors:
Routes:

> endpoint 56047/1
IEEE Address     : 00158D0002F81E0E
Network Address  : 56047
Endpoint         : 1
Device Profile   : 0x0104, ZIGBEE_HOME_AUTOMATION
Device Type      : 0x0000, ON_OFF_SWITCH
Device Version   : 0
Input Clusters   : (Server)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCLVersion                                
        S       1 r-- UNSIGNED_8_BIT_INTEGER    ApplicationVersion                        
        S       2 r-- UNSIGNED_8_BIT_INTEGER    StackVersion                              
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HWVersion                                 
        S       4 r-- CHARACTER_STRING          ManufacturerName                          
        S       5 r-- CHARACTER_STRING          ModelIdentifier                           
        S       6 r-- CHARACTER_STRING          DateCode                                  
        S       7 r-- ENUMERATION_8_BIT         PowerSource                               
        S      16 rw- CHARACTER_STRING          LocationDescription                       
        S      17 rw- ENUMERATION_8_BIT         PhysicalEnvironment                       
        S      18 rw- BOOLEAN                   DeviceEnabled                             
        S      19 rw- BITMAP_8_BIT              AlarmMask                                 
        S      20 rw- BITMAP_8_BIT              DisableLocalConfig                        
        S   16384 r-- CHARACTER_STRING          SWBuildID                                 
   0003 Identify
     - APS Security disabled
        S       0 rw- UNSIGNED_16_BIT_INTEGER   IdentifyTime                              
   0402 Temperature measurement
     - APS Security disabled
        S       0 r-s SIGNED_16_BIT_INTEGER     MeasuredValue                            Wed Apr 03 08:35:30 CEST 2019 1753
        S       1 r-- SIGNED_16_BIT_INTEGER     MinMeasuredValue                          
        S       2 r-- SIGNED_16_BIT_INTEGER     MaxMeasuredValue                          
        S       3 r-s UNSIGNED_16_BIT_INTEGER   Tolerance                                 
   0403 Pressure measurement
     - APS Security disabled
        S       0 r-s SIGNED_16_BIT_INTEGER     MeasuredValue                            Wed Apr 03 08:35:30 CEST 2019 939
        S       1 r-- SIGNED_16_BIT_INTEGER     MinMeasuredValue                          
        S       2 r-s SIGNED_16_BIT_INTEGER     MaxMeasuredValue                          
        S       3 r-- UNSIGNED_16_BIT_INTEGER   Tolerance                                 
        S      16 r-s SIGNED_16_BIT_INTEGER     ScaledValue                              Wed Apr 03 08:35:30 CEST 2019 9392
        S      17 r-- SIGNED_16_BIT_INTEGER     MinScaledValue                            
        S      18 r-- SIGNED_16_BIT_INTEGER     MaxScaledValue                            
        S      19 r-s UNSIGNED_16_BIT_INTEGER   ScaledTolerance                           
        S      20 r-- UNSIGNED_8_BIT_INTEGER    Scale                                    Wed Apr 03 08:35:30 CEST 2019 -1
   0405 Relative humidity measurement
     - APS Security disabled
        S       0 r-s UNSIGNED_16_BIT_INTEGER   MeasuredValue                            Wed Apr 03 08:35:30 CEST 2019 4663
        S       1 r-- UNSIGNED_16_BIT_INTEGER   MinMeasuredValue                          
        S       2 r-- UNSIGNED_16_BIT_INTEGER   MaxMeasuredValue                          
        S       3 r-s UNSIGNED_16_BIT_INTEGER   Tolerance                                 
   FFFF General
     - APS Security disabled
Output Clusters  : (Client)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCLVersion                                
        S       1 r-- UNSIGNED_8_BIT_INTEGER    ApplicationVersion                        
        S       2 r-- UNSIGNED_8_BIT_INTEGER    StackVersion                              
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HWVersion                                 
        S       4 r-- CHARACTER_STRING          ManufacturerName                          
        S       5 r-- CHARACTER_STRING          ModelIdentifier                           
        S       6 r-- CHARACTER_STRING          DateCode                                  
        S       7 r-- ENUMERATION_8_BIT         PowerSource                               
        S      16 rw- CHARACTER_STRING          LocationDescription                       
        S      17 rw- ENUMERATION_8_BIT         PhysicalEnvironment                       
        S      18 rw- BOOLEAN                   DeviceEnabled                             
        S      19 rw- BITMAP_8_BIT              AlarmMask                                 
        S      20 rw- BITMAP_8_BIT              DisableLocalConfig                        
        S   16384 r-- CHARACTER_STRING          SWBuildID                                 
   0004 Groups
     - APS Security disabled
   FFFF General
     - APS Security disabled

But if I try to read an attribute there is a timeout:
> read 56047/1 1026 0
Reading server cluster Temperature measurement (0402), attributes MeasuredValue
Error executing command: CommandResult [TIMEOUT]

Is the Xiaomi Sensor supported from this library or is this a problem with the console application?

cdjackson, 2019-04-04:
This is normal when devices don't respect the 7.6 second polling requirement. Clearly the framework cannot poll a battery device that is sleeping and doesn't wake up - hence the timeout.

DreyerAc, 2019-04-09:
Ok. Thanks. Is there any possibility to wake up such a device?

cdjackson, 2019-04-10:
You would need to read the manual. Often there is a button that you can press a number of times, or hold for a certain time, or something like this. 

It's not possible to wake a device over the air (since it is sleeping, and this means the receiver is powered off).

cdjackson, 2019-04-12:
Closing as above.

closed 2019-04-12 by cdjackson