zsmartsystems/com.zsmartsystems.zigbee #586: java.lang.ArrayIndexOutOfBoundsException makes SpiFrameHandler stop working

TomTravaglino, 2019-04-12:
I observed that sometimes when I restart our runtime the SpiFrameHandler stops working after a short time. The reason is that the handler receives this frame `FE 04 02 80 FF 00` and then dies by this Exception
```
2019-04-11_13:51:22:192	NA	ERROR:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - SpiFrameHandler Exception: 
java.lang.ArrayIndexOutOfBoundsException: 4
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.EzspFrame.createHandler(EzspFrame.java:398)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.processSpiEzsp(SpiFrameHandler.java:393)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.processSpiCommand(SpiFrameHandler.java:328)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler.access$300(SpiFrameHandler.java:57)
	at com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler$1.run(SpiFrameHandler.java:193)
```

What's wrong with the frame? Does it need to be handled? I attached the logs with the Exception at the bottom.
[logs-exception-in-spi-framehandler.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3072088/logs-exception-in-spi-framehandler.txt)

cdjackson, 2019-04-12:
It looks like the NCP is sending invalid data - we can catch this and ignore the frame. You can see below that the special value of 0xFF means that we have the extended frame - this is used to detect the old frame format, and the new frame format in the binding. With the new frame, we check for the frame type in byte 4, which is not in this frame - hence the exception.

```
RX SPI frame: FE 04 02 80 FF 00 
```

The EZSP frame starts with the ```02``` as per the packet definitions below.

[image]

[image]

This format changed a long time ago (EZSP 5 if I remember correctly, which was in firmware version around 5.10 I think), so it's nothing new. However it looks like you have updated your firmware to 6.5 - did this just start happening after the update? I wonder if there is an error, or change at least in the latest firmware (although there is no change in the doc).

I also note there is another exception in this log - you didn't comment on it so I guess you are happy it's not causing any issue (it's in your SPI reader code, so not part of the binding or the framework).

Do you want to trap this, or add length checks? I'll leave it up to you to provide a PR?

TomTravaglino, 2019-04-15:
>However it looks like you have updated your firmware to 6.5 - did this just start happening after the update?

Not sure if this issue came up after the version change. However, we need to handle it.

>I also note there is another exception in this log - you didn't comment on it so I guess you are happy it's not causing any issue

I see this Exception from time to time but I couldn't figure out why it's thrown. Regarding the impact, it seems that the `RetryTimer` in the `SpiFrameHandler` does his job so that the Exception is not causing any issues.

>Do you want to trap this, or add length checks? I'll leave it up to you to provide a PR?

I can surround the code in `EzspFrame` with a try-catch and handle the `ArrayIndexOutOfBoundsException`.

PR #594 (2019-04-15): Catch invalid Ezsp frame when detecting the frame type
Fixes #586 

In this PR `EzspFrame.createHandler(int[] data)` catches the `ArrayIndexOutOfBoundsException` thrown when an invalid frame is received . Now when the Exception occurs a null object is returned.
closed 2019-04-15 by cdjackson