zsmartsystems/com.zsmartsystems.zigbee #587: Android Sample

nanikjava, 2019-04-12:
Hi,

Trying to use zigbee inside Android application using XStick. Anybody has tried it before using it ?

Cheers

cdjackson, 2019-04-12:
Yes, this is used by a number of companies with Android.

nanikjava, 2019-04-13:
Thanks for the reply.  Are there any sample Android app available as looking at the source code there are none ?

cdjackson, 2019-04-13:
No - there’s currently no Android example publicly available. When I get some time I will publish one.

nanikjava, 2019-04-14:
Thank you. Appreciate the help.

Meantime while waiting for the sample code tried to use it inside Android. I managed to add jscc source code (NDK) and add the zsmartsystem source code into Android Studio and able to compile and run the application. However, facing a problem in code when it is trying to open the port inside the Java_jssc_SerialNativeInterface_openPort function call but managed to overcome it as it was permission issue, but after that it fails in the line 

if(tcgetattr(hComm, settings) == 0)

inside the same function. Any clue ?

[image]

cdjackson, 2019-04-14:
Sorry - I’m not familiar with this. I wrote Android USB drivers for Ember and TI chips rather than go down this route.

nanikjava, 2019-04-14:
The source code ZigBeeSerialPort.java is using the jssc project which is working in OpenHab (have tested this). Does this mean that it the same jssc will not work in Android ?

Sorry what you mean by Android USB driver ? did you write something to replace the jssc portion of the code ?

Thanks for your help.

cdjackson, 2019-04-14:
Sorry - I don’t know if jssc will work with Android. For Android, I wrote a different driver to work with the Android USB system for the Ember and TI chipsets only.

nanikjava, 2019-04-14:
Sorry for asking more questions but trying to get my head around it.

When you say "Android USB system" do you mean that you have rewritten the USB Manager service inside Android OR the driver inside the Linux Kernel ?

cdjackson, 2019-04-14:
I wrote it as an application side “driver" using the UsbManager and UsbInterface/UsbDevice etc classes thus avoiding any serial port driver (or other) dependencies.

nanikjava, 2019-04-14:
I see. So correct me if I'm wrong that means for your particular implementation you have complete replaced the UsbManager that is available inside the Android OS itself with your own implementation ? if that's the case that means you have does this in your own custom ROM ?

cdjackson, 2019-04-14:
No - I’ve “simply” written an application side driver that uses these interfaces.

nanikjava, 2019-04-14:
Ok cool. Will this code be made available in github ?

cdjackson, 2019-04-14:
Not at the moment as I’ve not had time to make this available publicly so I’ve currently only made it available to supported users. At some point I will likely make it available on GH, but it won’t be for a little while I’m afraid.

tunjid, 2019-06-17:
Hello Chris, thanks for the library and thorough documentation.

I'm trying to integrate this with an Android Things project with the Raspberry Pi using the Texas Instruments CC2531. I've managed to get as far as using the Android UsbManager and UsbInterface/UsbDevice APIs, to create an implementation of ```ZigBeePort``` by mirroring the ```ZigBeeSerialPort``` class as the ```AndroidSerialPort``` class.

To check if my implementation works, I attempted to initialize the Ti dongle:

```ZigBeeDongleTiCc2531(AndroidSerialPort(driver, 115200)).initialize()```

The result of that is some serial read and writes appear successful, however the result of the initialize call is ```COMMUNICATION_ERROR```. Further digging lets me see that in ```ZigbeeNetworkManager```, both ```dongleReset``` and ```bootloaderGetOut``` fail

```
        if (!dongleReset()) {
            logger.warn("Dongle reset failed. Assuming bootloader is running and sending magic byte {}.",
                    String.format("0x%02x", BOOTLOADER_MAGIC_BYTE));
            if (!bootloaderGetOut(BOOTLOADER_MAGIC_BYTE)) {
                logger.warn("Attempt to get out from bootloader failed.");
                shutdown();
                return null;
            }
        }
```

Which makes me think I may have the wrong firmware flashed on the CC2531. Please may I know the version of the Z stack you were able to get the CC2531 working with? I'm using the CC2531ZNP-Pro-Secure_LinkKeyJoin.hex.

tunjid, 2019-06-17:
Sorry, nevermind this. My issues stemmed from a poor byte to int conversion.

tunjid, 2019-07-08:
@cdjackson thanks a lot for the help above by suggesting to use Android's USB serial to communicate with the dongle. I've been able to port the ZigBee Console app to Android running Android Things on a Raspberry Pi and currently use it for my home automation needs.

If anyone is interested in the implementation, it's available here: https://github.com/tunjid/Android-Home-Hub

closed 2019-07-21 by cdjackson