zsmartsystems/com.zsmartsystems.zigbee #588: Two ZigBeeNetworkDiscoverer tasks started for one node

triller-telekom, 2019-04-12:
We just discovered while paring a device that there are to `ZigBeeNetworkDiscoverer` tasks started at the same time.

I looks like this leads to a lot of failed/concurrent transactions as one can see in the logs.

A side effect of this is this is also this exception:

```
java.lang.NullPointerException
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.requestPowerDescriptor(ZigBeeNodeServiceDiscoverer.java:376)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.access$800(ZigBeeNodeServiceDiscoverer.java:75)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer$NodeServiceDiscoveryTask.run(ZigBeeNodeServiceDiscoverer.java:602)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```

Please see the attached log file for more information:
[tmp8.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3073668/tmp8.txt)

We tested this on a branch which is based on the ZSS-1.2.0 branch and which also includes the XML-autocoder branch on top of it.

cdjackson, 2019-04-12:
Are you able to look at why this is duplicated?

I’m not sure that the failed transactions are related - I think that this is caused by sending a command to a device that the system is not aware of (hence the error about the unknown device with no queue). I think that the exception is also related to this - if the manager doesn’t queue the frame, it returns null, and this gives the exception, so I think that’s well understood and I’ll take a look at that.

triller-telekom, 2019-04-15:
Unfortunately I couldn't see so far why it was duplicated :( It happened while pairing a Bitron Video 902010/23 remote control.

triller-telekom, 2019-04-25:
Today I looked at this issue again and created a fresh log of a pairing from a CentraLite 3200-de smart plug. Its IEEE address is `000D6F000BBD2B5B` and I have marked in the logs where I see that 2 `com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer` scans are started.

@cdjackson Hopefully you are able to see more here. Because I still think that having multiple discoveries running might lead to problems.

[logs_doubleDiscovery.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3117563/logs_doubleDiscovery.txt)

cdjackson, 2019-04-27:
I will take a look.

Can you tell me what device node 17468 is?

cdjackson, 2019-04-27:
While I can't find exactly what's happening, what I do see is ```ZigBeeNetworkDiscoverer.deviceStatusUpdate``` is being called twice. I can identify two possible reasons for this, but can't see a mechanism for either.

* The ```ZigBeeAnnounceListener``` is being registered twice. This shouldn't happen, and I don't think it's the cause - however - the listeners is a ```List``` so technically it's possible for it to be registered twice. I will change this to a ```Set``` so this possibility will be eliminated.
* The ```ZigBeeDiscoveryExtension``` is being started twice. I suspect this might be the cause, although again, I can't really see the mechanism, but this is possible. Extensions are started after the network is ```ONLINE```. The ```ZigBeeNetworkManager``` should only call this if there is a transition to ```ONLINE``` from another state, so, if you had an ```ONLINE``` - ```OFFLINE``` - ```ONLINE``` situation, then this could occur. Starting the extension twice would create two ```ZigBeeNetworkDiscoverer```s and they would each register a ```ZigBeeAnnounceListener``` so we would see the double logging that I mention earlier.

I will shortly create a PR to resolve these two issues. It will eliminate the possibility of a ```ZigBeeAnnounceListener``` being registered twice, and will also ensure that the ```ZigBeeDiscoveryExtension``` is only started once so there will be no duplicate ```ZigBeeNetworkDiscoverer``` created. Hopefully that will resolve this, but if not, at least there will be less possibilities and I will look again.

PR #608 (2019-04-27): Avoid duplicate ZigBeeAnnounceListener registration
This PR prevents the same ```ZigBeeAnnounceListener``` being registered more than once, and also ensures that the ```ZigBeeDiscoveryExtension``` cannot be started more than once to prevent multiple ```ZigBeeNetworkDiscoverer``` tasks being started.

FYI @triller-telekom

Closes #588
closed 2019-04-27 by cdjackson