zsmartsystems/com.zsmartsystems.zigbee #595: Ember usb dongle not initialising the ASH protocol

triller-telekom, 2019-04-15:
Recently I had trouble to make an ember dongle work in openHAB.

I am using the ZSS 1.2. branch together with the Autocoder commits from https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/527

I have added some more debug output and this is what I see:

```
2019-04-15_15:49:39:662	NA	DEBUG:	[symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Connecting to serial port [/dev/ttyUSB0] at 57600 baud, flow control FLOWCONTROL_OUT_XONOFF.
2019-04-15_15:49:39:682	NA	DEBUG:	[symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Serial port [/dev/ttyUSB0] is initialized.
2019-04-15_15:49:40:012	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Stefan: Before frameHandler.start(serialPort)
2019-04-15_15:49:40:042	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Stefan: After framehandler.start
2019-04-15_15:49:40:052	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - AshFrameHandler thread started
2019-04-15_15:49:40:082	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:40:122	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Stefan: After FrameHandler.connect()
2019-04-15_15:49:40:152	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Stefan: ncp=com.zsmartsystems.zigbee.dongle.ember.EmberNcp@109141f
2019-04-15_15:49:40:212	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer: [248, 56, 188]
2019-04-15_15:49:40:212	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[0] = 248
2019-04-15_15:49:40:212	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[1] = 56
2019-04-15_15:49:40:212	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[2] = 188
2019-04-15_15:49:40:242	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - <-- RX ASH error: BAD PACKET 
2019-04-15_15:49:40:852	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.EmberNcp] - Stefan: getVersion: built request with desiredVersion=4 and set setDesiredProtocolVersion to=4
2019-04-15_15:49:40:862	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - TX EZSP: EzspVersionRequest [desiredProtocolVersion=4]
2019-04-15_15:49:40:892	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
2019-04-15_15:49:41:722	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:43:332	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:44:942	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:46:552	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:48:162	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:49:772	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:51:382	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:52:992	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:54:602	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:56:212	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:57:822	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:49:59:432	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:50:01:042	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:50:02:652	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:50:04:262	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:50:05:872	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:50:05:982	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer: [248, 56, 191]
2019-04-15_15:50:05:982	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[0] = 248
2019-04-15_15:50:05:982	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[1] = 56
2019-04-15_15:50:05:982	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[2] = 191
2019-04-15_15:50:05:982	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - <-- RX ASH error: BAD PACKET 
2019-04-15_15:50:07:482	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-04-15_15:50:09:092	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
```

So when it tries to set the desired protocol version to 4, it just gets broken packets from the chip.

I have seen
```
2019-04-15_15:49:40:212	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer: [248, 56, 188]
[... cut ...]
2019-04-15_15:56:25:942	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - Stefan: buffer[1] = 191
2019-04-15_15:56:25:942	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - <-- RX ASH error: BAD PACKET 
```

A full log from the initialisation phase is attached here: 
[log-post.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3080573/log-post.txt)

So basically the framework is stuck in the INITIALISATION phase.

triller-telekom, 2019-04-16:
I have debugged this a little further and it turned out that in my case the usb stick was in the bootloader mode!

The only way to get it back out again was to use some ncp.py script and flash a new firmware onto the stick. Afterwards it worked again.

So I am wondering whether this still is a ZSS bug. Probably not. But the application using ZSS should get the chance to know that the stick is in another mode to be able to react on it.

cdjackson, 2019-04-16:
I probably wouldn’t class this as a bug - we could add an enhancement to automatically recover from the bootloader, but it would require some thought as to how to do this effectively.

I’d suggest to close this - if you think it’s worth adding a feature to exit from bootloader, then feel free to open an issue specifically for that, although it won’t be too high on my personal list of things to do...

triller-telekom, 2019-04-17:
I am fine with closing this. If we have the need to recover from the bootloader mode we can create a new issue.

closed 2019-04-17 by triller-telekom