zsmartsystems/com.zsmartsystems.zigbee #597: Inconsistent data type mapping for UNSIGNED_48_BIT_INTEGER in branch "XML-autocoder"

TomTravaglino, 2019-04-17:
In the branch `XML-autocoder` I saw that the data type `UNSIGNED_48_BIT_INTEGER` is mapped to two different Java types
- In the enum `ZclDataType` it is mapped to `Integer`. 
- `DefaultDeserializer` converts such values to `Long`.

I ran into a ClassCastException when I was calling the method `getCurrentSummationDelivered` of the class 'ZclMeteringCluster'. In this method the attribute value is expected to be an `Integer`. But it gets a `Long` as this is what the DefaultDeserializer returns. I took a look the class `ZclDataType` on `master` and there `UNSIGNED_48_BIT_INTEGER` is mapped to `Long` which I assume is the correct type.

cdjackson, 2019-04-18:
Thanks @TomTravaglino - I will take a look at this. In fact though, it doesn't matter since the type in the ```ZclDataType``` enum is only used for compound types that implement ```ZclListItemField```. I agree it's not nice though and will take a look at how to resolve this as there are some other incorrect definitions in this enum, and I think it is probably best to remove these class definitions.

I'm no sure why you got the class cast exception - can you share a lot please? Maybe it is not related to this inconsistency?

>  I took a look the class ZclDataType on master and there UNSIGNED_48_BIT_INTEGER is mapped to Long which I assume is the correct type.

In the master branch, it is not defined. The ```UNSIGNED_48_BIT_INTEGER``` is only added in the new branch.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/protocol/ZclDataType.java#L50

cdjackson, 2019-04-18:
Closing this as I've updated the definition in the autocoder branch. However, I don't think it will resolve your issue as described above - I suspect there is either an error with the ZCL definition somewhere, or an error with whatever device you are testing.

If so, please open a new issue and we can discuss, but I really prompt you to provide log files when you discover an issue - especially if there is an exception - so that we can understand the cause.

TomTravaglino, 2019-04-23:
>In the master branch, it is not defined. The UNSIGNED_48_BIT_INTEGER is only added in the new branch.

It's defined in line 88. See [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/protocol/ZclDataType.java#L88) ;-).

cdjackson, 2019-04-23:
Hmmm - not sure why I didn't see that (maybe I looked in the other file with the same name which is in the autocoder package).

However, my comment still stands - I don't think this will impact your problem as I don't think this is used. The only time when the classes listed in this file are used is if they implement "Field List"  (I forget the exact class name) - otherwise the deserialiser is used. In this case - it's not going to use the definition here.

As requested - please provide a log showing your problem.

cdjackson, 2019-04-27:
@TomTravaglino I'm just wondering what you want to do here? Are you planning to provide the log, or do we not worry about this now?

TomTravaglino, 2019-04-29:
I will need to re-test it with the most recent version of ZSS 1.2.0 which - AFAIK - contains the new autocoder implementation.

cdjackson, 2019-04-29:
Do you have the log from the original error? That would be very useful to understand your problem...

TomTravaglino, 2019-04-29:
I managed to reproduce the logs. The code base is from the day when I opened this issue (maybe a little bit older). Let me know if you need more info.

[logs-clean.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3126347/logs-clean.txt)

cdjackson, 2019-04-29:
Thanks - that's fine. The exception is actually in the openHAB binding, and not in the ZigBee libs. So this is basically caused by the expectation in the binding that the data from the library is a Long. The change that was made in the autocoder will not help this - the data type is still dictated by the deserialiser - unless you are somehow using ZclDataType to get the class?

Anyway, as the issue isn't in this repo, and we've now made the deserialiser consistent with the enum, I will close this.

Thanks.

closed 2019-04-29 by cdjackson