zsmartsystems/com.zsmartsystems.zigbee #604: TransactionManager does not match Request and Response correctly

triller-telekom, 2019-04-25:
During initialization of an Ember chip connected via SPI we have observed the following:

In `ZigBeeDongleEzsp.initialize()`, we see

```
2019-04-25T09:33:30.116+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Ember local IEEE Address is 000D6F000F24146C
```

But we do not reach the debug output ""EZSP dongle initialize done: Initialised {}" at the end of that method.

Digging deeper into the issue we saw that we send the `EzspGetCurrentSecurityStateRequest`:

```
2019-04-25T09:33:32.563+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP: EzspGetCurrentSecurityStateRequest []
2019-04-25T09:33:32.564+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - --> TX SPI frame: FE 03 4C 00 69 
2019-04-25T09:33:32.564+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspGetNetworkParametersResponse [status=EMBER_SUCCESS, nodeType=EMBER_COORDINATOR, parameters=EmberNetworkParameters [extendedPanId=752C921BEE6FA922, panId=28456, radioTxPower=0, radioChannel=11, joinMethod=EMBER_USE_MAC_ASSOCIATION, nwkManagerId=0, nwkUpdateId=0, channels=07FFF800]]
```

and shortly afterwards a `EzspGetNodeIdResponse` arrives:

```
2019-04-25T09:33:32.678+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame: FE 05 4B 80 27 00 00 
2019-04-25T09:33:32.679+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - RX EZSP: EzspGetNodeIdResponse [nodeId=0]
```

Somehow this triggers the Network state to become ONLINE:

```
2019-04-25T09:33:32.683+0200   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Network state is updated to ONLINE
```

just before the `EzspGetCurrentSecurityStateResponse`, for which we are waiting, arrives:

```
2019-04-25T09:33:32.683+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame when lastFrameSent is null: FE 0E 4C 80 69 00 74 00 6C 14 24 0F 00 6F 0D 00 
```

We assume that `ZigBeeDongleEzsp.initialize()` is stuck within `SpiFrameHandler.sendEzspTransaction()` at `futureResponse.get()` (i.e. it waits for the `EzspGetCurrentSecurityStateResponse`) which is called by `ncp.getCurrentSecurityState()`. And because of that the network is not useable.

A sideeffect of this is that the `executorService` within `ZigBeeDongleEzsp.initialize()` is not set (we do not reach the line `executorService = Executors.newScheduledThreadPool(1)`), so we see a NPE in the logs.

@cdjackson Do you have an idea what is going wrong here? Did we hit a bug in the transaction manager? Its pretty hard to reproduce, so I only have the logs I have attached...

[failed-transaction-mapping.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3116328/failed-transaction-mapping.txt)

cdjackson, 2019-04-26:
As discussed, this is not related to the transaction manager. The transaction manager deals at the application level - the transactions here are down in the SPI handler.

Anyway, I don't see anything wrong really. I agree with your diagnosis, but I don't see why it gets stuck. The response is received ok - although what is interesting is that it takes 2 seconds to arrive and there is a retry to resend this command to the NCP.

[image]

Ultimately this command completes successfully, but I think it's the core of the problem. There is a second response a bit later, so both requests do ultimately get responses - but there is a delay...

Then, when we get to ```EzspGetCurrentSecurityStateRequest``` there is also no response, but this time there is no retry. Due to the fact that there is no response, this transaction stalls as there is no transaction timeout.

I think the retry is getting things out of sync. There should only be a single transaction that can be sent at once, and then the response should be received before the next is sent.

Ok, I think I know what's up.

The double response to the ```EzspGetNetworkParametersRequest```, caused by the retry gets things out of sync. There is another error in the log ```RX SPI frame when lastFrameSent is null:``` which indicates this.  This is caused by the ```SPI_CMD_EZSP``` being received for the second ```EzspGetNetworkParametersResponse``` - this resets the ```lastFrameSent``` and screws up the sequencing.

Currently the system doesn't check that the frame is the expected one before resetting ```lastFrameSent``` - probably this needs to be done to solve this. I will create a PR shortly and I would appreciate it if you can test this @triller-telekom.

cdjackson, 2019-05-05:
@triller-telekom I've had another look at this, and have a new theory. I wonder if the polling is stopping - this would explain the issue quite well as there should be no reason for the 2 second delay we see above. The 2 second delay is the retryTimer kicking in, but I actually think the retry timer should not be required in SPI as it's synchronous, and when this kicks it causes other problems as we are then out of sync.

I note that there is another exception in your log just before this - I don't think it would kill polling, but I'm not 100% sure where it happens -:

```
2019-04-25T09:33:29.145+0200   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - --> TX SPI frame: FE 05 3F 00 55 04 40 
java.util.concurrent.TimeoutException
	at java.util.concurrent.FutureTask.get(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```

As requested previously in the other issue, can you provide another log with ```TRACE``` enabled so we can see the polling.

There are also some potential synchronisation issues - I'll raise them separately as I can't see a direct link to this issue (although I wouldn't rule it out).

triller-telekom, 2019-05-06:
As mentioned earlier, unfortunately this issue is REALLY hard to reproduce by intention :( And I cannot run the embedded development hardware on TRACE level all the time to wait until it occurs.

So for now I cannot provide any new logs. If we know a potential reason and how we could theoretically reproduce it, I am more than happy to give it a try with TRACE debugging enabled.

cdjackson, 2019-05-06:
I have no suggestion on how to reproduce this. Can't you just run trace on this class? Or, can you change the poll logs to TRACE?

Alternatively, shall we close this then?

triller-telekom, 2019-05-06:
I just tried #622, please see my comment there.

I think #622 might help with this issue, though I don't know if it really fixes it. But since its hard to reproduce, I am fine with closing this issue once #622 is merged. I can reopen it in case it occurs again and I have more detailed logs.

cdjackson, 2019-05-06:
Ok, thanks. I'll merge #622 then and if this reappears please re-open this (although I think the next step is to understand the polling, so we will need to see this added to the logs).

triller-telekom, 2019-05-17:
I would like to re-open this issue as discussed in our call this morning.

I was even able reproduce it with TRACE logging. However or logging posted "DEBUG" instead of "TRACE" in the level column. But I see the single bytes being printed which only show up on trace level, so I assume its alright.

Unfortunately I only have the log from a pretty late point in initialization because with TRACE level there were too much logging statements for the buffer.

@cdjackson Would be nice if you could have a look at this. And I think we would need the discussed solution where we do not retry sending a packet, but rather set the network to offline if an error occurs and restart from the beginning, just like we already do in ASH.

[logs_post.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3190738/logs_post.txt)

cdjackson, 2019-05-17:
Interesting...

[image]

1) The retry timer is set to 500ms, but in this case, the timer seems to trigger 1.7 seconds after the frame is sent. I don't understand this.
2) More interestingly in this case is that we do receive the response to this message - it comes in just as the retry timer triggers (the first few bytes are received before, but you see it in the log above a few milliseconds after the retry is logged). This request should never take 1.7 seconds to respond - it's just getting a policy, so it's hard to know why this is delayed so long.

In this case, waiting a few milliseconds more would have avoided the issue, but it doesn't answer the question as to why it is so delayed. It might be due to the low power hardware you're running on (maybe!) - eg if something else takes the processor for a period that delays this response.  It's possible that the response in the earlier log is also received late as we see the response to the retry almost immediately, and again, if we'd held off before sending the retry, maybe it might have recovered (maybe!). If the hardware is "hanging", that might also explain why the timer is triggering after 1.7 seconds rather than 500mS.

So, I suggest to increase the retry delay to a few seconds (maybe even 5 or so), and also to remove the retry and instead set the system OFFLINE (as we discussed on the phone). That would give the system the best chance of recovering by waiting longer for the retry, and then aborting. As discussed, retries don't work with SPI, so we might as well wait a little longer, and if it fails, then restart.

What do you think?

triller-telekom, 2019-05-20:
The general idea looks good to me. Its plausible that we should not retry and rather restart from the beginning, since SPI is a synchronous protocol. Also giving it more time until we retry sounds like a good idea.

However, in the implementation, I found a bug and left a comment. Please have a look: https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/635/files#r285492070

cdjackson, 2019-05-20:
I don’t think the comment you made is really a bug. I agree in the case you have seen, it would not help, but I think that’s another issue.

Firstly, with this change, hopefully this issue will not occur. However, if it does, it will set the system to offline and allow the higher layers to account for this.

The point you make is a good one, but I think the failure to bring the system online during initialisation should be handled separately. The initialise and startup methods should always return an error if they don’t complete successfully (does this happen in this case?) and then the higher layer should manage that failure.

cdjackson, 2019-05-23:
Closed by #635 (hopefully :) ).

closed 2019-05-23 by cdjackson