zsmartsystems/com.zsmartsystems.zigbee #625: v 1.2.0 - Question regarding return types of ZclAttribute.readValue(long)

TomTravaglino, 2019-05-08:
We observed that the method returns completely different data types depending on whether it returns the cached value or it reads the attribute from the cluster.

Example: This is a code snippet used to call the method `ZclAttribute.readValue(long)`
```
            ZclAttribute attribute = otaCluster.getAttribute(ZclOtaUpgradeCluster.ATTR_CURRENTFILEVERSION);
            Object readValue = attribute.readValue(Long.MAX_VALUE);
```
We were expecting a result of type `Integer` here. 

This is `ZclAttribute.readValue(long)`
```
    public Object readValue(long refreshPeriod) {
        if (isLastValueCurrent(refreshPeriod)) {
            return getLastValue();
        }

        try {
            return cluster.readAttribute(id).get();
        } catch (InterruptedException | ExecutionException e) {
            // Eat me!
        }
        return null;
    }
```
If `getLastValue()` is called then the method returns an `Integer`. If `cluster.readAttribute(id).get()` is called then the method returns a `CommandResult`. This makes things more complicated as you always need a type check and extract the attribute value whereever you read the attribute. Was this intented or was `ZclAttribute.readValue(long)` meant to simply return the attribute value?

cdjackson, 2019-05-08:
I will change this. I copied this over from the cluster class, so didn't check it too closely - maybe I didn't copy it properly as I don't think this existed previously.

Thanks for spotting.

cdjackson, 2019-05-09:
@TomTravaglino this should be resolved now - let me know if there's any issues.

closed 2019-05-09 by cdjackson

TomTravaglino, 2019-05-10:
@cdjackson I looked at the PR and now `ZclAttribute.readValue()` always returns the attribute value instead of the `CommandResult`. Didn't test it but it looks good to me.