zsmartsystems/com.zsmartsystems.zigbee #630: Remove deprecated cluster attribute read commands

cdjackson, 2019-05-14:
As per #591, in ```1.3.0``` we should remove the deprecated methods that were deprecated in ```1.2.0```.

tejas-n, 2020-03-23:
Can I pick this one up or is it planned to be done later? 😄
I'm assuming this has two parts?
1. Replacing the internal refrences to the deprecated methods and
2.  Modifying the auto encoder to not generate those methods and generate the new cluster classes?

cdjackson, 2020-03-23:
Hi Tejas,
I didn't want to do this too soon as I want to avoid breaking users systems :)

That said, if you wanted to pick up point 1 (replace internal references) then that would be great. I did do some, but I don't recall if I systematically went through and removed all internal uses or not. This could certainly be done now if you wanted, and would be appreciated.

The autocoder part is pretty simple since all methods are already marked deprecated, and it's just a case of removing them. I don't really want to do that for a while though as it will break systems that are still using these methods - my plan was to do this shortly before the update from V1.3.x to 1.4.0 which is probably a little way off at the moment.

Is there a reason you wanted to remove them? eg is it causing resource issues in your system?

tejas-n, 2020-03-23:
Thanks for the reply, Chris.
There's no particular reason I wanted to do it. It's just that I had a bit of free time on the weekend and I saw this issue which I assumed would be easy to do :)
I can certainly pick up point 1, I think I saw a few internal usages.

cdjackson, 2020-03-23:
Thanks Tejas. Picking up the first point is most of the work really and would be much appreciated. 

I guess your IDE will provide a list of all deprecated warnings.

Thanks.