zsmartsystems/com.zsmartsystems.zigbee #640: Telegesis stick not working: Error setting Telegesis Prompt 1 register

triller-telekom, 2019-05-22:
By using the newest ZSS from the master branch (i.e. the merged 1.2 stuff) we noticed that a Telegesis stick is not working anymore.

It gets stuck on the call there it tries to set the "Prompt 1 register":

```
2019-05-22T16:42:39.496+0200   931  DEBUG          [symbolicName=org.openhab.binding.zigbee.telegesis, className=org.openhab.binding.zigbee.telegesis.handler.TelegesisHandler] - Initializing ZigBee Telegesis serial bridge handler.
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Initializing ZigBee network [zigbee:coordinator_telegesis:05001B8F].
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Channel 0
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - PANID XXX
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - EPANID XXX
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Network Key null
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Link Key null
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Config: zigbee_initialise found, initializeNetwork=false
2019-05-22T16:42:39.496+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Network Key String 
2019-05-22T16:42:39.536+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Network Key String has invalid format. Revert to default key.
2019-05-22T16:42:39.626+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Network key initialised 6352FBD0C1839B7AE418E39D8B23CDDD
2019-05-22T16:42:39.636+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Network key final array 6352FBD0C1839B7AE418E39D8B23CDDD
2019-05-22T16:42:39.636+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Link Key String 
2019-05-22T16:42:39.636+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Link Key String has invalid format. Revert to default key.
2019-05-22T16:42:39.666+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Link key final array 5A6967426565416C6C69616E63653039
2019-05-22T16:42:39.996+0200   931  DEBUG          [symbolicName=org.openhab.binding.zigbee.telegesis, className=org.openhab.binding.zigbee.telegesis.handler.TelegesisHandler] - ZigBee Telegesis Coordinator opening Port:'/dev/ttyUSB0' PAN:2a, EPAN:0000000000000042, Channel:0
2019-05-22T16:42:40.086+0200   148  DEBUG          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.common.registry.AbstractManagedProvider] - Added new element zigbee:coordinator_telegesis:05001B8F to ManagedThingProvider.
2019-05-22T16:42:40.146+0200   154  DEBUG          [symbolicName=org.eclipse.smarthome.core.thing, className=org.eclipse.smarthome.core.thing.events.ThingStatusInfoEvent] - 'zigbee:coordinator_telegesis:05001B8F' updated: UNKNOWN
2019-05-22T16:42:40.156+0200   154  DEBUG          [symbolicName=org.eclipse.smarthome.core.thing, className=org.eclipse.smarthome.core.thing.events.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:05001B8F' changed from INITIALIZING to UNKNOWN
2019-05-22T16:42:41.086+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - ZigBee network starting
2019-05-22T16:42:41.086+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Initialising ZigBee coordinator
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - Default: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=2, interTransactionDelay=50, maxRetries=2]
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - Broadcast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - Multicast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
2019-05-22T16:42:41.086+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Mesh Update Period 86400
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Adding supported cluster 1280
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNode] - ClusterMatcher starting
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNode] - ClusterMatcher adding cluster 1280
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Adding supported cluster 25
2019-05-22T16:42:41.086+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNode] - ClusterMatcher adding cluster 25
2019-05-22T16:42:41.116+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.internal.ZigBeeNetworkStateSerializerImpl] - Loading ZigBee network state: Start.
2019-05-22T16:42:41.116+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.internal.ZigBeeNetworkStateSerializerImpl] - Loading ZigBee network state: File does not exist
2019-05-22T16:42:41.136+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Network state is updated to INITIALISING
2019-05-22T16:42:41.136+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - null: networkStateUpdated called with state=INITIALISING
2019-05-22T16:42:41.266+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Telegesis dongle initialize.
2019-05-22T16:42:41.266+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Connecting to serial port [/dev/ttyUSB0] at 19200 baud, flow control FLOWCONTROL_OUT_NONE.
2019-05-22T16:42:41.306+0200   927  DEBUG          [symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Serial port [/dev/ttyUSB0] is initialized.

This is the error:

2019-05-22T16:44:02.276+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Error setting Telegesis Prompt 1 register
2019-05-22T16:44:02.276+0200   930  ERROR          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Unable to configure Telegesis dongle
2019-05-22T16:44:02.286+0200   925  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Network state is updated to OFFLINE
```

I just tested the latest OH-2.5 snapshot with ZSS-1.1.6 and the stick works correctly.

@cdjackson  Any idea what could have gone wrong or where to start debugging?

cdjackson, 2019-05-23:
It is working fine for me and if I check the repo, there were no changes to the handler in 1.2, just declaration changes - the last code changes were 3 to 4 months ago.

triller-telekom, 2019-05-28:
We have analyzed this a bit further and its a timing issue and the "Prompt 1 register" is just the result of it, whereas the bug occurs earlier.

When the `TelegesisFrameHandler` starts up it schedules the networkstate polling via `scheduleNetworkStatePolling()`. However, this starts with just 1 second delay.

On a slow machine it might happen, that the commands for setting the bits on the stick in `ZigbeeDongleTelegesis#initialiseCriticalConfiguration()` are not received by the chip, yet.

The following log shows that the `TelegesisDisplayNetworkInformationCommand` clashes right into the sequence of `TelegesisSetRegisterBitCommand`s:

```
2019-05-28T12:47:59.113+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TelegesisFrameHandler thread started
2019-05-28T12:47:59.173+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-05-28T12:47:59.183+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
2019-05-28T12:47:59.193+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis Data:ATS124=1

2019-05-28T12:48:02.213+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TELEGESIS Timer: Timeout 0
2019-05-28T12:48:03.133+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-05-28T12:48:03.133+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
2019-05-28T12:48:03.143+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2019-05-28T12:48:03.323+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,11,08,E313,9F755577FBB61A4B
2019-05-28T12:48:06.153+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TELEGESIS Timer: Timeout 1
2019-05-28T12:48:07.133+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-05-28T12:48:07.133+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
2019-05-28T12:48:07.133+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2019-05-28T12:48:07.183+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
2019-05-28T12:48:07.183+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-05-28T12:48:07.313+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,11,08,E313,9F755577FBB61A4B
2019-05-28T12:48:10.143+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TELEGESIS Timer: Timeout 2
2019-05-28T12:48:10.143+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - Error: number of retries exceeded [3].
2019-05-28T12:48:15.193+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
[... cut ...]
2019-05-28T12:48:31.213+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
2019-05-28T12:48:31.213+0200   930  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 4
...
```

Thus I suggest to increase the delay before we start the networkstate polling.

cdjackson, 2019-06-05:
Closed by #647

closed 2019-06-05 by cdjackson