zsmartsystems/com.zsmartsystems.zigbee #644: NPE in ZigBeeNetworkManager on network shutdown

TomTravaglino, 2019-05-28:
In `ZigBeeNetworkManager.setTransportState(ZigBeeTransportState)` a NPE is thrown when processing this statement
```
if (!validTransportStateTransitions.get(networkState).contains(state)) {
```
In that moment the value of `networkState` is `SHUTDOWN` which is not contained in the collection of valid state transitions, and thus `validTransportStateTransitions.get(networkState)` returns `null` followed by a NPE. My proposal is to add a transition `SHUTDOWN -> OFFLINE` to the collection. I will prepare a PR.

cdjackson, 2019-05-29:
I'm not sure that SHUTDOWN to OFFLINE should be allowed. Once the network is SHUTDOWN, it should not be allowed to change state as it has been shutdown.

triller-telekom, 2019-05-29:
If this transition is not allowed, I guess we stumbled across a bug. here is the exception to the explanation from Tommaso above:

```
2019-05-28T14:26:09.679+0200   159  ERROR          [symbolicName=org.eclipse.smarthome.core, className=org.eclipse.smarthome.core.internal.common.AbstractInvocationHandler] - An error occurred while calling method 'ThingHandler.dispose()' on 'org.openhab.binding.zigbee.telegesis.handler.TelegesisHandler@e325c6': null
java.lang.NullPointerException
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.setTransportState(ZigBeeNetworkManager.java:987)
	at com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis.shutdown(ZigBeeDongleTelegesis.java:476)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.shutdown(ZigBeeNetworkManager.java:564)
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.dispose(ZigBeeCoordinatorHandler.java:327)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.smarthome.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:153)
	at org.eclipse.smarthome.core.internal.common.Invocation.call(Invocation.java:53)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

closed 2019-05-29 by cdjackson

triller-telekom, 2019-05-29:
No you do not miss anything, changing the state which will be set from OFFLINE to SHUTDOWN would also be sufficient, thank you.

PR #646 (2019-05-29): Added SHUTDOWN->OFFLINE to allowed transitions in ZigBeeNetworkManager
Resolves #644