zsmartsystems/com.zsmartsystems.zigbee #651: NodeServiceDiscovery wrongly maps transactions to zigbee nodes

triller-telekom, 2019-06-07:
[log3.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3266353/log3.txt)
There seems to be an issue with a mapping of nodes to IEEE addresses in the discovery somewhere. We see that our discovery never completes, i.e. there is no DiscoveryResult created for openHAB.

Somehow we have ONE IEEE=00124B000420A2B5 address, but TWO nodes: 42581 and 33788. We see in the sniffer that 
42581 is the correct node which is responding to our requests.

Before we started the scan for new devices:

```
$zigbee nodes

Network  Addr  IEEE Address       Logical Type   EP   Profile                    Device Type      Manufacturer     Model
      0  0000  000D6F000AF902EA   COORDINATOR

```

After the scan with no DiscoveryResult for openHAB:

```
$zigbee nodes

Network  Addr  IEEE Address       Logical Type   EP   Profile                    Device Type      Manufacturer     Model
      0  0000  000D6F000AF902EA   COORDINATOR
  33788  83FC  00124B000420A2B5   UNKNOWN

```

So there is a wrong node in the NetworkManager somehow.

The problem seems to be within the NodeServiceDiscoveryTask:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/36f8412da5f6e2f812cd42a7f0b02768772de427/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeNodeServiceDiscoverer.java#L585-L619

There is the call to `networkManager.updateNode(node)` which overrides the content of the current node with the information from the node given as an argument. Whenever one of the discovery subtasks finishes, it updates the node. However, for the "wrong" node there are timeouts and thus no values available.

When there are timeouts for the non existing 33788 node, we will overwrite the already successfully discovered properties for 42581, because they both have the same IEEE address (That is because `networkManager.updateNode(node)` does: `currentNode = networkNodes.get(node.getIeeeAddress());`.

@cdjackson: Would be nice if you could have a look please, since you are more familiar with the code.

The log was done using a telegesis Dongle on one of our very slow devices, so maybe timing could also play a role as a side effect here.

cdjackson, 2019-06-07:
Please can you explain what you mean by -:

> Somehow we have ONE IEEE=00124B000420A2B5 address, but TWO nodes: 42581 and 33788
It is normal for a node to change address - especially if it changes parent. Since nodes are always registered in the system by their IEEE address, which is always static, and can only have one NWK address, I’m not sure what you mean by this. You also only show one node below (other than the coordinator) - not two, so I’m a bit confused.

My guess with what you are seeing is the node changed address, and possibly the system isn’t picking that change up,

triller-telekom, 2019-06-11:
I can only explain what we observed. We saw that we had one device and somehow ZSS was still sending requests to the "old" network address of that node (how this got in, I don't know).

The requests to this non-existing "old" network address did all time out. Requests to the "new" address where answered correctly. This led us to the odd situation where the node was updated during discovery with correct data from the "new" address and then these answers where overridden with `null` values from the timed out requests and the discovery never finished successfully.

That is why I said in my post before that I think there is a mismatch in transaction mapping, because apparently requests to one network address are considered as belonging to other requests, sent to another network address, during discovery.

cdjackson, 2019-06-11:
Ok, thanks. I'm still not sure I fully understand but I will take a look. Maybe in the discovery the node is cached (I think it is but need to check) and therefore maybe this doesn't get updated if the network manager updates the NWK address.

That said, one of the first things the discovery does is to request the NWK address given the IEEE address in order to avoid this sort of thing as NWK address is expected to change.

cdjackson, 2019-06-11:
Firstly, excuse the rambling thoughts that may be presented below - I'm just dumping my log analyses for starters...

> Whenever one of the discovery subtasks finishes, it updates the node.

The node should only be updated once ALL discovery tasks are complete. It only calls ```updateNode``` when there are no further discovery tasks.

When it does the update, the message ```Node SVC Discovery: complete``` is logged -:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/36f8412da5f6e2f812cd42a7f0b02768772de427/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeNodeServiceDiscoverer.java#L587

I don't see this in the log at all, so I don't think that this is the issue.

When the device starts, we see the ANNOUNCE message where the device announces its new address -:

[image]

At almost exactly the same time, the discovery is starting, and it is requesting the NWK address. This is the first thing that discovery does in case the address changed while the system was down -:

[image]

There is no response to this and it is retried - this time with a response -:

[image]
[image]

We see the response in the log, and the transaction is processed -:

```
2019-06-07T14:40:25.803+0200   2546 DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: NetworkAddressResponse [42581/0 -> 0/0, cluster=8000, TID=--, status=SUCCESS, ieeeAddrRemoteDev=00124B000420A2B5, nwkAddrRemoteDev=42581, startIndex=null, nwkAddrAssocDevList=[]]
2019-06-07T14:40:25.843+0200   2546 DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransaction] - Transaction response received - waiting TX_ACK: ZigBeeTransaction [queueTime=490, state=RESPONDED, sendCnt=1, command=NetworkAddressRequest [0/0 -> 65535/0, cluster=0000, TID=BC, ieeeAddr=00124B000420A2B5, requestType=0, startIndex=0]]
```

However in the ```ZigBeeNodeServiceDiscoverer ``` there is no message logged at that time - there is a log message for the processing of this a little later. I think it must be the same one as there is no other message in the meantime -:

```
2019-06-07T14:40:33.733+0200   2546 DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 42581: Node SVC Discovery: NetworkAddressRequest returned NetworkAddressResponse [42581/0 -> 0/0, cluster=8000, TID=--, status=SUCCESS, ieeeAddrRemoteDev=00124B000420A2B5, nwkAddrRemoteDev=42581, startIndex=null, nwkAddrAssocDevList=[]]
2019-06-07T14:40:33.733+0200   2546 DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B000420A2B5: Node SVC Discovery: request NWK_ADDRESS successful. Advancing to ACTIVE_ENDPOINTS.
```

I'm not sure why this transaction completion message is delayed and  I don't see any way for this to be out of order as the code is sequential - unless maybe they are different threads, but there is no indication of this (what is the 2546 that is logged in your log?) -:

When this transaction completes, it should update the NWK address and move to the request to send the ```ACTIVE_ENDPOINTS``` phase, and the following log shows it sending to the new address -:

[image]

```
2019-06-07T14:40:33.843+0200   2546 DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - 00124B000420A2B5: Added transaction to queue, len=1, transaction=ZigBeeTransaction [queueTime=0, state=WAITING, sendCnt=0, command=ActiveEndpointsRequest [0/0 -> 42581/0, cluster=0005, TID=--, nwkAddrOfInterest=42581]]
```

So, I think this is all working correctly.

If I look back further in this log, I see other activities ongoing with this address, so my guess is that there are other requests for this node, so I think that the problem started before this log starts. I see that there was probably already a discovery started for the device as seen below -:

[image]

I also see that you probably rejoined this device (which is probably also why it has a new NWK address -:

[image]

[image]

I think from this it's hard to see what is happening - are you able to provide a complete log please? My guess at the moment is that there was already a network discovery started with the original NWK address. In theory, when the new node is added we see the node discoverer starting, but I think that there was probably already one running. Looking at the discovery extension, this can happen if the node is re-added.

What I don't understand from the above is the first list doesn't contain this node, and the second one does. 

[image]

However, we can see from the log that the node is in fact known even before you discovered it here, so I'm wondering why this is? Did you remove this node somehow? That would explain everything if you started the system, removed the node when there was already a discoverer running. This discoverer would have referenced the old ```ZigBeeNode``` with the old address. When you remove this node, it is removed from the ```ZigBeeNetworkManager``` list and doesn't appear in your list of nodes above. However, this node does still exist within a running service discoverer.

Does that sound like what happened? It's about the only explanation I can see.

If that is correct, then I think the only issue is that when you removed the node, the discoverer remained, and we should look at how to kill this off as well.

triller-telekom, 2019-06-11:
Thank you for this very detailed analysis.

> Did you remove this node somehow? That would explain everything if you started the system, removed the node when there was already a discoverer running

Yes, the node was there in the first place and we have removed all nodes, except for the coordinator with `zigbee leave fnr <id>`.

> When you remove this node, it is removed from the ZigBeeNetworkManager list and doesn't appear in your list of nodes above. However, this node does still exist within a running service discoverer.

We assumed that if we remove a node, we would kill everything that is somehow associated with that node.

So yes, your explanation sounds plausible and meets what we saw while the log was created. Having the old discoverer still active in an old thread would explain why the new discovery thread never finishes, because the answers are mixed.

cdjackson, 2019-06-11:
> We assumed that if we remove a node, we would kill everything that is somehow associated with that node.

That's a fair assumption :) I'll take a look at removing this.

> Having the old discoverer still active in an old thread would explain why the new discovery thread never finishes, because the answers are mixed.

I think the key element is that the ```ZigBeeNode``` that is running in the old thread is not connected via the network manager. If it was still linked there, then the problem would probably not be seen as it would have been ensured that only one node existed, and when the NWK address changed, it would be updated.

Anyway, I'll rename this issue to avoid confusion and will take a look at this over the coming days.

cdjackson, 2019-06-12:
Closed by #652

closed 2019-06-12 by cdjackson