zsmartsystems/com.zsmartsystems.zigbee #654: IndexOutOfBoundsException

YordanDZhelev, 2019-06-13:
`IndexOutOfBoundsException` exception thrown while reading firmware file during firmware update of Bitron smart plug.
The exception occures at https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/otaserver/ZigBeeOtaFile.java#L423
After the second invocation of that method inside the while loop  https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/otaserver/ZigBeeOtaFile.java#L283
Do we really need this loop to read the OTA file?
After I removed this code I was able to successfully update firmware of Bitron 902010/28 from 0x0000000f to 0x00000030.
Since I updted the FW of my device I'm not able to reproduce the bug, therefore not able to provide stacktrace.

YordanDZhelev, 2019-06-13:
@cdjackson could you please take a look at that code. Do we use the information retrieved by this loop while reading the OTA file?

cdjackson, 2019-06-13:
The loop is required to read the file, which is what this code is doing.

Can you provide the file so I can test please?

YordanDZhelev, 2019-06-13:
I assume that I don't completely understand the code, so correct me if I'm wrong, but after we read all the metadata, which happens in `readOtaFile()` before we enter the loop all fields that we have getters for are already set. Then we enter the loop reading remaining bytes and not setting any fields based on that data. Also, as I described above, I was able to update the FW of a device successfully with the while loop removed. 
Here is the FW file which cause the `IndexOutOfBoundsException`, but with the while loop completely removed FW update works.
[902010-28-v48.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3286313/902010-28-v48.zip)

cdjackson, 2019-06-13:
Sure - I appreciate that the loop COULD be removed, but for debugging it is useful and I would therefore prefer to solve the issue without removing this.

YordanDZhelev, 2019-06-13:
Is there any documentation how the file should be structured? Because to me it looks that the metadata in the beginnign of the file is as we expected it to be, afther that the validation fails somehow inside the loop. However, the file is accepted by the device. If the code is based on a certain specification, then the file is not following that specification.

cdjackson, 2019-06-13:
Yes, the file is documented in the ZA docs.

YordanDZhelev, 2019-06-14:
After some research and looking at the bytes of the file provided above, it looks that the file does NOT follow the file structure described in the ZigBee Alliance documentation. The second tag after the headers appears to be with lenght 82 C0 E7 10 which evaluates to a number way beyond the file length. I susspect that it is a manufacturer specific file. @cdjackson I'm open to suggestions on how to fix this problem.

cdjackson, 2019-06-14:
Thanks for the investigation. If that's the case, I think probably the best thing is to add some sanity checks to the file.  I'm a little surprised that it gets this far with a custom file as if I remember correctly there is a file signature (I think the first 4 bytes need to be a specific value) and this should avoid this.

Let me take a look - unless you want to do something? We could simply trap the exception and ignore it - ie just logging something?

PR #655 (2019-06-14): Add sanity checking when reading OTA files
This adds some sanity checking when reading the OTA files.
Closes #654
closed 2019-06-14 by cdjackson