zsmartsystems/com.zsmartsystems.zigbee #663: ASH: ERROR received (code 81). Disconnecting.

openhab-5iver, 2019-06-16:
Moving over from https://github.com/openhab/org.openhab.binding.zigbee/issues/440. There are lots of details in that issue.

The issue started when using 1.1.11 libraries, but also occurred using 1.1.6. As reported [in this comment](https://github.com/openhab/org.openhab.binding.zigbee/issues/440#issuecomment-502476354), the error has stopped occurring when not using a manual install of the binding and a patch to openhab-core. The high load issue in the current OH snapshots could just be revealing an underlying issue in the libraries.

cdjackson, 2019-06-20:
I'm still discussing this with Silabs support - it's slow going...

At the moment, when an error is received the ASH handler disconnects. What I will look at is to retry the RESET if we're not connected - that might help and we can try it in the absence of info from Silabs.

cdjackson, 2019-06-21:
I spent some time trying to replicate this. By frigging around with the ASH protocol and putting a hack in to stop sending frames after some time, I can make the dongle go into error mode -:

```
18:26:54.125  DEBUG  ASH: TX EZSP queue size: 1
18:26:54.125  DEBUG  ******* LETS BLOW THIS BABY SKY HIGH
18:26:54.452  DEBUG  <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=1, reTx=true, data=79 80 FF 00 0B 02 FF FF]
18:26:54.453  DEBUG  ******* LETS BLOW THIS BABY SKY HIGH
18:26:57.579  DEBUG  <-- RX ASH frame: AshFrameError [version=2, errorCode=81, Error: Exceeded maximum ACK timeout count]
18:26:57.579  WARN   ASH: ERROR received (code 81).
18:26:57.579  DEBUG  ASH: ERROR received (code 81). Disconnecting.
```

Unfortunately though when I restart, the reset seems to work fine -:

```
18:27:47.566  DEBUG  --> TX ASH frame: AshFrameRst []
18:27:47.617  DEBUG  TX EZSP: EzspVersionRequest [desiredProtocolVersion=4]
18:27:47.624  DEBUG  ASH: TX EZSP queue size: 1
18:27:48.967  DEBUG  <-- RX ASH frame: AshFrameRstAck [version=2, resetCode=11, Reset: Software]
18:27:48.967  DEBUG  ASH: Connected
```

Currently in the driver whenever the ```ERROR``` is received, the driver is taken ```OFFLINE```. I was trying to test a modification I made to the ASH layer to resend the reset if the ```ERROR``` message is received before it's connected, but since in my case I can't get it to return an error after the restart, I can't test this.

@openhab-5iver I think I will add this to the driver anyway. Are you still getting these errors? If so, please can you try and test?

openhab-5iver, 2019-06-21:
The high CPU issue in OHC has been resolved, which prevents this error from occurring. I also have not been able to manually install the binding for a while, so testing will be tough, if not impossible.

cdjackson, 2019-08-20:
After a lot of discussion with Silabs their only real answer is that in some conditions it is possible for the NCP to stop receiving commands via the serial port if it gets into a bad state. While I fully agree with this, it's not what I think is happening since the NCP is responding in the cases we see here (and I've provided these logs). However, I've really reached the end of the line after 2 months or more of discussion that is going nowhere so I need to conclude this (initially their statement was that the ASH RST was not a real command that is sent over the wire - after about 3 weeks of discussion they finally agreed it was!!!).

So, all I can really offer is their statement that an ASH software reset is not considered reliable, and vendors must implement a hardware reset. I have therefore implemented a feature to allow users to provide a mechanism to perform such a reset. This is therefore closed by #720.

closed 2019-08-20 by cdjackson

openhab-5iver, 2019-08-21:
Do you plan to implement something in the binding to use this? None of my zigbee bulbs responded this morning and I found this... 

```
2019-08-21 05:27:54.672 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
```

cdjackson, 2019-08-21:
What do you suggest?

Currently there is already a system to restart the communications, but if there is no way to reset the device, then I don't think that I can do anything in the binding.

Unless you know of a way to reset the dongle, I can't do anything. I tried to work out if I could reset the CEL dongle using (for example) DTR, but this didn't work. I don't know if the HZSUSB-1 (whatever it's called) has a hardware reset as I don't have one to test with - I guess not, but I don't know.  If you can confirm, then I can potentially add something to support this.

openhab-5iver, 2019-08-22:
Sorry, no... I have no idea how you'd reset the HUSBZB-1.

cdjackson, 2019-08-22:
I did think about trying to contact the manufacturer, but the GoControl website has lots of different contact addresses depending on the product type and it wasn't clear where the HUSBZB fits. Maybe you want to try and chase this to see if you get any response?

The hooks are in place in the latest libs to add this if we can work out how (or if possible!) to reset the device through hardware.

openhab-5iver, 2019-08-22:
I've tried contacting Nortek a few times, but they've never responded. I'd give it a shot, but honestly, the priority for me would be pretty low.