zsmartsystems/com.zsmartsystems.zigbee #669: Cannot reconnect to roter devices after coordinator restarts

tunjid, 2019-06-28:
**Describe the bug**
When the coordinator is rebooted, previously paired routers do not reconnect to it, despite the successful storage and restoration of the routers to the ```DataStore```.

**To Reproduce**
I've been able to port console application to Android and successfully connect [Gledopto](https://www.amazon.com/GLEDOPTO-Dimmable-Compatible-SmartThings-Lightify/dp/B07QXS57FG/ref=sr_1_7?keywords=gledopto&qid=1561690068&s=gateway&sr=8-7&th=1) light bulbs using the TI CC2531 as seen [here](https://github.com/tunjid/BluetoothRcSwitch/blob/feature/zigbee/app/src/main/java/com/tunjid/rcswitchcontrol/nsd/protocols/ZigBeeProtocol.kt). It works a charm on first pairing, invoking ```ZigBeeConsoleNetworkStartCommand``` successfully causes the bulbs to join the network.

However, when the Android app is closed and reopened, the bulbs do not rejoin the network. A [ZigBeeDataStore](https://github.com/tunjid/BluetoothRcSwitch/blob/feature/zigbee/app/src/main/java/com/tunjid/rcswitchcontrol/data/persistence/ZigBeeDataStore.kt) is used, and the previous devices are restored on start up, they just don't reconnect. I've tried using the ```RediscoverCommand``` to try and rediscover the bulbs from their IEEE Address, but that doesn't seem to work well either.

**Expected behavior**
Previously connected devices should reconnect after the coordinator restarts.

**Additional context**
These are the console logs after restart:

```
2019-06-27 22:58:58.793 6147-6188/com.tunjid.rcswitchcontrol I/ZIGBEE: ZigBee network state updated to INITIALISING
2019-06-27 22:59:01.681 6147-6186/com.tunjid.rcswitchcontrol I/ZIGBEE: PAN ID          = 8194
2019-06-27 22:59:01.707 6147-6186/com.tunjid.rcswitchcontrol I/ZIGBEE: Extended PAN ID = 0000000987654321
2019-06-27 22:59:01.731 6147-6186/com.tunjid.rcswitchcontrol I/ZIGBEE: Channel         = CHANNEL_11
2019-06-27 22:59:01.847 6147-6186/com.tunjid.rcswitchcontrol I/ZIGBEE: *** Resetting network  * Channel                = 11  * PAN ID                 = 8192  * Extended PAN ID        = 0000000987654321  * Link Key               = 5A6967426565416C6C69616E63653039  * Network Key            = 77699859C1214B14CBD0EA25B38D3C35
2019-06-27 22:59:09.424 6147-6186/com.tunjid.rcswitchcontrol I/ZIGBEE: ZigBee console starting up ... [OK]
2019-06-27 22:59:09.537 6147-6208/com.tunjid.rcswitchcontrol I/ZIGBEE: Node Added ZigBeeNode [state=UNKNOWN, IEEE=00124B001C42FA37, NWK=35B2, Type=ROUTER]
2019-06-27 22:59:09.565 6147-6211/com.tunjid.rcswitchcontrol I/ZIGBEE: ZigBee network state updated to ONLINE
2019-06-27 22:59:09.564 6147-6187/com.tunjid.rcswitchcontrol I/ZIGBEE: Node Added ZigBeeNode [state=UNKNOWN, IEEE=00124B001C42ECCB, NWK=476D, Type=ROUTER]
2019-06-27 22:59:09.576 6147-6216/com.tunjid.rcswitchcontrol I/ZIGBEE: Node Added ZigBeeNode [state=UNKNOWN, IEEE=00124B0018E199A3, NWK=0000, Type=COORDINATOR]
2019-06-27 22:59:47.340 6147-6177/com.tunjid.rcswitchcontrol I/ZIGBEE: Sending rediscovery request for address 00124B001C42ECCB
2019-06-27 23:00:11.937 6147-6221/com.tunjid.rcswitchcontrol I/ZIGBEE: Node Added ZigBeeNode [state=UNKNOWN, IEEE=00124B0018E199A3, NWK=0000, Type=COORDINATOR]
```

cdjackson, 2019-06-28:
If you reset the network, it will not work. I don't completely recognise the logging here, so I can't be sure, but I see ```*** Resetting network ``` so I'm guessing this is resetting the network.

tunjid, 2019-06-29:
> If you reset the network, it will not work. I don't completely recognise the logging here, so I can't be sure, but I see `*** Resetting network ` so I'm guessing this is resetting the network.

Oh thank you! I currently reset on restart, much appreciated!

closed 2019-06-29 by tunjid