zsmartsystems/com.zsmartsystems.zigbee #673: Misuses of cryptographic APIs

mhp0rtal, 2019-07-02:
Hi
The following lines have cryptographic API misuses. <br> File name => com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/security/MmoHash.java: Line number => 152: API name => Cipher:First parameter (with value "AES/ECB/NoPadding") should be any of AES/{CBC, GCM, PCBC, CTR, CTS, CFB, OFB}<BR> File name => com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/security/MmoHash.java: Line number => 151: API name => SecretKeySpec:<BR> File name => com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/security/MmoHash.java: Line number => 154: API name => Cipher:<BR> File name => com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/security/MmoHash.java: Line number => 151: API name => SecretKeySpec:<BR> File name => com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/security/MmoHash.java: Line number => 154: API name => Cipher:<BR>

cdjackson, 2019-07-02:
Please feel free to provide a PR to suggest a solution.

Thanks.

cdjackson, 2019-07-13:
You state -:

> Cipher:First parameter (with value "AES/ECB/NoPadding") should be any of AES/{CBC, GCM, PCBC, CTR, CTS, CFB, OFB}

However the documentation (https://docs.oracle.com/javase/8/docs/api/javax/crypto/Cipher.html) states that the value ```AES/ECB/NoPadding``` is valid and must be supported by all implementation -:

>Every implementation of the Java platform is required to support the following standard Cipher transformations with the keysizes in parentheses:
AES/CBC/NoPadding (128)
AES/CBC/PKCS5Padding (128)
AES/ECB/NoPadding (128)

Since this is a simple hash use where we are simply generating a known key from a specific code (always the same hash will be generated from the same code) I don't really see any issue here. 

If you have further comments, please provide a PR that resolves the issue you are reporting for this specific usage.

closed 2019-07-20 by cdjackson