zsmartsystems/com.zsmartsystems.zigbee #681: Questions about Zigbee gateway choice and compatibility

sraillard, 2019-07-10:
Dear all,

This is not an issue related to the library, but more some questioning about the Zigbee ecosystem and especially how to choose a Zigbee gateway to be used with openHAB. I would like first to use an USB dongle but later my goal would be to use deported homemade gateways using serial over IP.

Silabs, TI and NXP seem to be the largest players in the Zigbee world, from a chip maker point of view. I'll let NXP here as there is no support for their chips in this library.

Silabs has acquired Ember in 2012 and Telegesis en 2015, so they have two lines of products.
TI is providing its Z-Stack with two versions, the older "archived" and the new one that is Zigbee 3.0 compatible.

So, I can see 4 major options for a gateway:
* Silabs Telegesis ETRX357 modules (not recommended for new design) and the ETRX3 USB dongle (EOL) that expose a simplified AT-Commands serial interface. While this dongle seems to be largely recommended by the community, it doesn't look like future proof and doesn't seem to support Zigbee 3.0
* Silabs Ember chips running the EmberZNet PRO Zigbee Protocol Stack and exposing a serial interface protocol called EZSP. This library is up to date, is supporting Zigbee 3.0 and is certified. But it isn't easy to find USB dongles or modules that tell the exact version of the library embedded. And it doesn't seem to be easy to upgrade or flash a new version as the binary files don't seem to be public. 
* TI older archived Z-Stack firmware that is exposing a serial interface protocol called ZNP
* TI newer Z-Stack firmware 3.x that seems to expose also the ZNP protocol and has been certified for Zigbee 3.0

This is my own analysis, I'm not a Zigbee specialist, so I expect to have made errors. But based on it, here are my questions:
* Can we say that the two serial protocols that are looking future proof are the EZSP of Silabs and the ZNP of TI?
* For EZSP, as there are many different Silabs chips that can support it, and different library versions, is there any recommendation for the firmware version supported by this project? Or specific chips supported?
* For ZNP, can we know which version of the Z-Stack is supported? Can we use the newer Z-Stack version 3.x? Can other chips like CC2530 (without USB) or CC2538 (more powerful) be used?
* Can this library communicate reliably with Zigbee 3.0 certified devices?
* By pure curiosity (and this isn't a remark), why the NXP chips like JN5168 are not supported?

Thank you for any corrections and information that can be provided!

cdjackson, 2019-07-10:
> Can we say that the two serial protocols that are looking future proof are the EZSP of Silabs and the ZNP of TI?

No - Silabs NCP firmware receives regular updates. TI have stated they will not update their ZNP at least for the 2531 any more.

> For EZSP, as there are many different Silabs chips that can support it, and different library versions, is there any recommendation for the firmware version supported by this project? Or specific chips supported?

Any NCP firmware should be fine in general.

> For ZNP, can we know which version of the Z-Stack is supported? Can we use the newer Z-Stack version 3.x? Can other chips like CC2530 (without USB) or CC2538 (more powerful) be used?

At the moment the ZB3.0.3 is not supported, however I have written a new driver for this at a customer request, but in the process found that it is no longer supported.

> Can this library communicate reliably with Zigbee 3.0 certified devices?

Of course.

> By pure curiosity (and this isn't a remark), why the NXP chips like JN5168 are not supported?

There's only so much time in the day - unless a customer wants it, I'm not writing drivers for no reason.

sraillard, 2019-07-13:
Thank you Chris for these answers!

So, the best way to go is to use the Silabs NCP firmware.

Silabs doesn't provide a USB dongle with a NCP chip supporting EZSP, the only one I have found is the ZM357S-USB from CEL: this dongle from CEL is using the EM357 circuit from Silabs and is preloaded with the EmberZNet PRO stack. If this device is a good starting point, maybe it can be added in the README for helping new users?

Regarding the Silabls NCP, does-it have like the TI CC2531 a restriction on the number of direct childrens it can communicate with? It have seen a limit of 15 direct childrens for the CC2531 and it doesn't look like large if mesh isn't an option.

cdjackson, 2019-07-14:
> the only one I have found is the ZM357S-USB from CEL

There are a few different versions available with EM357 and EM358 chipsets.

I have never managed to buy a CEL stick that has the NCP stack and bootloader pre-loaded. Every one I have purchased has had no software (and I have purchased 20 or 30 of them) so I'm a little hesitant to say that this is common. I know it's advertised as being available, but it's generally not in my experience.

> Regarding the Silabls NCP, does-it have like the TI CC2531 a restriction on the number of direct childrens it can communicate with?

Yes, there is always a restriction on all ZigBee routers with the number of devices that can connect. It is configurable, but limited to a maximum of 16 nodes.

sraillard, 2019-07-14:
Reading their datasheet, I was thinking the CEL USB sticks were preloaded with the NCP stack! I have ordered some of them, I'll check when they will arrive.

Can you advise some USB sticks that can be ordered with a NCP stack preloaded?

cdjackson, 2019-07-14:
I’ve not checked recently, but I have checked a number in the past, and they were all blank. Programming them isn’t so easy as you need the devkit to get access to the firmware, and also you need the programmer.

In the EU there is a Bitron device sold by DT (Qivicon) but it has special PID/VID codes which means it doesn’t work on Mac/Win. You can work around it on Linux IFRC.

sraillard, 2019-07-14:
I have just received a confirmation from CEL that their USB sticks are not preloaded.

That would have been great to have a provider of USB stick based on the recent MGM111 module and preloaded with the NCP Embserznet stack, but it's looking like a dream!

I have found explanations about limitations in the Ember stack (https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2014/03/24/guidelines_for_large-pgj4):
* Maximum of 16 routers
* Maximum of 64 end devices

cdjackson, 2019-07-14:
>  it's looking like a dream!

I am looking at manufacturing one with the EFR32 - I have the prototype boards here but need to find some more time.

> I have found explanations about limitations in the Ember stack 

This is also in the standard documentation, but it says 16 routes (as I mentioned earlier) and 32 children (not 64).

cdjackson, 2019-07-31:
Closing this as I think the question has been answered. If there's anything further please reopen.

closed 2019-07-31 by cdjackson