zsmartsystems/com.zsmartsystems.zigbee #682: Memory Leak (Overflow) using Ember Coordinator

Bennedetto, 2019-07-11:
I am not sure, if this is the proper place for me bug report to be posted. Please correct me if I should move it somewhere else.

The scenario is using OpenHab 2.4 with the zigbee binding. As a coordinator the "Ember EM35x Coordinator" (USB-Stick) is used. 
Wait - OpenHab is causing some issues: wrong location?!

Nope, As OpenHab is frequently crashing with some error-messages (like "[ERROR] [org.apache.felix.fileinstall        ] - In main loop, we have serious trouble
java.lang.OutOfMemoryError: Java heap space"). I configured OpenHab to create a HeapDump everytime it crashes.

This leads me to the assumption, that this project is the root cause. As I analyzed the HeapDump with Eclipse Memory Analyzer (MAT), it gives me some nice images. Please have a look

[image]

And the resulting report is:
[image]

As I had a look to the implementation of the ember AshFrameHandler (https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/ash/AshFrameHandler.java) this seems like a desired behaviour - to save all ever send packages and keep them forever.

My setup has 9 ikea tradfri bulbs. As I guess, this is not much, I suggest to introduce something like a expiration date for the packages to keep - but I have no idea of the mechanisms of the protokoll, thus I have no Idea how to tackle this issue down. All I know is, that using a Telegesis Coordinator does not result in that issue (same binding, same OpenHab version, just another stick).

cdjackson, 2019-07-11:
Please can you define what version of the library you're using? I suspect it's very old in which case you should upgrade to something newer.

For example, OH2.4 is about 8 months old.

cdjackson, 2019-07-11:
I think we can close this for now. If the problem persists with a newer version, then please reopen.

closed 2019-07-11 by cdjackson

Bennedetto, 2019-07-11:
The lib used is 1.1.6 which seems to be quite old. I'll build the openhab plugin using a more recent version and give a report whether the memory overflow is gone or not. This may take some time, but I'll try to give an update as soon as possible.

cdjackson, 2019-07-11:
As suggested, please use the latest version of OH. There is a script that will allow you to update.