zsmartsystems/com.zsmartsystems.zigbee #691: RejectedExecutionException in ZigBeeDongleEzsp.handlePacket()

TomTravaglino, 2019-07-23:
With #637 the instantiation of the Executor was moved from the ìnitialize()` method to the constructor. Currently we're observing an issue when the instance of `ZigBeeDongleEzsp` is reused/reinitialized after is has been shut down. In that case the Executor is not instantiated again and so the Exception mentioned above is thrown. Fyi: In our case this happenes when the reconnect job in the zigbee binding takes action.

My proposal is to have a check in the `initialize()` method and re-initialize the Executor if it was shut down.

cdjackson, 2019-07-23:
Currently you can't call ```initialize``` more than once - and I think once shutdown the system can't be restarted. There are other things elsewhere that likely won't work as well, and I think just reinitialising this executor won't be the end of the story.

It is probably cleaner to instantiate a new handler when initialising the Ember dongle through either of the (from memory) 3 entry points (MfgLib, firmware and normal useage).

What is your use case and what do you mean exactly by "reused"?

TomTravaglino, 2019-07-23:
>What is your use case and what do you mean exactly by "reused"?

As mentioned above the use case is the reconnect job in the binding (`ZigBeeCoordinatorHandler`). When the reconnect job runs it creates a new instance of `ZigBeeNetworkManager` whose constructor accepts the `ZigBeeTransportTransmit`. However the coordinator handler doesn't create a new instance of (in our case) `ZigBeeDongleEzsp` and instead passes the instance which has previously shut down.

cdjackson, 2019-07-23:
> As mentioned above the use case is the reconnect job in the binding (ZigBeeCoordinatorHandler).

You didn't actually mention this above ;)

I think in with this use case, you should really create a new transport when restarting the network. Once something is shut down, it should not be restarted. This was a discussion we had some time back about the overall lifecycle of the network and it should really extend throughout the system.

When the network closes, it closes down all elements, and I think it will be very messy to restart all of them. In your use case, you will also need to handle this same issue in multiple places in multiple drivers (eg TI, Ember, XBee, ConBee).

TomTravaglino, 2019-07-24:
OK. Then let's close this one. I will think about another approach.

closed 2019-07-24 by TomTravaglino