zsmartsystems/com.zsmartsystems.zigbee #692: NotificationService cannot be recreated after shutdown

triller-telekom, 2019-07-30:
In PR #686 the scheduler in the `NotificationService` is shut down once the `ZigbeeNetworkManager` is shut down:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/b9c434d061a12f7b3f2887049368a232f1639cf7/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/internal/NotificationService.java#L31

However, at least the openHAB binding uses a `reconnectPollingTimer` and shuts down the networkmanager as well in order to initialize it again:

https://github.com/openhab/org.openhab.binding.zigbee/blob/master/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeCoordinatorHandler.java#L331

But after the shutdown there is no way to reinitialize the stack because the `executorService` in the `NotificationService` is `static`. So one cannot instantiate a new class of type `NotificationService`.

cdjackson, 2019-07-30:
As per the recent discussion with @TomTravaglino in #691, once the ```ZigBeeNetworkManager``` is shutdown, it cannot be restarted.

The documentation regarding to ```ZigBeeNetworkManager``` also states this -:

 * To shutdown the network and free all resources, call {@link #shutdown()}. The network may not be restarted and a new
 * {@link ZigBeeNetworkManager} must be instantiated. The network state will be {@link ZigBeeNetworkState#SHUTDOWN}.

triller-telekom, 2019-07-31:
Maybe there is a slight misunderstanding.

The binding does not "restart" the `ZigBeeNetworkManager` but it creates a **new** instance of the `ZigBeeNetworkManager`. This new instance will use the very **same** `static` `NotificationService` class and thus the shutdown executor. Do you see the problem now?

So even if you create a new instance of `ZigBeeNetworkManager`, you won't get a working zigbee stack again, because you have to shutdown the whole JVM to get rid of the `static NotificationService` which contains the shutdown executor.

cdjackson, 2019-07-31:
Ah - ok, interesting point that I'd not considered.  I'll have a think about this.

Thanks.

PR #699 (2019-07-31): Add ability to reinitialise NotificationService
Closes #692
closed 2019-07-31 by cdjackson