zsmartsystems/com.zsmartsystems.zigbee #693: java.lang.ArrayIndexOutOfBoundsException in EzspFrame reintroduced see #586 and #641

triller-telekom, 2019-07-30:
PR #688 reintroduces the issue #586 which was fixed in #594, see

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/688/files#diff-29a5822c0fa62c8f8e2df9c53b307d9aL396

the try-catch block was removed, leading to the exception and a non working stack if there is an invalid frame.

This has happened for the second time now, see issue #641 and its fix #642.

cdjackson, 2019-07-30:
I think the issue here is that this change was introduced into the autogenerated code, so it gets overwritten when the autocoder is run.

triller-telekom, 2019-07-31:
Thanks for pointing this out. Should have read the comment in the class about this O_o

I have created an PR to fix this in the code and in the auto-coder, please have a look.

cdjackson, 2019-07-31:
Should be fixed now :)

closed 2019-07-31 by cdjackson