zsmartsystems/com.zsmartsystems.zigbee #709: Problem with ZDO transaction matcher when short response received

cdjackson, 2019-08-11:
It seems that some devices responding with a non ```SUCCESS``` response (eg ```NOT_SUPPORTED```) will not respond with the ```nwkAddrOfInterest``` field. This prevents the matcher from correlating the response with the request, resulting in the transaction timing out even though it was actually successful.

The following ```PowerDescriptorResponse``` is received from a UbiSys S2 (although admittedly not a new one).

```
15:17:06.564  DEBUG  RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0, clusterId=32771, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=59], lastHopLqi=230, lastHopRssi=-57, sender=61585, bindingIndex=255, addressIndex=255, messageContents=84 84]
15:17:06.564  DEBUG  RX APS: ZigBeeApsFrame [sourceAddress=61585/0, destinationAddress=0/0, profile=0000, cluster=8003, addressMode=null, radius=0, apsSecurity=false, apsCounter=59, payload=84 84]
15:17:06.564  DEBUG  RX CMD: PowerDescriptorResponse [61585/0 -> 0/0, cluster=8003, TID=--, status=NOT_SUPPORTED, nwkAddrOfInterest=null, powerDescriptor=null]
```

@AfromD how old is the firmware in your devices? I wonder if you are able to provide a log from your system with these devices? This should be achieved by deleting the XML file that OH creates for this device (or moving to another folder so you can replace it if needed) then restart the OH binding. If it does not come back after a minute or so, then check if the XML file is back. If it is, then please provide the log - if not, click the switch and hopefully it should be generated - then provide the log.

I found this while working on the configuration for the S2 switch discussed in https://github.com/openhab/org.openhab.binding.zigbee/issues/321 so would like to resolve this so I can look at the S2 configuration problem.

cdjackson, 2019-08-11:
It seems that the only option here is to use the source address. However I recall in the past that some ZDO messages to battery devices will actually get the response from their parent - so source address and nwkAddrOfInterest are not the same (normally they should be, but there are exceptions).

AfromD, 2019-08-11:
This is what I get in PaperUI
`

hardwareVersion | 5
-- | --
modelId | S2 (5502)
vendor | ubisys
zigbee_datecode | 20170718-DE-FB0
`
So I would suspect the firmware is from July 17
Do I need to switch on zsmartsystems debug level logging?

cdjackson, 2019-08-11:
Thanks for the quick response :)

By removing the request for the PowerDescriptor, I an at least get the discovery to complete.  The info from my device is -:

```
Node Info     Value
APPVERSION    1
DATE          20150317-DE-FB0
HWVERSION     0
MANUFACTURER  ubisys
MODEL         S2 (5502)
STKVERSION    1
ZCLVERSION    4
```

So we have the same device (5502) but I have different hardware and firmware versions. I can probably flash the firmware, but I don't want to do that right now at least as I'd like to use this device to work around the matcher problem here.

Are you using the latest OH2.5? I don't think this issue is new to the latest version as the matchers have worked this way for a long while.

So... Yes, if you can enable the logging for ZSmartSystems (as per the OH binding docs) then that would be good. I'd like to see what your S2 responds with so I know how big a problem this is.

I need to look at an issue with ZWave now, but will try to get the S2 configuration sorted this week if I can.

AfromD, 2019-08-11:
I'm on 2.5 M1 for OH and the binding, zsmartsystems library is on 1.1.6

AfromD, 2019-08-11:
I am not sure which XML file to delete. Thus far, I have found the XML file for my ember dongle which the binding creates. Is that it?

cdjackson, 2019-08-11:
Ok, the XML is different for your version so this doesn’t work so well.

Let’s not worry about this - I will see if I can flash the firmware for my S2 to check this. Either way I need to have a think about how to resolve the matcher issue here.

Note that I have managed to read the configuration information from the S2, although I’m not quite sure how to write this at the moment. I have some ideas to at least get this working for now.  Do you still need to change these settings on the S2?

AfromD, 2019-08-11:
I think I will update to M2 in the next few days.

I have modified my switches to momentary with bits of foam. Works quite well. So right now I'm ok, but I could see how this is an issue to others. Another thing I had to "hack" was to get the switch inputs by watching the output. I have a rule that watches for changes, and then checks whether there was a command (OH turned it on) or not (the switch must have turned it on).

cdjackson, 2019-08-11:
If/when you update to M2, it would be great if you could grab the logs when the device initialises - I’d definitely like to see these.

Hopefully I’ll have a way to set these parameters in the next week or two - bits of foam may work, but doesn’t sound ideal ;)

Thanks.

AfromD, 2019-08-13:
[openhab.log](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3497321/openhab.log)
Restarting the bundles on Karaf didn't produce the expected result (maybe it was me), so I did a full restart of the OH service. Then the file appeared. I have attached the log starting from the restart.

cdjackson, 2019-08-13:
Perfect - thanks. That shows that your device is correctly responding to the PowerDescriptor request so I guess this bug was fixed by Ubisys.

Cheers
Chris

cdjackson, 2019-08-26:
I believe that this is an invalid response. According to the ZigBee spec, the PowerDescriptor can be omitted if the response reports an error, but the address should be included. I guess this is why UbiSys have fixed this in newer versions.

I'm not sure it is easy to work around this since without the address it's hard to correlate the response.

closed 2019-11-08 by stale[bot]