zsmartsystems/com.zsmartsystems.zigbee #712: Centralite plug turns off and on by itself

TomTravaglino, 2019-08-13:
We're experiencing an issue with a Centralite 3200 Smart Outlet. After pairing the device and having it turned on we can observe the following behavior: About every 5 minutes the plug turns off by itself and after about 1.5 seconds it turns on again. In the sniffer logs we see a `ZCL: ReportAttributes` when the plug turns on but nothing when it turns off. We also see that this always occurs in combination with these commands:
[image]

And there is another anomaly in the logs: Every 2 seconds the plug broadcasts a `IEEE Address Request` and does not get a response.

We can reliably reproduce all this on two copies.

I attached a zip containing runtime logs and the sniffer logs produced by Simplicity Studio. The network key for the sniffer logs is `C42FA032417A93E4EC6EA460302B35D0`.

[logs.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3495394/logs.zip)

TomTravaglino, 2019-08-23:
@cdjackson Friendly reminder. Do you have an idea what's wrong with the device?

cdjackson, 2019-08-23:
Sorry - I’ve not really looked at this as it seems to be an issue with the device from your description.

cdjackson, 2019-08-26:
> every 5 minutes the plug turns off by itself and after about 1.5 seconds it turns on again.

The only thing I can think of here is that this is a feature of the device, possibly in relation to how the library is handling some low level acks (I think I mentioned this to @triller-telekom in our last call). I might be wrong though as it would be non-standard and therefore the only way to know is from the device manufacturer.

> And there is another anomaly in the logs: Every 2 seconds the plug broadcasts a IEEE Address Request and does not get a response.

What response are you expecting? Devices are allowed to send such requests - the framework does not have to reply and in this case the request does not appear valid from my initial look.

cdjackson, 2019-09-22:
Closing this due to inactivity. If the issue persists, and you think it's related to the framework, then please provide further information.

closed 2019-10-02 by cdjackson