zsmartsystems/com.zsmartsystems.zigbee #719: Battery powered devices are re-added to the node list after being removed

TomTravaglino, 2019-08-20:
Battery powered devices are re-added to the node list after being removed

We're observing that battery powered devices are still in the node list after they have been removed from the ZigBee network. We researched the issue and saw that the node is first removed from the node list and later re-added. This happens because after the node has been removed the frame handler (ZigBeeDongleEzsp) receives a `EzspChildJoinHandler` and subsequently starts processing an `UNSECURED_JOIN` which ends up with the node being re-added to the node list. What's strange about the `EzspChildJoinHandler` is that I can't see it in the sniffer logs when I remove the device.

I used a Centralite 3315 (water sensor) and a Centralite 3305 (motion sensor) to reproduce this issue. I attached the logs (runtime and Simplicity Studio) produced when I removed the 3305. The network key for the sniffer logs is C42FA032417A93E4EC6EA460302B35D0.

[logs.zip](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3519802/logs.zip)

TomTravaglino, 2019-08-20:
Is this issue already fixed in #716?

cdjackson, 2019-08-20:
Yes, I think this will be fixed in #716 as the ```EzspChildJoinHandler``` notification is for the device leaving.

TomTravaglino, 2019-08-22:
We can close this one.

closed 2019-08-22 by TomTravaglino