zsmartsystems/com.zsmartsystems.zigbee #731: Incorrect MatchDescriptionRensponse when nwkAddrOfInterest is a broadcast address?

TomTravaglino, 2019-08-30:
In #712 we reported an abnormal behavior of the Centralite 3200 SmartPlug. I investigated the issue and found a possible reason for it in the the class `ClusterMatcher` where the `MatchDescriptorResponse` is created. There the `nwkAddrOfInterest` used to create the `MatchDescriptorResponse` is always the one provided by the `MatchDescriptorRequest`, even then when `nwkAddrOfInterest` is a broadcast address (`0xFFFD`). Apparently this broadcast address is responsible for the behavior described in #712.
I made a small change to the code so that the `localNwkAddress` is put into `MatchDescriptorResponse` when we have to do with a broadcast address. Several tests showed that none of the issues described in #712 occurred anymore.

This is the change I made in the `ClusterMatcher`
```
    @Override
    public void commandReceived(ZigBeeCommand command) {
        // If we have local servers matching the request, then we need to respond
        if (command instanceof MatchDescriptorRequest) {
...            
            Integer nwkAddrOfInterest = matchRequest.getNwkAddrOfInterest();
            if (nwkAddrOfInterest == 0xFFFD) {
                matchResponse.setNwkAddrOfInterest(networkManager.getLocalNwkAddress());
            } else {
                matchResponse.setNwkAddrOfInterest(nwkAddrOfInterest);
            }
...
        }
    }

```

Is our assumption correct that the `nwkAddrOfInterest` in the `MatchDescriptorResponse` must be the `localNwkAddress` when the `MatchDescriptorRequest` comes with a broadcast address?

cdjackson, 2019-08-30:
Thanks @TomTravaglino. For completeness can you post the received messages here as well so we have this all captured.

I agree that something similar to what you have suggested is probably needed - I would not use the explicit address check, but there is a ```ZigBeeBroadcastAddress.isBroadcastAddress()``` method available for this which will also check other broadcast addresses (note that this method name is from memory so it might be slightly different).

TomTravaglino, 2019-08-30:
> For completeness can you post the received messages here as well so we have this all captured.

I guess the logs should contain the messages you're asking for.
[logs.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/3559359/logs.txt)

cdjackson, 2019-08-30:
Ok, I will look through the logs to find them, but I was hoping that the log entries that you are talking about could be simply added to the issue so that we have this captured here without having to do that!

cdjackson, 2019-08-30:
For completeness, this is the log entry that I assume you are referring to -:

```
2019-08-30T11:28:10.895+0200   3122 DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: MatchDescriptorRequest [9637/0 -> 0/0, cluster=0006, TID=--, nwkAddrOfInterest=65533, profileId=260, inClusterList=[25], outClusterList=[]]
```

TomTravaglino, 2019-08-30:
Sorry, I misunderstood your request. I thought you were asking for detailed communication logs in order to better understand what happens there.

cdjackson, 2019-08-30:
No problem.

I guess you know, but the lack of response from the binding to this request will not matter, and will not be the reasons for the switch turning on/off. This is an OTA server request - not every network will have an OTA server, and this is simply the device trying to see if there is on. It won't (or shouldn't!) cause the device to power on/off.

TomTravaglino, 2019-09-02:
I understand your explanation that this issue shouldn't be the cause for the behavior described in #712. Probably the device doesn't handle the response (where nwkAddrOfInterest = 0xFFFD) as it should and the mentioned behavior is just a side effect. Anyway, I tested your PR #732 and the issues mentioned in #712 don't occur anymore.

closed 2019-09-02 by cdjackson

cdjackson, 2019-09-02:
My point is not really that the device doesn't handle the response - my point is that it is irrelevant in any case. A network does not have to have an OTA server - so most networks will not respond to this message at all and the device must not expect to have a response. This is simply a search for services - if the network has no such service, then there will be no response and the device should operate normally.

PR #732 (2019-09-02): Manage broadcast addresses in cluster matcher
This adds address checking to the cluster matcher to ensure that the request is addressed to our local address or a broadcast address, and then ensures that the response uses the local address.
Fixes #731 
@TomTravaglino this should solve the issue you reported - as mentioned though this will not be related to the issue you report in #712.