zsmartsystems/com.zsmartsystems.zigbee #739: How to find out the "type" of ZigbeeCommand received?

lumin-admin, 2019-09-05:
It would be great if you could help me out with this one. I was relying on `ZclCommandType` and `ZdoCommandType` enum to find out the type(and name) of the ZigbeeCommand received in `commandReceived(ZigBeeCommand command)` callback. 
The way I was doing it was I was iterating over all the `ZclCommandType` and `ZdoCommandType` enum values to see if the received command was an instance of any of the commands contained in those enums.
Since ` ZclCommandType` enum was removed from the library recently, can you suggest an alternative way to find out the type of the `ZigbeeCommand` received?

cdjackson, 2019-09-05:
I'm not sure how you are listening for these commands? Is it through the cluster command listener (this is the recommended way)?

In general, the easiest way is to simply use ```instanceof``` to work out the type of a command.

eg -:

```
if (command instanceof ZclSomeCommandType) {
   someCommand = (ZclSomeCommandType) command;
   someCommand.getSomeMethod();
   ...
}
```

lumin-admin, 2019-09-05:
Thanks for the reply. We use `zigbeeNetworkManager.addCommandListener(listener)` to listen for the commands.
We do use the approach that you mentioned but there's a scenario where we want to log all the 'type' of incoming commands to our internal database. So was wondering if there's any data structure(similar to `ZclCommandType` enum) so that we could iterate over the values to find the 'type' of the commands and log those?

lumin-admin, 2019-09-05:
Actually, I just realized that I could use `zigBeeCommand.javaClass.simpleName`(in Kotlin) to log the subclass/type of incoming `ZigBeeCommand`.

cdjackson, 2019-09-06:
This seems like the easiest approach. There is no complete list of commands available as manufacturers are able to add their own. This is why the enum was removed as relying on this enum limited the system by not allowing the manufacturer specific extensions.

cdjackson, 2019-09-06:
Just one thought - there is no guarantee of uniqueness with the simple name so you might want to use the full name - depending on your use case.

cdjackson, 2019-09-06:
I think this is answered now - shall we close this?

lumin-admin, 2019-09-06:
That makes sense. Thanks for your replies.
I'll close the issue.

closed 2019-09-06 by lumin-admin