zsmartsystems/com.zsmartsystems.zigbee #749: Weird IEEE Address Requests from samjin motion devices (Smartthings motion detector 2018)

sesuncedu, 2019-09-11:
I'm pretty sure these are being generated incorrectly on the device, but I wanted to double check, as the destination address doesn't seem to be recorded properly(?)

The messages are being reported up to the Ezsp handler as being of type EMBER_INCOMING_BROADCAST, which should be impossible for a request of this kind; however the APS frame seems to end up with a destination address of 0/0, which is what I would expect if it were unicast. 

The network address of interest is 0xFFFD ( Broadcast address for ```macRxOnWhenIdle = TRUE```) which seems a bit off.

Am I reading this right? If so, then the c.z.z issue is where, if anywhere, the destination address should be being recorded? 

Has anyone else seen similar behavior?  Each device is doing this every 7-8 seconds, which is probably not wasting as much battery as I think it is, but represents the vast bulk of all traffic. Is there anything that might have induced this behavior?

(of course, given that these devices report a value for PowerSource of 0x03, rather than setting bit 3, so they are apparently using mains power).  

##### Log Sample

``` 
[e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=1, ackNum=7, reTx=false, data=17 90 45 04 00 00 01 00 00 00 08 00 00 00 83 FF BC 61 2C FF FF 05 C5 FD FF 00 00 04]

[zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_BROADCAST, apsFrame=EmberApsFrame [profileId=0, clusterId=1, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=83], lastHopLqi=255, lastHopRssi=-68, sender=11361, bindingIndex=255, addressIndex=255, messageContents=C5 FD FF 00 00]

[tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=11361/0, destinationAddress=0/0, profile=0000, cluster=0001, addressMode=null, radius=0, apsSecurity=false, apsCounter=83, payload=C5 FD FF 00 00]

[tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: IeeeAddressRequest [11361/0 -> 0/0, cluster=0001, TID=--, nwkAddrOfInterest=65533, requestType=0, startIndex=0]

[transaction.ZigBeeTransactionManager] - notifyTransactionCommand: IeeeAddressRequest [11361/0 -> 0/0, cluster=0001, TID=--, nwkAddrOfInterest=65533, requestType=0, startIndex=0] 
```

cdjackson, 2019-09-12:
I have seen a number of devices doing this and I don't think it's correct either. I discussed this with one of the Silabs FAEs when we were debugging an issue earlier this year and he also wasn't clear why that was doing this.

So I would also agree with you - to me, this device is asking for the IEEE address of the device that has the network address ```FFFD```, which is not a valid unicast address (as you point out, this is one of the broadcast addresses).

sesuncedu, 2019-09-13:
It doesn't seem like there's any obvious way to figure out what the destination address for an incoming broadcast message was when using ezsp *sigh*. All we know is that it was a broadcast. 

Currently the destination address in the incoming ZigbeeApsFrame is being set in ZigBeeDongleEzsp::handlePacket : 

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/2d6f26fb9de58d01f8abc9b3e737dff4b5dd4baa/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ZigBeeDongleEzsp.java#L632

This is correct when ```incomingMessage.getType()``` is ```EMBER_INCOMING_UNICAST``` or ```EMBER_INCOMING_UNICAST_REPLY```, but is wrong for Broadcast and Multicast.

The destination address field in ZigbeeApsFrame (ZAF) is defined as ```int```, so there are bits available to indicate unknown broadcast address- alternatively one of the reserved values could be used, on the assumption that it will fire an error if it's copied and used ass an address (I have a suspicion that something similar to this lies at the root of the problem with the device in question). 

There's a few other things that aren't being translated cleanly

#### Consider adding an Enumeration and field corresponding to the APS Delivery Mode
The enumeration should have three values, corresponding to UNICAST, GROUP, and BROADCAST.
The value of the field might not match an APS Frame (e.g if  network multicast were to intervene). 

#### When getType() indicates multicast:
* The groupId  in the ember APS frame should be copied into the ZAF groupAddress 
* The groupId in the ember APS frame should be copied into the ZAF destinationAddress
* The address mode in the ZAF should be set to GROUP
* The delivery mode should be set to GROUP
* the getter for ZAF destination endpoint should either throw an exception or be set to an illegal value    

#### When getType() indicates broadcast:
*  destinationAddress should be set to an address as discussed above.  possibilities
     *  ```0x0fffd``` (rx_on)
     *  ```0x0fffe``` (reserved broadcast))
     * ```-1``` (invalid address value,  low 16 bits are 0xffff (all hands broadcast))
* The address mode in the ZAF should be set to DEVICE
* The delivery mode should be set to BROADCAST

cdjackson, 2019-09-14:
> It doesn't seem like there's any obvious way to figure out what the destination address for an incoming broadcast message was when using ezsp sigh. All we know is that it was a broadcast.

Isn't it ```FFFD```? I believe it is - have you actually checked this using a sniffer?

Regarding multicast - this is not supported at the moment. There is an open issue on this and it's something I will look at soon.

sesuncedu, 2019-09-20:
[The broadcast address may not be available and ZigbeeAPSFrame is carrying the indication.]

I need to get a sniffer since there are some more wrinkles- the ieee addr request  issue doesn't seem to happen when the sensor is directly connected to the coordinator, but seems to kick off when there's a router in the way (in this case, GE/JASCO wall switches).  I need to sniff in situ, and then  to have something else be a router.

cdjackson, 2019-09-20:
Remember the APS Frame, and the ZDO Frame are different layers of the protocol.

sesuncedu, 2019-09-25:
I think I  may have found the problem, and it looks like it should be fixed by #732  

I set up a brand new unit whilst sniffing with wireshark+nrf52840 dongle. That let me catch the problem starting; there was a suspicious action right before things went pear-shaped.  I then went to the openhab logs and found the appropriate entries. 

1. The device (33752) broadcast a match_req for ZHA (260) OTA Cluster (25). The address of interest was 65533 (broadcast, rx-on).

```
2019-09-25 14:38:50.013 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 
   RX CMD: MatchDescriptorRequest [33752/0 -> 0/0, cluster=0006, TID=--,
      nwkAddrOfInterest=65533, profileId=260, inClusterList=[25], outClusterList=[]]
```
2. openhab / czz sent a match_rsp, with the address of interest field set to 65533 
```
2019-09-25 14:38:50.020 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 
   TX CMD: MatchDescriptorResponse [0/0 -> 33752/0, cluster=8006, TID=2C, 
        status=SUCCESS, nwkAddrOfInterest=65533, matchList=[1]]
```

3. After a couple more seconds of fast-polling, the device  issued it's first weird ieee request. 

This fits with the expected behavior from ZCL section 11.8 -Discovery of the Upgrade Server.
>After discovering the OTA server’s short ID via the ZDO Match descriptor request
the client SHALL discover the IEEE address of the upgrade server via ZDO IEEE address discovery command and store the value in UpgradeServerID attribute

So it should be fixed by the change to the cluster matcher.  Now to see if I can make openhab deploy an updated bundle (I guess I should fix and create PRs for both projects with proper OSGI versions)

cdjackson, 2019-09-25:
> Now to see if I can make openhab deploy an updated bundle

This is already included in openHAB

sesuncedu, 2019-09-25:
I switched to openhab unstable,  but that pulled in czz-1.2.1?

cdjackson, 2019-09-25:
I’m not sure what you are doing - maybe you should discuss on the openhab forum to ensure you get the right support. The latest oh uses 1.2.3

sesuncedu, 2019-09-25:
That's an openhab issue - the features.xml in openhab-distro  refers to old versions. 

I fixed  and deployed from a local feature, and can confirm that the issue is fixed.

Now the devices are sending OTA requests.  They're not being handled, but if they were, I don't have samsung firmware files, and if I did, and they fixed the bug, it doesn't really matter because the issue is fixed...

closed 2019-09-25 by sesuncedu