zsmartsystems/com.zsmartsystems.zigbee #751: Telegesis test failure

cdjackson, 2019-09-14:
Following the merge of #738 we have occasional test failures with CI. I ran this 3 times, and it failed twice.

@yoadey can you have a look at this please. I've not looked at this, but this is the sort of thing I was concerned about where tests may run ok (most of the time) on local computers, but when run on slower/different cloud CI servers we end up with occasional failures.

```
-------------------------------------------------------
 T E S T S
-------------------------------------------------------
Running com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesisTest
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Tests run: 13, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 2.711 sec
Running com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisEventFactoryTest
Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.012 sec
Running com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandlerTest
Tests run: 10, Failures: 1, Errors: 0, Skipped: 1, Time elapsed: 4.076 sec <<< FAILURE!
testRetriesReset(com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandlerTest)  Time elapsed: 3.687 sec  <<< FAILURE!
org.mockito.exceptions.verification.NeverWantedButInvoked: 
zigBeeDongleTelegesis.notifyStateUpdate(
    false
);
Never wanted here:
-> at com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandlerTest.testRetriesReset(TelegesisFrameHandlerTest.java:153)
But invoked here:
-> at com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler$2.run(TelegesisFrameHandler.java:655)
	at com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandlerTest.testRetriesReset(TelegesisFrameHandlerTest.java:153)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.apache.maven.surefire.junit4.JUnit4Provider.execute(JUnit4Provider.java:252)
	at org.apache.maven.surefire.junit4.JUnit4Provider.executeTestSet(JUnit4Provider.java:141)
	at org.apache.maven.surefire.junit4.JUnit4Provider.invoke(JUnit4Provider.java:112)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.apache.maven.surefire.util.ReflectionUtils.invokeMethodWithArray(ReflectionUtils.java:189)
	at org.apache.maven.surefire.booter.ProviderFactory$ProviderProxy.invoke(ProviderFactory.java:165)
	at org.apache.maven.surefire.booter.ProviderFactory.invokeProvider(ProviderFactory.java:85)
	at org.apache.maven.surefire.booter.ForkedBooter.runSuitesInProcess(ForkedBooter.java:115)
	at org.apache.maven.surefire.booter.ForkedBooter.main(ForkedBooter.java:75)
```

yoadey, 2019-09-14:
Will take a look at it, when I'm at home, I thought I had handled all possibilities for this. Tested it with muß smaller timeouts on local machine and always worked, would it be OK if I add a logger during tests, as it is difficulz

cdjackson, 2019-09-14:
Thanks @cdjackson - sure - I have no problem if you want to add the logger.

For now I've ```@Ignore```d the test so there's no major hurry.

cdjackson, 2019-10-25:
@yoadey is there any update on this please?

closed 2020-03-08 by stale[bot]