zsmartsystems/com.zsmartsystems.zigbee #763: UNSUP_GENERAL_COMMAND DefaultResponse generated by incoming default response

sesuncedu, 2019-09-27:
Something odd is going on here. I haven't looked deeper yet (only had ZigbeeNetworkManager logging  at debug) 

### Log extract. 

#### Send OffComand

- Command is sent with default response enabled. 
```
TX CMD: OffCommand [On/Off: 0/0 -> 9036/1, cluster=0006, TID=88]
TX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND,
    manufacturerSpecific=false, direction=CLIENT_TO_SERVER,
    disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=136, commandId=0]
```

#### Receive DefaultResponse

```
RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, 
     manufacturerSpecific=false, direction=SERVER_TO_CLIENT,
     disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=136, commandId=11]
RX CMD: DefaultResponse [On/Off: 9036/1 -> 0/1, cluster=0006, TID=88,
       commandIdentifier=0, statusCode=SUCCESS]
```

- disable default response flag is not set by device
- I believe it should be, since this is a single response to a command

#### Send DefaultResponse

Response is sent in response to default response. 

___

- Unsupported command error
- Maybe something isn't being consumed? Transaction related? 
- disable default response isn't set here either.
```
TX CMD: DefaultResponse [On/Off: 0/0 -> 9036/1, cluster=0006, TID=88,
        commandIdentifier=11, statusCode=UNSUP_GENERAL_COMMAND]
TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, 
        manufacturerSpecific=false, direction=CLIENT_TO_SERVER,
        disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=136, commandId=11]
```

cdjackson, 2019-09-27:
To me, it seems correct - if you send an ```OnOff``` command to a device, it will respond with a ```DefaultResponse```. This is normal and required by the standard.

Obviously the ```DefaultResponse``` should not be sent in response to a ```DefaultResponse```. I will fix this.

sesuncedu, 2019-09-27:
Default reply to default is explicitly forbidden, but there seem to be
other cases where this is happening.

One example seems to involve duplicated report configuration responses (
with a suspicious RX_NAK that requires restarting with  ASH logging
enabled).

I am also seeing correct default responses for unsupported cluster commands
sent to these switches for the client endpoint  OnOff client cluster (that
seems to be an openhab issue).

cdjackson, 2019-09-27:
> Default reply to default is explicitly forbidden

Yes, I agree, and this is what the PR I've created fixes.

> One example

??

Please provide the example. Without logs it's very hard for me to understand what you have seen and what the example is.

sesuncedu, 2019-09-28:
"Here's one I made earlier".

Should I open a new ticket or do you want to reopen this one?

7C is the relevant TID.

```ConfigureReportingCommand``` induces ```DefaultResponse``` with 
```Power Configuration: 0/0 -> 2760/1, cluster=0001, TID=7C, commandIdentifier=7, statusCode=UNSUP_GENERAL_COMMAND]]```

[Note: There's a ```SUCCESS``` DefaultResponse for another transaction on the queue. ]

```
2019-09-27 16:18:47.836 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - 
  RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=260, clusterId=1, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=7E], lastHopLqi=255, lastHopRssi=-60, sender=2760, bindingIndex=255, addressIndex=255, messageContents=08 7C 07 00]

2019-09-27 16:18:47.836 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 
   RX APS: ZigBeeApsFrame [sourceAddress=2760/1, destinationAddress=0/1, profile=0104, cluster=0001, addressMode=null, radius=0, apsSecurity=false, apsCounter=7E, payload=08 7C 07 00]

2019-09-27 16:18:47.837 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 
    RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=124, commandId=7]

2019-09-27 16:18:47.838 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 
    RX CMD: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null]
 
2019-09-27 16:18:47.838 [DEBUG] [transaction.ZigBeeTransactionManager] - 
   notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] 

2019-09-27 16:18:47.839 [DEBUG] [transaction.ZigBeeTransactionManager] - 
     notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=16729, state=DISPATCHED, sendCnt=2, command=ConfigureReportingCommand [Power Configuration: 0/0 -> 2760/1, cluster=0001, TID=7C, records=[AttributeReportingConfigurationRecord: [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=33, direction=0, minimumReportingInterval=600, maximumReportingInterval=7200, reportableChange=1]]]]

2019-09-27 16:18:47.840 [DEBUG] [transaction.ZigBeeTransactionManager] - 
    notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=1595, state=DISPATCHED, sendCnt=1, command=ManagementLqiRequest [0/0 -> 9036/0, cluster=0031, TID=8A, startIndex=0]]

2019-09-27 16:18:47.840 [DEBUG] [transaction.ZigBeeTransactionManager] - 
    Transaction complete: ZigBeeTransaction [queueTime=16730, state=COMPLETE, sendCnt=2, command=ConfigureReportingCommand [Power Configuration: 0/0 -> 2760/1, cluster=0001, TID=7C, records=[AttributeReportingConfigurationRecord: [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=33, direction=0, minimumReportingInterval=600, maximumReportingInterval=7200, reportableChange=1]]]]

2019-09-27 16:18:47.841 [DEBUG] [transaction.ZigBeeTransactionManager] - 
    notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=14286, state=DISPATCHED, sendCnt=1, command=DefaultResponse [Temperature Measurement: 0/0 -> 21671/1, cluster=0402, TID=50, commandIdentifier=10, statusCode=SUCCESS]]
 
2019-09-27 16:18:47.841 [DEBUG] [transaction.ZigBeeTransactionManager] - 
    notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=2215, state=DISPATCHED, sendCnt=1, command=ConfigureReportingCommand [Level Control: 0/0 -> 46495/1, cluster=0008, TID=89, records=[AttributeReportingConfigurationRecord: [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, direction=0, minimumReportingInterval=0, maximumReportingInterval=900, reportableChange=1]]]]

2019-09-27 16:18:47.842 [DEBUG] [transaction.ZigBeeTransactionManager] - 
     notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=12609, state=DISPATCHED, sendCnt=1, command=IeeeAddressRequest [0/0 -> 33752/0, cluster=0001, TID=8C, nwkAddrOfInterest=33752, requestType=1, startIndex=0]]

2019-09-27 16:18:47.843 [DEBUG] [transaction.ZigBeeTransactionManager] - 
    notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=11271, state=DISPATCHED, sendCnt=1, command=ReadAttributesCommand [Temperature Measurement: 0/0 -> 2676/1, cluster=0402, TID=8B, identifiers=[0]]]

2019-09-27 16:18:47.844 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ConfigureReportingResponse [Power Configuration: 2760/1 -> 0/1, cluster=0001, TID=7C, status=SUCCESS, records=null] ZigBeeTransaction [queueTime=266, state=DISPATCHED, sendCnt=1, command=ManagementRoutingRequest [0/0 -> 46495/0, cluster=0032, TID=8D, startIndex=0]]

2019-09-27 16:18:47.845 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - 
[... cut ...]
    286D97000102418F: Added transaction to queue, len=6, transaction=ZigBeeTransaction [queueTime=0, state=WAITING, sendCnt=0, command=DefaultResponse [Power Configuration: 0/0 -> 2760/1, cluster=0001, TID=7C, commandIdentifier=7, statusCode=UNSUP_GENERAL_COMMAND]]

2019-09-27 16:18:47.846 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - 
    <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=0, reTx=false, data=B8 90 59 C8 0A 8F 41 02 01 00 97 6D 28 FF C4 01 4C 23]

2019-09-27 16:18:47.846 [DEBUG] [e.transaction.ZigBeeTransactionQueue] -
     286D97000102418F: transactionComplete COMPLETE 0
```

cdjackson, 2019-09-28:
I'm clearly missing something here. Can you explain what this issue is? I assumed that it was related to your earlier comment about "One example seems to involve duplicated report configuration responses". There is however only one command in this log, so if it's duplicated, I can't see what it's duplicated with?

Also, can you supply logs without changing them (eg adding more line breaks) as it makes it impossible to process without having to change everything back (thanks).

sesuncedu, 2019-09-28:
Will not edit in future.

This sequence of packets has shows end of transaction 7C. 

1. RX  ConfigureReportingResponse- EZSP, APS, ZCL, CMD
5. TransactionManager completing transaction. 
6. Transaction added for default response with UNSUP_GENERAL_COMMAND.

cdjackson, 2019-09-28:
Ok, I was looking for the packet duplication as per your report and not for this response. I will take a look at this.

cdjackson, 2019-09-28:
I suggest to open a new issue for this as it's quite a different problem than the original issue,

closed 2019-09-28 by cdjackson

sesuncedu, 2019-09-28:
[markers for new ticket]

I'm still not sure duplication isn't involved, but it's speculation and not
as obvious in the log bits above. The request retransmitted, but I didn't
dig out the prior log entries to see if an earlier reply ever arrived.

I will look for a better set of logs and matched sniffage. I'll look for
something that is non-sleepy for simplicity.

cdjackson, 2019-09-28:
> I'm still not sure duplication isn't involved

There is no duplication in the log you've provided, so unless there is something to work with I think we can for now ignore it.  If you find logs showing some sort of duplication, then please open an issue.

PR #764 (2019-09-27): Don't send a DefaultResponse to a DefaultResponse
This doesn't send the received ```DefaultResponse``` to listeners. This should not be needed by listeners, and this also prevents us sending a ```DefaultResponse``` to a received ```DefaultResponse```.

Closes #763