zsmartsystems/com.zsmartsystems.zigbee #766: Ember: coordinator should not set  APS_OPTION_ENABLE_ROUTE_DISCOVERY

sesuncedu, 2019-09-28:
Pe Si Labs: https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2014/03/24/guidelines_for_large-pgj4
>In order to ensure the benefits of source routing and many-to-one routing are achieved, the concentrator should make sure to suppress route discovery (disallow the setting of EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY) for outgoing messages, instead deriving source routes from incoming Route Records. 

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/75e8931f597f6c21939a7d37b1014b92b6c4ddc1/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ZigBeeDongleEzsp.java#L610

cdjackson, 2019-09-29:
Clearly this is conditional on the system being a coordinator so this would require some thought.

sesuncedu, 2019-09-29:
I think you've already done all the prep for this - the status is  saved in the config (it looks like it's only true if coordinating is successfully enabled.

sesuncedu, 2019-09-29:
(Also Si Labs ZB documentation is so much more helpful than Nordic/DSR's).

sesuncedu, 2019-10-07:
I am now not sure if source routes are ever being used in the current set-up. I see the MOTRRs go out per schedule, and lots of incoming route records, but can't seem to spot any packets on the air with routes attached.

Next thing to try is not setting enable_route_discovery to see if the ncp use the on board source route cache, and trying SetSourceRoute with a minimal one item cache.

closed 2019-12-20 by stale[bot]