zsmartsystems/com.zsmartsystems.zigbee #767: Ember: source routes not saved and sent to NCP

sesuncedu, 2019-09-28:
High-RAM concentrator mode is not handled properly on the host side. 

Incoming source routes are not saved, and are not provided to NCP via SetSourceRoute.

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2012/06/29/how_does_source_rout-CD8I

>Route Storage/Management
If EZSP_CONFIG_SOURCE_ROUTE_TABLE_SIZE is configured as non-zero size (call the arbitrary size X ) for the NCP (via SetConfigurationValue 0×1A), the NCP will use a FIFO queue to buffer the last X source routes (seen in incoming Route Record packets) in its internal source route table. Regardless of the value of X (even if X=0), there is also always a currentSourceRoute buffer in the NCP that is used for the host to manually set/override the NCP's source route to the next destination. This buffer is managed by the host with SetSourceRoute EZSP command (frame ID 0×5A). CurrentSourceRoute can only be set by the host, but it is always cleared by the NCP after each Unicast or ACK has been sent.
>
>The notion is that the host would store some number of source routes (call this quantity Y ) in its own table, via Ember’s utility code from app/util/source-route-host.c (and source-route-common.c). For a High-RAM concentrator, Y is assumed to be large enough to accommodate all possible source routes from the concentrator (call that quantity T ), or at least all the source routes the concentrator would ever need to use (for ACKs and Unicasts). For a Low-RAM concentrator, Y is assumed to be less than T but greater than 0. (Technically, since the NCP can store a few table entries internally if configured to do so, the host table size Y could be 0, and the NCP would handle the cache of source routes for a Low-RAM concentrator.)

cdjackson, 2019-09-29:
Clearly this is a conditional requirement and not something all systems want to use. This will significant'y increase communications with the NCP which can be a problem so this needs some thought.

sesuncedu, 2019-09-29:
An alternative might be to disable High-ram, since the backend support isn't coded yet?

cdjackson, 2019-09-29:
Agreed. I don’t think that’s done in the library is it? I assume you’re therefore talking about openHAB?

Feel free to propose something if you can think of a nice way to implement this.

sesuncedu, 2019-09-29:
I'm not 💯% sure (I'm not 20% sure), but I think that overall ncp traffic should be a wash. Incoming traffic should go down because of fewer route records coming in, but outgoing gets a bump. Bandwidth feels like it should be fine - I think the SetSourceRoute message is pretty small compared to outgoing unicasts. Not sure about whether window usage will cause backpressure.

cdjackson, 2019-09-29:
I'm also not 100% sure, but what I do know is that some customers using the SPI system (which is slower as it requires polling) have problems with the throughput so I'm a little hesitant to make this "worse".

In any case, I agree that the high-ram concentrator isn't right at the moment, so as long as a) we fix that, and b) make it configurable, then I think it is fine.

sesuncedu, 2019-09-29:
Concentrator config setting performed  in ZigBeeDongleEzsp::setConcentrator, which is called from updateTransportConfig.
High-ram could be downgraded, or rejected.

Can't the SPI be interrupt driven and clocked at 5MHz?

 I had been assuming that the config I have ( 56kbps/XON-XOFF) would be the worst case?

cdjackson, 2019-09-29:
The SPI can’t be changed - it’s embedded in a commercial system and the interrupt isn’t available (which is the root of the problem).

I’m not sure what you mean by :high-ram could be downgraded”? Do you mean remove it? Or do you mean change the openHAB configuration to use low-RAM? If the second (use low RAM) yes, we could do that in OH, but the issue still remains that for this repository, and improvement would be welcome.

cdjackson, 2019-09-29:
Is BDB an official acronym? I’ve not heard it before - to me, BDB is simply Base Device Behaviour.

ZigBee Alliance test specs provide information on what is tested if you refer to them.

closed 2019-12-13 by stale[bot]