zsmartsystems/com.zsmartsystems.zigbee #768: Capabilities from DeviceAnnounce should be saved when discovering/rediscovering nodes

sesuncedu, 2019-09-29:
It shouldn't differ from the Node Descriptor. Looking at you, Samsung/Samjin...

It look like the power and node descriptors for the current smartthings  motion detectors  have pre-compiled descriptors that weren't edited from defaults, but the capabilities in the device announce match what is going on at the mac layer.

This means that czz doesn't see them as sleepy, which messes up java level timeouts.  I'm not 100% sure about this. I'm not sure if this also messes up Ember APS timeouts;  there's a mechanism for telling ezsp that specific addresses should be treated as sleepy for timeout purposes. 

There is actually a pretty good heuristic that happens to work well in this case;  after the node descriptor has been received, and the endpoint info is fetched, the lists of clusters are available. 
 poll control in the inClusters  pretty much guarantees sleepy, and this can be figured out when sleepy devices should be in fast poll mode. 

BDB == Broken Device Behaviour 

I am so glad that  the zigbee alliance doesn't strip email addresses from the certification devices.

cdjackson, 2019-09-29:
I guess this is a tough one as it's device specific. In the case of these devices, you've found they differ, and you've worked out what one is correct. However for other devices, they may differ, and we may not know what one is correct, and it might be the descriptor. I'm not really sure how we should handle these sort of discrepancies.

We just need to be careful here that if we fix a broken device we're not breaking devices that have worked previously.

> BDB == Broken Device Behaviour

No - it's Base Device Behaviour (although I'm not sure why you added this as it isn't referenced in your text).

sesuncedu, 2019-09-29:
Base Device Behavior specifies what happens  for devices that work correctly.
Broken Device Behavior is for everything else.

The most reliable source for sleepy info that is available at more than one hop is probably in the neighbor table in an lqi_rsp ; that should be dynamic, and sourced from something that has to be correct for things to work. It should also be part of the certified platform that is not supposed to be messed with. 

It would be nice to know what certification actually test (though there are some device certification packs that leak some of the detailed test protocols).

sesuncedu, 2019-09-29:
Maybe a simple extension api that could post-process discovered nodes, keyed on manufacturer and model?

cdjackson, 2019-09-29:
What would you actually want to do with this exactly? “Simple” extensions still require work, and there still needs to be a definition, although in general this could be possible now if you want to write your own discovery handler. 

Generally I’m open to ideas, but please clearly specify what it needs to do before you start work so that there’s no surprises.

sesuncedu, 2019-09-29:
Separately, it does seem that device announce is a good place to get this information from:
  EmberZnet uses it for its own extended timeout decision. 

>The stack will automatically extend the timeout:- For our own sleepy children.- When an address response is received from a parent on behalf of its child.- When an indirect  transaction expiry route error is received.- **When an end device announcement is received from a sleepy node.**

BTW, does turning on the APS retry option  obviate the need to use the library level retransmission in use and vice versa?

cdjackson, 2019-09-29:
> BTW, does turning on the APS retry option obviate the need to use the library level retransmission in use and vice versa?

No - it's recommended to use a higher level of retransmission. Each level (MAC/NWK/APS) has mechanisms to cope with errors of different types.

If the APS retry works, then clearly higher level retry will not be used.

sesuncedu, 2019-10-07:
Closing this issue to open two more specific ones

closed 2019-10-07 by sesuncedu