zsmartsystems/com.zsmartsystems.zigbee #773: Heisenbug in test case introduced for #770

sesuncedu, 2019-10-02:
The final test for invocation count for sendTransaction in the test for ```handleCommand``` generally fails.

Mockito sometimes reports a count of 6 instead of 5. Stepping through the relevant code makes the problem go away, as does adding println. 

Replacing the mocked listener with a lambda does  not change the behavior  (unless it prints). 

This seems to suggest some weird timing or concurrency related mockito issue.

The problem has been observed with oracle jdk 8 and 11, under Linux on amd64 and arm, from maven and under intellij, and in the latter case, when the failing test is the only test invoked.

Workaround may be to isolate the last test in a separate test method.

I will try with mockito verboselogging turned on (though that may count as observing for heisenbug purposes) .

sesuncedu, 2019-10-02:
The root  cause seems to be in ```ZclCommand::notifyListeners```.  

This method seems rather odd:
1. If there are n listeners, of which m return true, the list will be decremented m*n times
2. If all listeners take too long to complete, the method will return false. 
2. If any listener tasks are executing or queued, they will still be dispatched / run to completion.
3. More than one listener can  handle the command. more than one listener will handle the command. Each listener might send a response.

The simplest fix would be to require all listeners to complete without blocking for an extended time, and execute them sequentially on the main thread.  If a listener wishes to handle the command, it can submit a task to an executor service. 

A more complicated approach could use  ExecutorCompletionService;  the listener would need to be split into decision and handling parts, and if the number of execution threads is bounded it is still possible for slow listeners to cause problems even if a queued  listener would be fast. 

The simple approach seems reasonable here.

cdjackson, 2019-10-02:
What is the actual root cause though?

It is not suitable to make everything run on the main thread - this will cause significant issues. I don’t think there are any issues with this as it currently stands other than the test, and I don’t understand what you are suggesting is actually the issue with the current implementation?

cdjackson, 2019-10-02:
> The root cause seems to be in ZclCommand::notifyListeners.

Can you also clarify where you think the error is since the ```ZclCommand``` class doesn't have such a method. I assume you probably are referring to ```ZclCluster::notifyCommandListener```?

cdjackson, 2019-10-02:
Can you also clarify what test fails - or provide a log showing the failure. I can't replicate it here and it's pretty unclear from your report where the test failure is.

sesuncedu, 2019-10-03:
The failing test is 
```ZclClusterTest::handleCommand```

The test failures are generated from the verify in: 
```java
        Mockito.when(listenerMock.commandReceived(command)).thenReturn(true);
        cluster.handleCommand(command);
        assertEquals(command, zclCommandCapture.getValue());
        Mockito.verify(endpoint, 
             Mockito.timeout(TIMEOUT).times(5)).sendTransaction(commandCapture.capture());
```

cdjackson, 2019-10-03:
> The test failures are generated from the verify

And what is the error you get and is it consistent?

sesuncedu, 2019-10-03:
Just getting to that part :)

sesuncedu, 2019-10-03:
With unmodified  code, performing 10- runs of ```mvn -X test``` passed 3 times and failed 7. 

The failure was:
```
Tests run: 16, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 5.17 sec <<< FAILURE!
handleCommand(com.zsmartsystems.zigbee.zcl.ZclClusterTest)  Time elapsed: 5.073 sec  <<< FAILURE!
org.mockito.exceptions.verification.TooManyActualInvocations: 
zigBeeEndpoint.sendTransaction(
    <Capturing argument>
);
Wanted 5 times:
-> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.handleCommand(ZclClusterTest.java:388)
But was 6 times:
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
```
The test case was then modified to allow for mockito logging. See following message.

sesuncedu, 2019-10-03:
The test case was modified as per https://github.com/sesuncedu/com.zsmartsystems.zigbee/commit/13c3eda56e0b89379b6d3bf54c26ddafee9511cf

The individual test method was run from intellij until a failure occurred, and the output recorded.  The complete output is at the end of this comment. 

### Problematic part of test output

#### 1. Mocked ```commandReceived``` returns true
```
############ Logging method invocation #37 on mock/spy ########
   stubbed: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.handleCommand(ZclClusterTest.java:392)
zclCommandListener.commandReceived(
    ConfigureReportingResponse [Basic: null -> 0/0, cluster=0000, TID=80, status=null, records=null]
);
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclCluster$5.run(ZclCluster.java:1131)
   has returned: "true" (java.lang.Boolean)
```
#### 2. ```sendTransaction``` is still invoked
```
############ Logging method invocation #38 on mock/spy ########
zigBeeEndpoint.sendTransaction(
    DefaultResponse [Basic: null -> 0/0, cluster=0000, TID=80, commandIdentifier=7, statusCode=SUCCESS]
);
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
   has returned: "null"
```
#### 3. Inducing test failure 
```
############ Logging method invocation #39 on mock/spy ########
zigBeeEndpoint.sendTransaction(null);
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.handleCommand(ZclClusterTest.java:396)
   has thrown: class org.mockito.exceptions.verification.TooManyActualInvocations with message
zigBeeEndpoint.sendTransaction(
    <Capturing argument>
);
Wanted 5 times:
-> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.handleCommand(ZclClusterTest.java:396)
But was 6 times:
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)
-> at com.zsmartsystems.zigbee.zcl.ZclCluster.sendDefaultResponse(ZclCluster.java:705)

```

#### Complete test output
```
############ Logging method invocation #1 on mock/spy ########
zigBeeEndpoint.getEndpointId();
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.createEndpoint(ZclClusterTest.java:78)
   has returned: "0" (java.lang.Integer)

############ Logging method invocation #2 on mock/spy ########
zigBeeEndpoint.getEndpointAddress();
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.createEndpoint(ZclClusterTest.java:79)
   has returned: "null"

############ Logging method invocation #3 on mock/spy ########
zigBeeEndpoint.getParentNode();
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.createEndpoint(ZclClusterTest.java:84)
   has returned: "null"

############ Logging method invocation #4 on mock/spy ########
zigBeeEndpoint.sendTransaction(null, null);
   invoked: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.createEndpoint(ZclClusterTest.java:85)
   has returned: "null"

############ Logging method invocation #5 on mock/spy ########
   stubbed: -> at com.zsmartsystems.zigbee.zcl.ZclClusterTest.createEndpoint(ZclClusterTest.java:79)
[... 252 lines cut ...]
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at com.intellij.junit4.JUnit4IdeaTestRunner.startRunnerWithArgs(JUnit4IdeaTestRunner.java:68)
	at com.intellij.rt.execution.junit.IdeaTestRunner$Repeater.startRunnerWithArgs(IdeaTestRunner.java:47)
	at com.intellij.rt.execution.junit.JUnitStarter.prepareStreamsAndStart(JUnitStarter.java:242)
	at com.intellij.rt.execution.junit.JUnitStarter.main(JUnitStarter.java:70)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.intellij.rt.execution.application.AppMainV2.main(AppMainV2.java:131)

```

sesuncedu, 2019-10-03:
sendTransaction still being invoked despite the listeners commandReceived returning true indicates that ZclCluster::notifyCommandListener is failing to pass on the result of the notification and is returning false, since otherwise ZclCluster::handleCommand would return without sending a default response.

sesuncedu, 2019-10-03:
Examining the code in notifyCommandListener, it can be seen that this can happen if an InterruptedException is thrown OR if the latch counter times out before the response has been set, and the response is not set before the result is returned. 

The concurrency related code in notifyCommandListener has a number of issues, most noted above. Combined with the non-deterministic nature of the failures, the first step in investigation was to switch to a serial mode of execution.   This caused the tests to consistently pass.  

More detailed work could be performed to isolate the precise error in concurrency control that induced the test failure. However, even if the code works, it cannot behave correctly if more than one listener attempts to handle the command in ways that might have side effects.

sesuncedu, 2019-10-03:
One possible cause of the failure is the latch count being decremented before the response is set (the response is also set once for every listener.

sesuncedu, 2019-10-03:
See PR

cdjackson, 2019-10-03:
What do you mean by this -:

> even if the code works, it cannot behave correctly if more than one listener attempts to handle the command in ways that might have side effects.

When I wrote this I first thought about having a sendReply method to try and correlate the responses, but this introduced problems. I don't really see a problem with multiple listeners responding - ideally it should not happen, and clearly it would result in multiple responses which is also not good, but this code I think should still work correctly.

sesuncedu, 2019-10-03:
It is an error if multiple listeners generate responses. The current approach requires that at most one  listener respond to a given message, and this must be handled by private arrangement between all listeners associated with a cluster.  

This seems awkward.  Anyway, a fix for two  race conditions, one of which was the cause of the immediate problem, is in #780.

cdjackson, 2019-10-03:
If multiple listeners generate responses, then yes, there will be multiple responses. But that's not really a concurrency issue - the threads etc will still work fine I think? Or am I missing something?

sesuncedu, 2019-10-03:
Its an issue that could be introduced in serial form if listeners were processed in sequence, with the final result code updated in a loop, and no short-circuiting, but in that situation the library would at least be able to detect that something was going on.  In a concurrent situation there is no way to handle it other than every listener applied to a given cluster being written to adopt some arbitrary convention.  

The timeout that will return false if no listener completes within one second, which causes things to fall through to the default handler, but leaves any unfinished listener tasks to run to completion, could be fixed, but would require some pain.

cdjackson, 2019-10-03:
> Its an issue that could be introduced in serial form if listeners were processed in sequence

I think this is a bad idea - sorry. It leaves the whole system open to a poorly written application that is delayed.

> with the final result code updated in a loop

Are you referring to the result code that is returned in the DefaultResponse? You can't assume that a handler is going to return a DefaultResponse - other responses do exist.

> other than every listener applied to a given cluster being written to adopt some arbitrary convention.

?? I don't really know what you mean by this.

As I've said elsewhere, I don't think that multiple listeners ought to be sending responses in most cases. If that does happen, then I suspect that there are other problems in the application (eg two handlers are handling the OTA responses, or any other response, then that's going to result in problems anyway).

sesuncedu, 2019-10-03:
Clarifying one part:

Suppose there's only a single listener.
If the listener accepts a command (and thus will return true), and  takes more than one second to process a command and generate a response, the timeout will fire, and false will be returned, causing the DefaultResponse code to be activated. 
Meanwhile, the listener is still running and could still generate a response, even though the library has already generated a DefaultResponse for it.

cdjackson, 2019-10-03:
Yep - well we can’t wait too long. Maybe 1 second isn’t the best answer, but devices SHOULD respond quickly or there will be other problems due to retrying of commands. If a command takes 2 seconds to respond, and a sending device resends, then it takes 4 seconds and that’s bad news. Listeners should really respond in a reasonable timeframe.

Do you suggest a different timeout? If so, what is it and why?

cdjackson, 2019-10-04:
I wanted to cover a few of your points and explain the design... I will try and consolidate things here since we are discussing this in two places now.

**Basic Principals**

All listeners should have equal access to incoming messages. Somewhere you indicated that by sequentially calling listeners the caller could perform some sort of arbitration on the results to reduce the number of responses, however that would mean that there is a priority - it could be first listener to register, first to be called, or first to respond that wins, but somewhere there is a priority that is not controlled.

If we were to think of the application as a node on the ZigBee network, there should probably just be one entity receiving incoming messages, and therefore only one entity sending responses. There should not normally be multiple entities receiving an incoming message and sending a response, or there is likely some sort of application conflict. Note that I am specifically talking about incoming messages here - outgoing transactions are handled differently.

This code is really here to provide a ```DefaultResponse``` in a timely manner if a received command is not responded to - it's not there to manage or arbitrate against multiple responders since that probably already means there's an issue in the application design.

**Concurrent versus Sequential notification**

You indicate that you could perform some sort of arbitration if the notifications were sent sequentially and not concurrently. To me, it makes no difference - in both cases you can send the notification and receive the reply. The concurrent implementation I've used does exactly this - it receives a response from the notification to say if it responded - exactly the same as if we'd used a sequential approach.

The advantage (IMHO) of the concurrent approach is that all listeners are not blocked by one poor process and all get equal chance to respond. As I've said earlier, there should really not be multiple responders in any case.

**Boolean Return Value**

I think you indicated that the listeners could respond with some other response to allow the caller to perform some sort of arbitration. This was something I initially looked at, but as the response may be something other than the ```DefaultResponse```, this becomes difficult. I did at one stage consider that the listener could return the actual command it wants to send, but again that has problems in some circumstances where (for example) a delay may be required. One example of this is in the OTA where there are delays between some responses.

The best approach in the end seemed to just ask listeners to respond quickly, and allow them to simply say if they have, or will send a response.

**Listener Removal**

You mentioned that if a listener is removed after the notification has been sent, then it will not complete and the timeout is required to catch this. I don't agree. If a listener is called, and then the listener is removed, it will still complete. The removal of a listener simply means it will not be called again when subsequent commands are received - it will still complete the executing call, so the concurrent implementation will still wait for the thread to complete.

I hope that helps you to understand some of the background behind the design and implementation of this. As is often the case, there are compromises in here, and it makes assumptions that may not always be valid for all implementations. I do try to document my assumptions both in the code and the project documentation where possible, however it's not always as good as it could be I'm sure :) .

I'm also sure there are some corner cases where this may not work as well as it should, so please feel free to propose improvements (both in the code but also in the inline documentation as necessary).

cdjackson, 2019-10-04:
Ok, you are right that there is a VERY small opportunity - 1 line of code between the latch being created and the listeners being notified. Yes, we should resolve this, but it's not particularly likely to occur at the moment. It does not require a snapshot of the list necessarily - it could simply be synchronised.

cdjackson, 2019-10-04:
Closed by #780

closed 2019-10-04 by cdjackson

sesuncedu, 2019-10-04:
Right- the bug that was actually causing the test failure was the latch
decrement before response update, so that's what got properly fixed.

The concurrent modification race condition is really unlikely to happen. I
really only  noticed it because I was looking at the concurrency code in
the method.

Using toArray to snapshot the listener set is simple.
It should  be inexpensive for the size of sets used with COW.  The new
array will almost always be allocated from the TLAB, so it is likely to
preserve non-blocking.

Using a lock of some kind to guard the add/remove listener methods and the
getting of size & creation of iterator block is enough for safety.

There are faster solutions,but they are not worth the effort (need access
to the length of the snapshot in the iterator).