zsmartsystems/com.zsmartsystems.zigbee #776: Disable Default Response flag should be set in outgoing responses to received commands

sesuncedu, 2019-10-02:
§ 2.4.1.1.4 
> This field SHALL be set to 1, for all response frames generated as the immediate and direct effect of a previously received frame. Still required to generate a response on receipt unless it's a DefaultResponse, but recipient is allowed to resent it.

A sensible place to do this might be a new sendReply method on ZclCluster, which could also take care of copying transaction ids etc.

cdjackson, 2019-10-02:
Please feel free to propose a PR.  I’m not against the sendReply idea - this is something I looked at before, but I have other things to work on so it was not implemented at this stage.

sesuncedu, 2019-10-02:
Feel free to assign. 

There's a related issue of being able to send commands with DefaultResponse disabled, and also with or without aps acks required.  The former makes sense for commands where success doesn't tell us much useful, and errors would be generalizable beyond the individual performance of the command (e.g. If an acked OnCommand fails and generates a DefaultResponse, it probably affects all on commands for this cluster instance, and if OnOff is being reported, the attribute report is more useful).

Unacked would be suitable for... default responses to responses that shouldn't be asking for them anyway. Ideally they would be sent with an incorrect source route.

cdjackson, 2019-10-02:
Let's not mix issues - if there's something different, then please open a separate issue. The lack of default response, or disabling APS acks is more problematic I think as it impacts more generally on the framework.

closed 2019-10-07 by cdjackson

sesuncedu, 2019-10-07:
[To me: This issue should be split] 
There seem to be a few variants of sendReply that would be useful. 

1: Terminal responses, where no response is expected, and where any errors generated could be handled by the cluster instance. 

2: Terminal responses where no response is expected in the usual case, but where error responses are specified, and transaction specific processing may be desirable.
I believe that this situation is one that is problematic for the transaction manager.

An example of a response message where the general requirement to set the disableDefaultResponse bit is explicitly reiterated, and an error response to a response specified is ```QueryNextImageResponse```. Per ZCL 11.13.5.5:

>[The]  command SHALL have the disable default response bit set. [... ]However, the default  response SHALL be generated to indicate the error cases below. 
**11.13.5.5.1 Malformed Command**
Upon reception a badly formatted [...] command, for example, the command is missing one of the payload field, [...] the image type value included in the command does not match that of the device [...] ; the client [...] SHALL send default response command with MALFORMED_COMMAND status to the server.

This particular case could be handled by a handler at the cluster level. For the stock OTA server, the only sensible response to this message is to stop trying to serve the provided file and report an error. Thus a one-way send reply method could be used, with no need for any transaction matching.

So net-net, a one-way sendReply method on ZclCluster is useful (also, making ```send``` public rather than protected might be useful too).

I'm working on a PR to fix the incorrect use of DefaultResponses for image queries in the OTA server, which will include a one way sendReply method. That will be a test of one approach to fixing this ticket, but is probably not enough for closure. 

#### Send methods that return a result should have DDR clear. 
This can either be handled by clearing the bit, or by signaling an error. 

#### Handling sends of messages with Disable DefaultResponse set. 

A one-way send method with an error handler callback would be nice, but needs some more extensive changes at the transaction manager. 

One-way sends with an error callback would follow the usual path up until the send is acknowledged. The transaction would then be retired from any queues, and moved to a separate collection of transactions with live error callbacks, with a timeout set. 

If an error is received, the transaction is removed from the collection, and the callback invoked, with the original message and the subsequent error response provided as arguments. 
If the timeout fires, the callback is removed from the set, and processing terminated. It would also be possible to invoke the callback with the error response absent.

In both cases, the transaction can be marked as completed.

Since possible pending error messages should not prevent the dispatch of new transactions,  there may be issues with available TIDs if many commands are sent and no errors occur (and if the timeout is too long). 
For fresh incoming requests, this can be addressed by only matching DefaultResponse messages to error handlers. For outgoing messages this can be handled by canceling any error handler with whose TID matches. This could be treated the same as no error being received.

sesuncedu, 2019-10-07:
Gah

PR #784 (2019-10-07): Enhance ZclCluster to support local attributes
This PR adds methods to support local attributes. This means the local attributes are the "opposite side" of the remote - so if the remote cluster is a server, then this will provide the client attributes.

The new methods are ```getLocalAttribute()```, ```getLocalAttributes()```.  In addition, the ```ZclAttribute``` class has been enhanced with a method ```setValue()``` to allow the value to be explicitly set.

The ```implemented``` flag in the ```ZclAttribute``` class is also now implemented and this is used when responding to attribute requests [... cut]