zsmartsystems/com.zsmartsystems.zigbee #791: Ember: Race conditions assigning sequence numbers in  EzspFrameRequest

sesuncedu, 2019-10-10:
EZSP frame sequence numbers are integers in the range [1,254].  

EzspFrameRequest  stores the value of the next sequence number to assign  in an ```AtomicLong```.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/d7398dd5edcfb5f78a3e367a82146ea7d73d95a2/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ezsp/EzspFrameRequest.java#L43

Sequence numbers are assigned in the constructor

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/d7398dd5edcfb5f78a3e367a82146ea7d73d95a2/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ezsp/EzspFrameRequest.java#L48-L53

This code is subject can return invalid values due to the non-atomic manipulation of the sequence when the value overflows.  For example, if a thread is assigned sequence number 254, but other threads increment the value before the first thread has had a chance to reset it to 1, the those threads can  be assigned sequence numbers of 255 or higher.  

The fix is to get and update the sequence in a single atomic operation. 
e.g.
```java
    private final static AtomicInteger sequence = new AtomicInteger(1);

    /**
     * Constructor used to create an outgoing frame
     */
    protected EzspFrameRequest() {
        sequenceNumber = sequence.getAndUpdate(n -> n <254 ? n+1 : 1);
    }
```

I need to stash some stuff so I can create a  PR with just this change.

cdjackson, 2019-10-11:
Thanks. As commented on the PR, I’d suggest to just synchronise the existing code. We can probably change to using an AtomicInteger - I think in the past I needed to keep track of the total number of frames sent which is why it is done the way it is, but that’s not used now, but we can’t use streaming.

sesuncedu, 2019-10-11:
Using getAndUpdate is faster and less typing, so what's not to like :) 

[Being able to use streaming again would be nice  if setSourceRoute is used.]

cdjackson, 2019-10-11:
Unfortunately streaming is not possible due to Android limitations.

sesuncedu, 2019-10-11:
I'm confused about android. Is this an issue with android usbserial drivers? 
AtomicInteger::getAndUpdate was added to Android in API 24 (Nougat).

sesuncedu, 2019-10-11:
https://github.com/felHR85/UsbSerial is aysnc and supports flow control for most of the relevant devices

cdjackson, 2019-10-11:
I need to support older Android, and the streaming API is not available. This can't currently be changed - sorry.

I'm not sure that I understand the reference to the serial driver for Android?

sesuncedu, 2019-10-12:
I was confused by your reference to streaming. getAndUpdate doesn't use streams. It does take an argument that is an instance of IntUnaryOperator, which in the  code above is a lambda. Lambda is available for old android because it is desugared. getAndUpdate isn't until 24.

 I thought you were talking about streaming ezsp+ash from android; some older usbserial code for Android didn't do flow control; the library I linked to does (for the most popular usb-uart bridges). 

If you have specific android api constraints it might be setting up animal sniffer?

cdjackson, 2019-10-12:
Yes, I know - I think I said that somewhere earlier that I misread the code (don’t try and review code on your phone ;) ).

I’ve not heard of “animal sniffer” before (I’’m not really an Android user), but yes, this does seem a good idea to catch issues - thanks.

sesuncedu, 2019-10-12:
Animalsniffer is maven ; it checks for api usage. Obviously it's not a an android thing because it isn't gradle. Hmm- say... Also there would be kotlin... Hmm-say 🤔

cdjackson, 2019-10-25:
@sesuncedu is this still relevant since you have closed the PR?

cdjackson, 2019-10-26:
@sesuncedu do we now close this issue or is it still something you are looking at? You've closed the PR, so I'm assuming it's now not an issue?

cdjackson, 2019-11-03:
@sesuncedu please can you comment on the status of this? You raised the issue, and opened a PR that you have now closed - can you update us on what is happening and why you closed the PR please?

cdjackson, 2019-11-07:
@sesuncedu where are we with this and the PR you closed?

PR #953 (2020-01-10): Add support for EZSP V8
This implements support for EZSP v8 frame format and improves the allocation of sequence numbers to Ember frames.

Closes #791 
Closes #944
closed 2020-01-10 by cdjackson