zsmartsystems/com.zsmartsystems.zigbee #794: Add a method to support changing the network channel

sesuncedu, 2019-10-12:
There's a method that updates the config, but not one that does the update with magic scan time so that all the rx_on devices will follow.

sesuncedu, 2019-10-12:
1. Should a coordinator assume that it is the network manager unless it affirmatively knows different?
2. If we are the network manager, and we are online, should we enforce the frequency agility rate rules?
3. If we will be the network manager but are not online, but we have been previously, when we are activated, should we connect on the old channel, perform frequency agility, then switch to the new one?
4. Should there be an exposed method for changing the channel immediately?
5. Should it be possible to delegate the process to the ncp if the ncp supports it?

sesuncedu, 2019-10-12:
1. Should it be possible to specify a mask instead of a specific channel in places where a channel is needed and where selection is sensible?
2. Should these be saved to the local apsChannelMask?
3. Should a network manager attempt to propagate this to the entire network?
4. How many different layers are violated by recording energy scans in the AIB?

cdjackson, 2019-10-12:
Thanks @sesuncedu this is an area I keep thinking about looking at, but have not had the time/priority, so it would really be great if you can take a look. I did start to look at the channel management a while back and can push this branch if you like.

I would quite like to expose a higher level interface for performing scans, setting channels and channel masks etc. Currently where I need to do this, it has always been on the Ember NCP, and I always deal directly with the ```EmberNcp``` class which is arguably a bit low level.

Just something else to think about as part of this, even if it's not directly related to channels. If you are looking at scanning, please also consider active scans as well as the energy scans as part of any initialisation.

> Should a coordinator assume that it is the network manager

It should know shouldn't it? This is defined in one of the descriptors (```NodeDescriptor``` if I remember correctly.

> should we enforce the frequency agility

It should be optional, but if enabled, why not...

> Should there be an exposed method for changing the channel immediately?

There is already a ```setChannel``` method so I think it should do this.

> Should it be possible to delegate the process to the ncp if the ncp supports it?

Probably. I think that the Ember NCP can do all this.

> Should it be possible to specify a mask instead of a specific channel

Yes.

> Should a network manager attempt to propagate this to the entire network?

Yes.
> How many different layers are violated by recording energy scans in the AIB?

What is the AIB?

cdjackson, 2019-10-12:
For reference, this was my starter on this topic. Note that some of this was updating the ZDO messages which will now need a re-think as the autocoder has changed since...

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/network_channel_manager

sesuncedu, 2019-10-12:
AIB is the APS Information Base.

sesuncedu, 2019-10-22:
Note to self: It is a good idea to make sure that the destination address is always set when requesting a remote energy scan.  SiLabs stacks at least do not handle this error gracefully (Jasco switches (HA) lock up. Innr branded Leedarson light bulbs (zigbee 3) generate a new response every second or so.) 

"Have you tried turning it off and on again?"
 "It's a light switch. That's the problem." 

Nothing that can't be fixed by cycling power at the breaker...

sesuncedu, 2019-10-24:
Progress note:
#### Request generation
I'm using private zdo subclasses to marshall the three different forms of management network update request. As @cdjackson notes above, the existing request class doesn't generate valid PDUs since each variant uses a different set of fields. 

Remote scan requests work.

Changing the Network Manager has not been tested in case it worked. 

Channel change request broadcast work, with a temporary hack for persisting data in place. Changing the radio channel on the ncp immediately is not done. 

Setting the channel on the network manager then calling reinitialize did not go through cleanly when  I tried it, probably due to where I had inserted the change procedure for testing (I was trying to find a statewhere I could send and messages but with no other  tasks active, but ended up losing  the endpoint on the ncp).

There needs to be an api for pausing all transaction queues other than those of a certain priority. This is needed to stop messages bring sent to listeners who are changing channels, and also to those who may be performing an energy scan and thus non responsive. (There's a separate but related issue involving backpressure on broadcast queue when non ZSS-generated  broadcast transactions may be occupying all available slots.)

closed 2020-01-06 by stale[bot]