zsmartsystems/com.zsmartsystems.zigbee #810: Ember & TransactionManager : synchronizing maximum number of outstanding transactions.

sesuncedu, 2019-10-17:
Currently this information is configured separately. If the number of unacked messages is changed in the ncp configuration/initialization , this is not passed back to the the transaction manager. Thus the transaction manager will not be able to take advantage of any extra slots.

The transaction manager is created prior to the ncp initialization, and since it is not in general possible to determine whether the desired configuration value is usable until an attempt is made to set it, the value in the transaction manager needs to be updated to the one successfully configured.

I have added this feature experimentally to my extended timeout branch, which has access to the NetworkManager; however since the transaction manager is split out from the network manager the network manager either needs a method that exposes this functionality, or a method that exposes the transaction manager.

This is a bit icky, because ZigBeeNetworkManager provides both API and implementation, so any new  method becomes part of the API.

An alternative approach would be to add methods to the various NCP drivers that would allow actual configured values to be retrieved. This could potentially require changes to all drivers, though if the return type is made an Optional, a default method returning an empty is plausible. Note that because of the requirement to support android 4.4, Optional would have to be the guava one, not java (are there many developers using CZZ for deployment on Android? Do they have current needs to support pre- api 24?)

cdjackson, 2019-10-17:
Isn’t it simple enough for the software architect to configure these parameters? If they change the number of outstanding messages in the NCP, then they can also set the number of transactions. Sure - synchronising this is nice, but here you add a lot of complexity simply to avoid setting two configuration options instead of one don’t you?

sesuncedu, 2019-10-17:
The issue is that the actual number of  available maximum transactions may not be known  until configuration has been performed (e.g. if the setting is not compatible with other settings).

cdjackson, 2019-10-17:
Isn’t it set explicitly? Are you thinking about the maximum number of buffers, or the maximum number of outstanding transactions - there are two configurations if I remember correctly. The total number of buffers can be configured to be set based on the amount of available memory by setting EZSP_CONFIG_PACKET_BUFFER_COUNT to FF, and therefore may not be known until runtime, but the maximum number of outstanding APS transactions is set explicitly with EZSP_CONFIG_APS_UNICAST_MESSAGE_COUNT.

Can you explicitly state what configuration you are talking about please?

sesuncedu, 2019-10-17:
EZSP_CONFIG_APS_UNICAST_MESSAGE_COUNT is the setting.  Not all values may be permitted by all builds in all situations. 

An in-flight transaction will occupy a minimum of two buffers, but I believe that those are borrowed from the pool on demand.  I haven't messed with the use-whats-left setting for  buffers (0xff).

cdjackson, 2019-10-17:
My understanding is that this is not related to buffers - this is the number of messages that may be outstanding at once. This is explicitly configured so I don’t think there’s a need to link this directly to transactions since the application architect can already do this.

sesuncedu, 2019-10-17:
It isn't related to buffers. 

An explicit value is *requested* when the ncp is initialized, but that value may not be accepted.  That means that the *actual* value  is not known until after the NCP has been configured. 

Each possible unacked transaction uses 6 bytes from reading header files. When an actual unacked transaction is in flight an additional minimum of two buffers are used.

cdjackson, 2019-10-18:
Sorry - can you explain what you mean by "requested"? This is a configuration parameter that is set by the host and I don't think it changes after startup. The number of buffers (ie EZSP_CONFIG_PACKET_BUFFER_COUNT) does change if it is set to FF - if it's set to another number, then I don't think it changes either. I think if you set it to another value it will simply fail.

As far as I know, it is only the EZSP_CONFIG_PACKET_BUFFER_COUNT that changes as that is configured to use all available memory for buffers after all other configuration parameters have allocated their memory.

Maybe you have further information, or you've tried something and found more information by testing? If so, please clarify.

cdjackson, 2019-11-07:
@sesuncedu ping?

closed 2020-01-20 by stale[bot]