zsmartsystems/com.zsmartsystems.zigbee #811: Feature: using exceptions to signal errors.

sesuncedu, 2019-10-17:
Initial motivating case comes from ember.

 Performing a channel change has several steps. One of these steps requires broadcasting an update request. This returns an ezsp status. 

Another requires changing the radio channel. This returns an ember status.

If the method is called at an invalid time, there is a library level error.

If exceptions are used, each case can throw an appropriate exception. A suitable exception hierarchy would allow for appropriate error handling specificity.

One possible use for exceptions would be in providing an api for ZCL operations, especially those that have a specific response type, but which may receive default responses in the case of errors. These default responses could generate exceptions. Non-success default responses in general could be thrown as exceptions, so actions that only have DefaultResponses would return void. 

Exceptions could be subclasses of RuntimeException if checked exceptions are disfavored.

Exceptions thrown from callable methods in futures are wrapped in ExecutionExceptions. These can be unwrapped by the method calling get, and handled or rethrown as desired (some exceptions could be rewapped as zigbee exceptions).

Also, some places where interrupted exceptions are swallowed might want to rethrow them to avoid lost interruptions.

cdjackson, 2019-10-17:
I'm not super keen on wholesale changes like this as this will require changes to applications that could well be problematic for some. If there are specific instances then we can look at it but I don't think there should be a wholesale change in approach.

cdjackson, 2019-10-27:
Closing this as we should not simply go around changing things for the sake of it. If there are specific examples, then for sure they can be looked at on a case by case basis.

closed 2019-10-27 by cdjackson