zsmartsystems/com.zsmartsystems.zigbee #832: ZigbeeNetworkManager sets ZigBeeNetworkState.INITIALISING state twice

triller-telekom, 2019-10-25:
I just stumbled across this piece of code:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c85f91dd32b48f6d6a1863d1f119300a962d1d7c/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L356-L370

I am pretty sure the second setting of the state is a left over from the past and should be removed.

Also I noticed that there is no synchronization when setting the network state within these methods: `reinitialize`, `shutdown`, `startup`. I think this is a bug as well?

cdjackson, 2019-10-25:
> Also I noticed that there is no synchronization when setting the network state within these methods: reinitialize, shutdown, startup. I think this is a bug as well?

I think this is already synchronised? All these methods call ```setNetworkState``` which is synchronised - I don't really think it requires synchronisation inside the calling method as well if that's what you are inferring?

triller-telekom, 2019-10-29:
Oh right, the `setNetworkState` is already synchronized. But then there is a double synchronization within `initialize`: 

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/a12a1d54c9291e3986d2107ae675560db6425b33/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L356-L361

it might be necessary to sync for comparing the current network state within the if-statement. However, the call to `setNetworkState` could be moved to after the synchronize, or am I mistaken there?

cdjackson, 2019-10-29:
> But then there is a double synchronization within initialize:

Does that matter? I don't think so...

The reason is that the ```networkState``` is checked here - we don't want another thread changing ```networkState``` - even though it is synchronised in the ```setNetworkState``` it also needs synchronising here.

> the call to setNetworkState could be moved to after the synchronize, or am I mistaken there?

Well, I think you're mistaken ;)

I don't think it's wise to move the ```setNetworkState``` call to after the synchronized block as that leaves a (small) window where the check has been made, but the update has not been - ie this thread checks that we're != ```ZigBeeNetworkState.UNINITIALISED```, it then unsynchonises leaving another thread open to manipulate the state, and then we would call ```setNetworkState``` - this doesn't seem good to me?

closed 2019-10-31 by triller-telekom