zsmartsystems/com.zsmartsystems.zigbee #837: Responses to outgoing messages addressed to clusters for which explicit support has not been registered appear to be rejected

sesuncedu, 2019-10-26:
The changes in #826 appear to be causing the framework to reject incoming messages that are from clusters that the framework has sent messages to. 

I have created a PR #836 with a little  debugging code to illustrate. 

Maybe #826 should be backed out for now?

### Examples deployed openhab:

#### Read Attributes Command for On/Off
```
17:55:52.572 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 0022A30000252FBF: Polling zigbee:device:536f0a01:0022a30000252fbf:0022A30000252FBF_1_switch
17:55:52.575 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [On/Off: 0000/0 -> 4092/1, cluster=0006, TID=F4, identifiers=[0]]
17:55:52.577 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=244, commandId=0]
17:55:52.579 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=16530/1, profile=0104, cluster=0006, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=24, payload=00 F4 00 00 00]
17:55:52.582 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - TX EZSP: EzspSendUnicastRequest [type=EMBER_OUTGOING_DIRECT, indexOrDestination=4092, apsFrame=EmberApsFrame [profileId=0104, clusterId=0006, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=24], messageTag=F4, messageContents=00 F4 00 00 00]
17:55:52.585 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
17:55:52.588 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=2, ackNum=4, reTx=false, data=24 00 34 00 92 40 04 01 06 00 01 01 40 11 00 00 24 F4 05 00 F4 00 00 00]
17:55:52.603 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=4, ackNum=3, reTx=false, data=24 80 34 00 86]
17:55:52.606 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=2, ackNum=4, reTx=false, data=24 00 34 00 92 40 04 01 06 00 01 01 40 11 00 00 24 F4 05 00 F4 00 00 00]
17:55:52.609 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=86]
17:55:52.611 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=5, notRdy=false]
17:55:52.641 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=5, ackNum=3, reTx=false, data=24 90 59 92 40 BF 2F 25 00 00 A3 22 00 FF C6 00]
17:55:52.644 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [source=4092, sourceEui=0022A30000252FBF, lastHopLqi=255, lastHopRssi=-58, relayList=]
17:55:52.646 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=6, notRdy=false]
17:55:52.663 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=3, reTx=false, data=24 90 59 92 40 BF 2F 25 00 00 A3 22 00 FF C7 00]
17:55:52.666 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [source=4092, sourceEui=0022A30000252FBF, lastHopLqi=255, lastHopRssi=-57, relayList=]
17:55:52.668 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=7, notRdy=false]
17:55:52.692 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=7, ackNum=3, reTx=false, data=24 90 45 00 04 01 06 00 01 01 40 01 00 00 4C FF C6 92 40 FF FF 08 08 F4 01 00 00 00 10 00 02]
17:55:52.695 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0006, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=4C], lastHopLqi=255, lastHopRssi=-58, sender=4092, bindingIndex=255, addressIndex=255, messageContents=08 F4 01 00 00 00 10 00]
17:55:52.698 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=16530/1, destinationAddress=0/1, profile=0104, cluster=0006, addressMode=DEVICE, radius=0, apsSecurity=false, apsCounter=4C, payload=08 F4 01 00 00 00 10 00]
17:55:52.701 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=244, commandId=1]
17:55:52.703 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - Unsupported local client cluster 0006
17:55:52.704 [ERROR] [rtsystems.zigbee.ZigBeeNetworkManager] - Rejecting incoming message for cluster 0x0006, but we have sent outgoing non-default-response to it
17:55:52.706 [INFO ] [rtsystems.zigbee.ZigBeeNetworkManager] - rejected command was ReadAttributesResponse [On/Off: null -> null, cluster=0006, TID=F4, records=[ReadAttributeStatusRecord [status=SUCCESS, attributeIdentifier=0, attributeDataType=BOOLEAN, attributeValue=false]]]
17:55:52.709 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: DefaultResponse [On/Off: 0000/0 -> 4092/1, cluster=0006, TID=F4, commandIdentifier=1, statusCode=FAILURE]
17:55:52.711 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=244, commandId=11]
17:55:52.713 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=16530/1, profile=0104, cluster=0006, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=25, payload=10 F4 0B 01 01]
17:55:52.716 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - Incoming message from node 4092 did not translate to command
```

#### Report Attribute Command for temperature measurement
```
17:48:39.875 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0402, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=DF], lastHopLqi=255, lastHopRssi=-66, sender=0AC8, bindingIndex=255, addressIndex=255, messageContents=08 13 0A 00 00 29 FC 08]
17:48:39.881 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=2760/1, destinationAddress=0/1, profile=0104, cluster=0402, addressMode=DEVICE, radius=0, apsSecurity=false, apsCounter=DF, payload=08 13 0A 00 00 29 FC 08]
17:48:39.886 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=19, commandId=10]
17:48:39.891 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - Unsupported local client cluster 0402
17:48:39.896 [ERROR] [rtsystems.zigbee.ZigBeeNetworkManager] - Rejecting incoming message for cluster 0x0402, but we have sent outgoing non-default-response to it
17:48:39.900 [INFO ] [rtsystems.zigbee.ZigBeeNetworkManager] - rejected command was ReportAttributesCommand [Temperature Measurement: null -> null, cluster=0402, TID=13, reports=[AttributeReport [attributeDataType=SIGNED_16_BIT_INTEGER, attributeIdentifier=0, attributeValue=2300]]]
17:48:39.907 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: DefaultResponse [Temperature Measurement: 0000/0 -> 0AC8/1, cluster=0402, TID=13, commandIdentifier=10, statusCode=FAILURE]
```

cdjackson, 2019-10-26:
Yes, this is not really a surprise. I'm not sure what application you are using, but clearly it will need to be adjusted following this change.

This change is required for certification so I'm not planning to revert it. You should change your application to add the supported local clusters.

sesuncedu, 2019-10-26:
ZSS Discovery is  broken. So is openhab zigbee binding. 

Is there really a PICS that allows you to send commands but requires you to reject the responses?

cdjackson, 2019-10-27:
As I've stated in the other issue, there is a requirement that a device does not respond to any endpoints not registered in the node descriptor, and it should not respond to any clusters not registered in the simple descriptor., So, yes, this is tested (at least in the SEP1.1b TH that I'm currently working through).

As stated - openHAB will not currently work with the latest version of the framework without changes to set this configuration.

cdjackson, 2020-01-01:
Closing this due to no response, and it appears to simply be a misunderstanding on how the system works.

closed 2020-01-01 by cdjackson