zsmartsystems/com.zsmartsystems.zigbee #872: Telegesis sets wrong source endpoint

triller-telekom, 2019-11-19:
I was debugging while I was not able to pair a device using a telegesis dongle and saw this:

```
2019-11-19T13:32:08.654+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - 00124B000EEA2880: Added transaction to queue, len=1,
transaction=ZigBeeTransaction [queueTime=0, state=WAITING, sendCnt=0, command=ReadAttributesCommand [Basic: 0000/0 -> 0B62/1, cluster=0000, TID=--, identifiers=[5, 4, 3, 2, 6, 0, 1]]]
2019-11-19T13:32:08.654+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - 0B62/1: Sending ZigBeeTransaction [queueTime=0, state=WAITING, sendCnt=0, command=ReadAttributesCommand [Basic: 0000/0 -> 0B62/1, cluster=0000, TID=11, identifiers=[5, 4, 3, 2, 6, 0, 1]]]
2019-11-19T13:32:08.694+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
2019-11-19T13:32:08.694+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Basic: 0000/0 -> 0B62/1, cluster=0000, TID=11, identifiers=[5, 4, 3, 2, 6, 0, 1]]
2019-11-19T13:32:08.694+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=17, commandId=0]
2019-11-19T13:32:08.724+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/1, destinationAddress=0B62/1, profile=0104, cluster=0000, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=55, payload=00 11 00 05 00 04 00 03 00 02 00 06 00 00 00 01 00]
2019-11-19T13:32:08.754+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Telegesis send: TelegesisSendUnicastCommand [address=2914, sourceEp=0, destEp=1, profileId=260, clusterId=0, messageData=00 11 00 05 00 04 00 03 00 02 00 06 00 00 00 01 00, messageId=null]
2019-11-19T13:32:08.754+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-11-19T13:32:08.774+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=2914, sourceEp=0, destEp=1, profileId=260, clusterId=0, messageData=00 11 00 05 00 04 00 03 00 02 00 06 00 00 00 01 00, messageId=null]
2019-11-19T13:32:08.784+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.database.ZigBeeNetworkDatabaseManager] - 00124B000EEA2880: Data store: Writing node.
2019-11-19T13:32:08.914+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+SENDUCASTB:11,0B62,00,01,0104,0000
          

2019-11-19T13:32:08.934+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SEQ:33
2019-11-19T13:32:08.934+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2019-11-19T13:32:08.984+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisSendUnicastCommand [address=2914, sourceEp=0, destEp=1, profileId=260, clusterId=0, messageData=00 11 00 05 00 04 00 03 00 02 00 06 00 00 00 01 00, messageId=51, status=SUCCESS]
2019-11-19T13:32:08.984+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=11 state=TX_ACK
2019-11-19T13:32:08.984+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=11, state=TX_ACK, outstanding=1
2019-11-19T13:32:08.984+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransaction] - Transaction state updated: TID 11 -> TX_ACK == TRANSMITTED
2019-11-19T13:32:09.034+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:ACK:33
2019-11-19T13:32:09.034+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisAckMessageEvent [messageId=51]
2019-11-19T13:32:09.034+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=11 state=RX_ACK
2019-11-19T13:32:09.034+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=11, state=RX_ACK, outstanding=1
2019-11-19T13:32:09.034+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransaction] - Transaction state updated: TID 11 -> RX_ACK == ACKED
2019-11-19T13:32:09.264+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:RX:0B62,0104,00,01,0000,41:  B	902010/25  BBitron Video    ?  B20180816       0,-40,FE
2019-11-19T13:32:09.304+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=2914, profileId=260, destinationEp=0, sourceEp=1, clusterId=0, messageData=18 11 01 05 00 00 42 09 39 30 32 30 31 30 2F 32 35 04 00 00 42 0C 42 69 74 72 6F 6E 20 56 69 64 65 6F 03 00 00 20 02 02 00 86 06 00 00 42 08 32 30 31 38 30 38 31 36 00 00 00 20 01 01 00 00 20 30, rssi=-64, lqi=254]
2019-11-19T13:32:09.344+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0B62/1, destinationAddress=0000/0, profile=0104, cluster=0000, addressMode=null, radius=0, apsSecurity=false, apsCounter=--, payload=18 11 01 05 00 00 42 09 39 30 32 30 31 30 2F 32 35 04 00 00 42 0C 42 69 74 72 6F 6E 20 56 69 64 65 6F 03 00 00 20 02 02 00 86 06 00 00 42 08 32 30 31 38 30 38 31 36 00 00 00 20 01 01 00 00 20 30]
2019-11-19T13:32:09.394+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Unknown local endpoint for APS frame ZigBeeApsFrame [sourceAddress=0B62/1, destinationAddress=0000/0, profile=0104, cluster=0000, addressMode=null, radius=0, apsSecurity=false, apsCounter=--, payload=18 11 01 05 00 00 42 09 39 30 32 30 31 30 2F 32 35 04 00 00 42 0C 42 69 74 72 6F 6E 20 56 69 64 65 6F 03 00 00 20 02 02 00 86 06 00 00 42 08 32 30 31 38 30 38 31 36 00 00 00 20 01 01 00 00 20 30]
2019-11-19T13:32:09.394+0100   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Incoming message from node 0B62 did not translate to command
```

Apparently there is a `ReadAttributesCommand` send out with a wrong source endpoint. The reply is send to this wrong endpoint as a destination and then discarded.

The culprit seems to be a hard-coded "0" here:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c994033c26cbc4cc4be67afa3bca1d7472f721d2/com.zsmartsystems.zigbee.dongle.telegesis/src/main/java/com/zsmartsystems/zigbee/dongle/telegesis/ZigBeeDongleTelegesis.java#L674

IMHO this should be

```
unicastCommand.setSourceEp(apsFrame.getSourceEndpoint());
```

After changing the code as suggested, responses are send to the correct endpoint and the pairing works again.

Shall I provide a PR to fix this?

cdjackson, 2019-11-19:
Hi Stefan,
Yes please - a PR would be great, thanks.  I’m a bit surprised that it’s taken so long to find this bug to be honest!

Thanks
Chris

triller-telekom, 2019-11-20:
Hi Chris,

yes I was also wondering why we didn't spot this issue earlier. Maybe there is another mechanism now in place which does a more restrictive checking on source and destination endpoints and thus it didn't match anymore. I don't know.

But it is good that we have spotted the problem and fixed it, see my linked PR.

Regards,

Stefan

cdjackson, 2019-11-20:
Yes, there is now more checking, but in this case I think that setting the source endpoint to 0 should have prevented the remote device from responding, and that's what we see here as there is no response.

An endpoint of 0 makes this a ZDP message rather than ZCL, so it's no surprise that there's no response, but I'm confused that it ever worked (but it clearly did).

Thanks for the PR and reporting anyway.

PR #873 (2019-11-20): Telegesis commands should use the source endpoint from APS frame
In the current implementation it uses a hard-coded value which is clearly
wrong.

Fixes #872
commit 8fc4d67f (2019-11-20): Telegesis commands should use the source endpoint from APS frame (#873)

In the current implementation it uses a hard-coded value which is clearly
wrong.

Fixes #872
closed 2019-11-20 by cdjackson