zsmartsystems/com.zsmartsystems.zigbee #874: [question] Should the fetched attribute values be serialized automatically when using the ZigBeeDataStore?

tejas-n, 2019-11-20:
I noticed that when you call `readValue(long)` on a `ZclAttribute`, it fetches the attribute from the end device and also updates the `lastValue` field but it doesn't automatically persist the value using the `ZigBeeNetworkDataStore` implementation. Is this the intended behavior?
Shouldn't it persist the value given that technically the node has been updated? If it's intentional, what's the recommended way of persisting it? Would calling `serializeNetworkDataStore(IeeeAddress)` after reading the attribute be the best way?

cdjackson, 2019-11-21:
Persisting the data after every single attribute update would be very time consuming. Given that it can take a significant amount of time to write the data to storage, and often gateways are running on low performing systems, this causes a lot of problems. If you had a high powered computer, then it would probably be fine, but in most instances, that is simply not the case.

There should however be no need to persist the data after every read. So long as you are doing a controlled shutdown of your system (using the ```ZigBeeNetworkManager.shutdown()``` method) then the data will be persisted when the system is closed.

If you are not doing a controlled shutdown, then if you are using the mesh monitor, the nodes are normally updated every minute or so anyway, so data will be periodically persisted.

tejas-n, 2019-11-21:
Thanks for the reply. That makes sense.
I am using ZigBeeDiscoveryExtension but doesn't look like it's persisting the previously fetched attribute value. Is it because the end devices(pendants) are sleeping when it does it periodic runs even though I had fetched the attribute values when they were awake?

cdjackson, 2019-11-22:
When the node is persisted, it will persist the attribute states as well as the rest of the node information.

Maybe the mesh update is not actually updating - it wont necessarily update mesh information on sleepy nodes.

You can also persist the data periodically manually if you want - there is the ```ZigBeeNetworkManager.serializeNetworkDataStore(address)``` method that does this - alternatively it will be serialised when the system closes, and then restored on startup, so no data should be lost.

tejas-n, 2019-11-23:
Thanks, Chris. I think that answers all of my questions.

cdjackson, 2019-11-23:
No problem - just reach out if there's anything further.

closed 2019-11-23 by cdjackson