zsmartsystems/com.zsmartsystems.zigbee #875: Queue in ZigbeeTransactionManager is blocked by old transaction

triller-telekom, 2019-11-22:
- Pair a device with for example openHAB.
- Remove the device via openHAB (we used a battery powered one and removed the battery before deleting it to make the leave request fail)
- Try to pair the device again.

The problem we found is in the `ZigbeeTransactionManager` whose `nodeUpdated()` get called so he can clean up all transactions corresponding to the node. We saw that the queue for the removed device contains a transaction with a destination address (network address) that is the OLD address of the device we are trying to pair again.

All new transactions, like discovery of descriptors or cluster bindings are queued AFTER this old transaction. The policy for sending data has a `maxOutstandingTranscations=1` and thus the queue is effectively blocked.

I made a proof of concept:

In the `ZigbeeTransactionManager#nodeUpdated()` I call a new method on the `ZigBeeTransactionQueue` and I named it 

```
void removeTransactionsNotBelongingToNode(ZigBeeNode node) {
        ZigBeeTransaction zt = null;
        boolean allCleared = false;
        while ((zt = queue.peek()) != null && !allCleared) {
            ZigBeeAddress destinationAddress = zt.getDestinationAddress();
            if (destinationAddress != null) {
                if (destinationAddress.getAddress() != node.getNetworkAddress()) {
                    // we found an old transaction which should not be send to this updated node anymore -> remove it
                    // from the queue
                    logger.debug("Stefan: Removing old transaction from queue: {}", zt);
                    queue.poll();
                } else {
                    // found the point in the queue were only new transactions are
                    allCleared = true;
                }
            } else {
                logger.debug("No destination address set in transaction={}", zt);
            }
        }
    }
```

This method removes all transactions from the queue which are not targeted towards the networkaddress of the current (updated) node.

After cleaning the queue, I am able to pair the device again.

Please see the log attached.

cdjackson, 2019-11-24:
Thanks Stefan,

I guess there are two possible solutions to this - one is as you've suggested, to clear the queue, but I'm not sure this is the right solution. The other solution is to change the destination network address to the updated value.

The reason I say I'm not sure it's best to remove the transactions with the old number is that your case where a device has left and rejoined is only one reason why the NWK address can change - it could also change under normal use. This is especially the case for battery devices (ZEDs) where they can rejoin via a new parent, and the new parent will provide a new NWK address. In this case, removing the transactions would not be right. If you example we had a mobile device (eg a portable remote) that was moving around the house, we may get into the situation where we can never send frames to it if every time it wakes up it has a new NWK address (even if this is probably an extreme example that's unlikely in reality).

Regarding your implementation - if we were to remove transactions from the queue, I think they also need to be completed/cancelled, or you don't complete the ```Future``` which leaves the caller thread hanging.

My preference is probably to update the queue to change the address, but I've not looked at such an implementation...

What do you think?

triller-telekom, 2019-11-29:
> My preference is probably to update the queue to change the address, but I've not looked at such an implementation...

> What do you think?

Considering the point you mentioned with the sleepy devices and we would potentially discard all their messages once they rejoin, I would also vote for rewriting those addresses.

A possible first sketch up for such an implementation is easy. Instead of the method above, use this one:

```
void rewriteDestinationAddresses(ZigBeeNode node) {
        LinkedList<ZigBeeTransaction> transactions = new LinkedList<>();

        ZigBeeTransaction zt = null;
        while ((zt = queue.poll()) != null) {
            ZigBeeAddress destinationAddress = zt.getDestinationAddress();
            if (destinationAddress != null) {
                if (destinationAddress.getAddress() != node.getNetworkAddress()) {
                    logger.debug("Stefan: Rewriting transaction destination address from {} to {} in transaction={}",
                            destinationAddress.getAddress(), node.getNetworkAddress(), zt);
                    destinationAddress.setAddress(node.getNetworkAddress());
                }
            } else {
                logger.debug("Stefan: No destination address set in transaction={}", zt);
            }
            transactions.add(zt);
        }

        for (ZigBeeTransaction trans : transactions) {
            logger.debug("Stefan: Enquing transaction={}", trans);
            queue.add(trans);
        }
    }
```

It would be called by the `ZigbeeTransactionManager#nodeUpdated` like this:

```
public void nodeUpdated(ZigBeeNode node) {
        // Make sure that the queue is set to the correct type for this node
        // This is handled here as this may change after the device is initially paired
        // since the {@link NodeDescriptor} isn't initially known
        ZigBeeTransactionQueue queue = nodeQueue.get(node.getIeeeAddress());
        if (queue == null) {
            logger.debug("Stefan: nodeUpdated(): no queue for nodeIEEE={}", node.getIeeeAddress());
            return;
        }

        //rewrite queue here with method sketched above
        queue.rewriteDestinationAddresses(node);

       if (setQueueType(node, queue)) {
            synchronized (outstandingTransactions) {
                int sleepyCnt = 0;
                // The queue type changed - resync the sleepyTransactions counter
                for (ZigBeeTransaction transaction : outstandingTransactions) {
                    ZigBeeTransactionQueue transactionQueue = getTransactionQueue(transaction);
                    if (transactionQueue != null && transactionQueue.isSleepy()) {
                        sleepyCnt++;
                        continue;
                    }
                }
                logger.debug("Sleepy transaction count resynchronised: was {}, now {}", sleepyCnt, sleepyTransactions);
                sleepyTransactions = sleepyCnt;
            }
        }
}
```

However, I looked into this a bit closer and saw that in some cases we might also have transactions with a wrong destination address already within `outstandingTransactions`. So I guess we would also need to somehow clean those up as well.

The problem with doing this is, that the transaction inside `outstandingTransactions` might belong to communications with **different** devices (i.e. different IEEE address), because all device-specific queues are polled for new ready2send transactions and then shuffled into `outstandingTransactions`.

What we have to do is rewriting those transactions which belong to the device whose node has changed its network address. The only matching I would see is the IEEE address, because it always stays the same.

Unfortunately the transactions (and their commands within) only carry zigbee network addresses and the information about the IEEE address is lost once they are moved from their device-specific queues into the `outstandingTransactions` set.

So how can we preserve some more information so we can do the filtering? Or am I completely mistaken and there is some easier solution to this?

triller-telekom, 2019-12-02:
@cdjackson Did you have time to have a look at this again?

My problem would be that we have to clean up the `outstandingTransactions` and do not have any connection from the commands within to the physical IEEE address of the device so we could track them across network address changes.

cdjackson, 2019-12-02:
No - not yet, sorry.

cdjackson, 2019-12-02:
My suggestion is to add the ```IeeeAddress``` to the ```ZigBeeTransaction```. Currently there is the destination ```ZigBeeAddress```, and if we add the ```IeeeAddress``` we could add a ```getIeeeAddress``` and ```setDestinationAddress(ZigBeeAddress)``` which would allow updating the address.

Another alternative is to have a ```Map<IeeeAddress, NetworkAddress>``` in the transaction manager and use this to detect the network address change in ```updateNode```, and subsequently update the ```outstandingTransactions```. I'm not sure I'm such a fan of caching this information though.

WDYT?

triller-telekom, 2019-12-04:
I also thought about adding the `IeeeAddress` to the `ZigBeeTransaction` object so we always have the possibility to determine to which device it belongs to.

And you are right, caching with such a map will only be error-prone and thus I would not do that.

How shall we proceed? Do you want to create a PR to integrate the `IeeeAddress` into the `ZigBeeTransaction` and offer the necessary getters and setter? And then afterwards I could create a PR to clean up the transactions queues like suggested before?

If you like you can also grab the code from above and integrate/improve it yourself. Just let me know which option your prefer.

triller-telekom, 2019-12-09:
@cdjackson I have just opened a PR for the IEEEaddress, can you have a look please? https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/897

In a follow-up PR I can provide the code for cleaning up the queues.

PR #898 (2019-12-17): Rewrite transaction destinations if node rejoins with new network address
- to be merged AFTER #897 (compilation depends on it)

Fixes #875
commit 488d24da (2019-12-17): Rewrite transaction destinations if node rejoins with new network address (#898)

Fixes #875
closed 2019-12-17 by cdjackson