zsmartsystems/com.zsmartsystems.zigbee #876: ZigBeeNode: status is not correctly set in all cases

triller-telekom, 2019-11-22:
- pair a device with for example openHAB.
- remove the device from openHAB (we used a battery powered one and removed the battery before deleting it to make the leave request fail)

During that process, the `ZigbeeNetworkManager#nodeStatusUpdate()` sets the status of the `ZigbeeNode` WITHIN `ZigbeeNetworkManager.networkNodes` to OFFLINE. I highlighted "within", because it fetched the node via `getNode()` from its `networkNodes` collection and then changes the status. So the status of the existing `ZigbeeNode` is updated and only after that `updateNode()` is called.

The result is an OFFLINE node within the ZSS network manager.

Now we try to pair the device again and ZSS will pick up the node from `ZigbeeNetworkManager.networkNodes` and will update its values, like for example with a new network address.

However, the problem is that `ZigbeeNetworkManager#updateNode()`, which calls `ZigbeeNode#updateNode()` does NOT copy any status value of a `ZigbeeNode` towards itself (neither UNKNOWN, OFFLINE, nor ONLINE). This results in a `ZigbeeNode` that has all properties from the current device state, but which is still `OFFLINE` and will also stay `OFFLINE` throughout any future node updates.

cdjackson, 2019-11-24:
Let me see if I understand - you mean the following case doesn't result in an update -:

```
node = networkManager.getNode();
node.updateSomething()
networkManager.updateNode(node);
```

If that's what you're reporting, then yes, this is correct and we should change the code to always present a new object to ```updateNode()``` for change detection to work. I've recently been through and changed some instances of this in the Smart Energy client, but I'm sure there are others.

TomTravaglino, 2019-11-25:
Allow me to join the conversation since I investigated this issue together with Stefan.

I'm not sure if you entirely got Stefans point (but maybe I'm wrong :wink:). He describes what happens when we want to pair a device after it previously left the network. 
When the node leaves its status is set to OFFLINE and the node remains in the network manager. When the device joins again a new instance of the node is created but it doesn't replace the old one. Instead, just its values are copied to the old instance (in `ZigBeeNode.updateNode(...)`). The issue here is that the status of the old instance is not updated and so the node stays OFFLINE forever.

I hope I do not cause any confusion now.

cdjackson, 2019-11-25:
Hi Tomaso,
Yes, I understood. When the device leaves, the node is kept in the manager. I was assuming that the reason it is not updated if it rejoins is because (maybe) in the discovery system, the code directly updates the existing node rather than updating a new object and passing that to ```updateNode```. This would mean there is no notification.

>when the device joins again a new instance of the node is created but it doesn't replace the old one.

This is correct - the node should not be replaced - it should be updated.

> The issue here is that the status of the old instance is not updated and so the node stays OFFLINE forever.

As said, I assume that this is because of the updating of the existing node rather than the updating of the node via a new object.

ie, the code should not do what I said earlier -:

```
node = networkManager.getNode();
node.updateSomething()
networkManager.updateNode(node);
```

Instead, it should do something like -:

```
node = new ZigBeeNode(address);
node.updateSomething()
networkManager.updateNode(node);
```

Does that make sense?

triller-telekom, 2019-11-29:
Hi Chris,

I think we have a misunderstanding here. the original question from me isn't about using a new node vs. updating the old one.

My issue is that in ´ZigbeeNode#updateNode(ZigBeeNode node)` all field from `node` are copied over to the current instance. However the private field `nodeState` is NOT touched!

so what I am saying is, no matter if you create a new node or update the old one, as long as you are using the current implementations for updating a node, you will NEVER update the state of the node.

In my issue, this results in a node that is always OFFLINE, because even when the device appears again we do not change its `nodeState` field.

Is this explanation a bit clearer?

triller-telekom, 2019-11-29:
In code the part that is missing would look like this inside `ZigbeeNode#updateNode(ZigBeeNode node)`

```
if(nodeState != node.getNodeState()) {
            nodeState = node.getNodeState();
            updated = true;
        }
```

cdjackson, 2019-11-29:
Thanks Stefan - my misunderstanding and I will update this.

Thanks.

PR #880 (2019-11-30): Update node status in ZigBeeNode.updateNode()
Closes #876
closed 2019-11-30 by cdjackson