zsmartsystems/com.zsmartsystems.zigbee #899: ZigBeeDongleEzsp does not detect stick unplugged state in certain cases

dtodor, 2019-12-13:
If the ZigBee USB dongle gets unplugged, it might happen that the change won't be registered within the ```ZigBeeDongleEzsp#handleLinkStateChange``` method. 

This occurs in the case, where the ```linkState``` is reported ```true``` before the initialization (i.e. ```initialised``` variable) has finished. Thus ```networkStateUp``` won't be set to ```true``` and once we unplug the stick, the change back to ```false``` won't be registered.

The behavior can be observed in the following logs:

```
2019-12-13T11:08:40.480+0100   224  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Ember: Link State change to true, initialised=false, networkStateUp=false
2019-12-13T11:08:40.490+0100   224  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Ember: Link State change to true ignored.
...
2019-12-13T11:08:40.750+0100   224  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Startup complete. NWK Address = 0000, State = EMBER_JOINED_NETWORK
```

triller-telekom, 2019-12-19:
I think this is a serious problem, because for applications like for example openHAB this results in the behaviour that the zigbee bridge does not go offline if one unplugs the stick.

Also I am not sure about the side effects inside ZSS if the setting of the `linkState` is not set to `true` at all.

@cdjackson Since you know the insides of ZSS better, did you have a chance to look at this issue, yet?

cdjackson, 2019-12-19:
I will not have time to look at this for a few weeks - I'm not completely sure I understand it at the moment, but again, I've not looked at it at all due to other commitments.

While it might be a "serious problem" - I think if people are unplugging the dongle, then this is something they are consciously doing.  Also, if I understand this, it is only happening in certain cases which is prior to the network coming online so the opportunity is limited? (not to mean that it shouldn't be looked at - just that it's not high priority as I have ZigBee certification things to do right now).

triller-telekom, 2019-12-19:
Thank you for your feedback.

Actually the problem occurs after the system has been fully initialized. I.e. you can use the system, pair and control devices. Afterwards you unplug the zigbee stick and then it will not go OFFLINE.

IMHO the reason is that the change to OFFLINE is skipped, because the `linkState` has never been set to ONLINE correctly. So it is a general problem, not one that only occurs in "special scenarios".

cdjackson, 2019-12-19:
I will try look at this in future, but as mentioned this is not currently a high priority.

PR #914 (2019-12-21): Set Ember online start at end of startup()
Setting the ember network state is disabled during initialisation to prevent the situation where the network goes online before being reset. However, this holdoff means that we must set the state at the end of ```startup()``` when we finally know the startup state.
Closes #899
closed 2019-12-21 by cdjackson