zsmartsystems/com.zsmartsystems.zigbee #907: Fail to initialize Telegesis dongle

mikomarrache, 2019-12-17:
Hi,

This is not bug but I didn't find another place to ask for help, so I apologize.

I tried to start the console application provided with the library and I get the following logs:

```
Initialising ZigBee console...
09:50:34.191  DEBUG  Default: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=2, interTransactionDelay=50, maxRetries=2]
09:50:34.197  DEBUG  Broadcast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
09:50:34.200  DEBUG  Multicast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
09:51:07.945  ERROR  Error creating network database folder database\TELEGESIS
09:51:23.674  DEBUG  ZigBeeNetworkManager initialize: networkState=UNINITIALISED
09:51:23.705  DEBUG  Network state is updated to INITIALISING

ZigBee network state updated to INITIALISING
09:51:23.722  DEBUG  Telegesis dongle initialize.
09:51:23.726  DEBUG  Opening port COM6 at 115200 baud with FLOWCONTROL_OUT_NONE.
09:51:24.066  DEBUG  TelegesisFrameHandler thread started
09:51:24.197  DEBUG  TX Telegesis queue: 1
09:51:24.197  DEBUG  TX Telegesis: TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
09:51:24.201  DEBUG  TX Telegesis Data:ATS124=1

09:51:27.205  DEBUG  TELEGESIS Timer: Timeout 0
09:51:32.197  DEBUG  Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
09:51:32.199  DEBUG  TX Telegesis queue: 1
09:51:32.199  DEBUG  TX Telegesis: TelegesisSetRegisterBitCommand [register=14, bit=1, state=false, password=null]
09:51:32.202  DEBUG  TX Telegesis Data:ATS0E1=0

09:51:35.203  DEBUG  TELEGESIS Timer: Timeout 1
09:51:40.200  DEBUG  Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=14, bit=1, state=false, password=null]
09:51:40.201  DEBUG  TX Telegesis queue: 1
09:51:40.201  DEBUG  TX Telegesis: TelegesisSetRegisterBitCommand [register=14, bit=0, state=false, password=null]
09:51:40.206  DEBUG  TX Telegesis Data:ATS0E0=0

09:51:43.208  DEBUG  TELEGESIS Timer: Timeout 2
09:51:43.208  DEBUG  Error: number of retries exceeded [3].
09:51:48.203  DEBUG  Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=14, bit=0, state=false, password=null]
09:51:48.211  DEBUG  TX Telegesis queue: 1
09:51:56.213  DEBUG  Telegesis interrupted in sendRequest TelegesisSoftwareResetCommand []
09:51:56.214  DEBUG  TX Telegesis queue: 2
09:52:04.216  DEBUG  Telegesis interrupted in sendRequest TelegesisSoftwareResetCommand []
09:52:04.217  DEBUG  TX Telegesis queue: 3
09:52:12.216  DEBUG  Telegesis interrupted in sendRequest TelegesisSoftwareResetCommand []
09:52:12.218  DEBUG  TX Telegesis queue: 4
09:52:20.219  DEBUG  Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
09:52:20.219  DEBUG  TX Telegesis queue: 5
09:52:28.221  DEBUG  Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=14, bit=1, state=false, password=null]
09:52:28.223  DEBUG  TX Telegesis queue: 6
09:52:36.224  DEBUG  Telegesis interrupted in sendRequest TelegesisSetRegisterBitCommand [register=14, bit=0, state=false, password=null]
09:52:36.238  DEBUG  TX Telegesis queue: 7
09:52:44.239  DEBUG  Telegesis interrupted in sendRequest TelegesisSetPromptEnable1Command [configuration=8804]
09:52:44.239  DEBUG  Error setting Telegesis Prompt 1 register
09:52:44.240  ERROR  Unable to configure Telegesis dongle
09:52:44.244  DEBUG  Network state is updated to OFFLINE

ZigBee network state updated to OFFLINE
```

Here are the arguments I use:

`-dongle TELEGESIS -port COM6 -baud 115200 -reset`

I'm using the dongle ETRX3USB.

I also installed the last driver version (6.6.1).

Windows 10 64 bits.

Thanks

cdjackson, 2019-12-17:
Normally the telegesis works at 19k2 by default - unless you have manually changed it?

mikomarrache, 2019-12-17:
You're right - it's working now.

The console application expects the baud rate to be set so I could not simply let the Telegesis module set it for me.

closed 2019-12-17 by mikomarrache

cdjackson, 2019-12-17:
The console application must be told the baud rate - it's not possible to open a serial port without knowing the baud rate and other such parameters.