zsmartsystems/com.zsmartsystems.zigbee #912: Unable to pair any device

triller-telekom, 2019-12-19:
Using commit 9dc77ee9d7876fc4be03b5c1ed729fca0eff90cf (Add option to set profile id as console command line argument (#910)) I cannot pair any device.

Last working commit that I used was 4cea309d175fd7c8d72616873f75235ae8e44103 (Document EMBER_MAC_COMMAND_TRANSMIT_FAILURE in active scan (#886)).

I have tested it with a telegesis dongle and an ember stick. According to the zigbee sniffer, both die right after the `DeviceAnnounce`.

This is what I see in the logs while trying to pair a Centralite 3200 smart plug:
```
2019-12-19T15:42:08.136+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - Permit join to FFFC/0 for 60 seconds.
2019-12-19T15:42:08.137+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - Broadcast: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=null queueTime=1, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=--, permitDuration=60, tcSignificance=true]]
2019-12-19T15:42:08.138+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - FFFC/0: Sending ZigBeeTransaction [ieeeAddress=null queueTime=1, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=04, permitDuration=60, tcSignificance=true]]
2019-12-19T15:42:08.138+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
2019-12-19T15:42:08.139+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=04, permitDuration=60, tcSignificance=true]
2019-12-19T15:42:08.140+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=FFFC/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=04, payload=04 3C 01]
2019-12-19T15:42:08.141+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - 000D6F000F24146C: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=000D6F000F24146C queueTime=0, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=--, permitDuration=60, tcSignificance=true]]
2019-12-19T15:42:08.142+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - 0000/0: Sending ZigBeeTransaction [ieeeAddress=000D6F000F24146C queueTime=2, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=05, permitDuration=60, tcSignificance=true]]
2019-12-19T15:42:08.142+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - transactionListenerAdded: 2 outstanding
2019-12-19T15:42:08.143+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=05, permitDuration=60, tcSignificance=true]
2019-12-19T15:42:08.144+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=0000/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=05, payload=05 3C 01]
2019-12-19T15:42:08.149+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP queue: 1
2019-12-19T15:42:08.151+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP: EzspSendBroadcastRequest [destination=FFFC, apsFrame=EmberApsFrame [profileId=0000, clusterId=0036, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=04], radius=31, messageTag=04, messageContents=04 3C 01]
2019-12-19T15:42:08.152+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - --> TX SPI frame: FE 18 04 00 FF 00 36 FC FF 00 00 36 00 00 00 40 11 00 00 04 1F 04 03 04 3C 01 
2019-12-19T15:42:08.159+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame: FE 07 04 80 FF 00 36 00 36 
2019-12-19T15:42:08.160+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - RX EZSP: EzspSendBroadcastResponse [status=EMBER_SUCCESS, sequence=36]
2019-12-19T15:42:08.162+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP queue: 1
2019-12-19T15:42:08.162+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendBroadcastResponse [status=EMBER_SUCCESS, sequence=36]
2019-12-19T15:42:08.190+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame: FE 1B 04 88 FF 00 45 05 00 00 36 00 00 00 00 01 00 00 36 FF 00 00 00 FF FF 03 04 3C 01 
2019-12-19T15:42:08.192+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_BROADCAST_LOOPBACK, apsFrame=EmberApsFrame [profileId=0000, clusterId=0036, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=36], lastHopLqi=255, lastHopRssi=0, sender=0000, bindingIndex=255, addressIndex=255, messageContents=04 3C 01]
2019-12-19T15:42:08.193+0100   445  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_BROADCAST_LOOPBACK, apsFrame=EmberApsFrame [profileId=0000, clusterId=0036, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=36], lastHopLqi=255, lastHopRssi=0, sender=0000, bindingIndex=255, addressIndex=255, messageContents=04 3C 01]
2019-12-19T15:42:08.193+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=0000/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=0, apsSecurity=false, apsCounter=36, payload=04 3C 01]
2019-12-19T15:42:08.194+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=04, permitDuration=60, tcSignificance=true]
2019-12-19T15:42:08.195+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=04, permitDuration=60, tcSignificance=true] 
2019-12-19T15:42:08.195+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=04, permitDuration=60, tcSignificance=true] ZigBeeTransaction [ieeeAddress=null  [... cut ...]
2019-12-19T17:32:47.758+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: MatchDescriptorRequest [2214/0 -> 0000/0, cluster=0006, TID=09, nwkAddrOfInterest=FFFD, profileId=0104, inClusterList=[25], outClusterList=[]]
2019-12-19T17:32:47.759+0100   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: MatchDescriptorRequest [2214/0 -> 0000/0, cluster=0006, TID=09, nwkAddrOfInterest=FFFD, profileId=0104, inClusterList=[25], outClusterList=[]]
```

cdjackson, 2019-12-19:
Just so I'm clear what you are doing -:

What software are you using the libraries with? Is this still openHAB? What version of OH? Have you refactored for the latest changes?

I probably won't have time to look at this until late next week, but please answer the above...

triller-telekom, 2019-12-20:
I am using the openHAB binding, but it is adjusted to work with the latest ZSS master. However I have documented the solution of my problem here: https://github.com/openhab/org.openhab.binding.zigbee/issues/531

If everything would only be that easy :)

@cdjackson: So fortunately no real big issue with the latest master. Have a nice christmas and a good start into 2020!

closed 2019-12-20 by triller-telekom