zsmartsystems/com.zsmartsystems.zigbee #958: Provide API to allow user to perform a scan

cdjackson, 2020-01-12:
It would be useful to allow users to perform a scan when creating a network to allow a clear channel to be selected prior to forming the network.

tejas-n, 2020-08-25:
Hi, is anyone already working on this? I asked a related question about setting a channel mask here https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1051#issuecomment-679424019 and heard back from @triller-telekom that it's not been implemented.

I can pick this up if no one's working on it. Are you thinking of exposing methods for doing energy scan and finding active networks in `ZigBeeNetworkManager` which under the hood calls the transport layer?

cdjackson, 2020-08-26:
Hi @tejas-n . I don't think anyone is working on this, and for sure, it would be great to do so.

> Are you thinking of exposing methods for doing energy scan in ZigBeeNetworkManager which under the hood calls the transport layer?

Yes, basically the same as my thinking - we probably want the following API in both the `ZigBeeNetworkManager` and  `ZigBeeTransportTransmit` . 

* `setChannelMask()`: to set the channel mask used within the system. This is used for the scan, but may also be used for any subsequent channel changes etc.
* `performScan(options)`, or `performEnergyScan()` and `performActiveScan` or something - to actually kick off the scan.

Exposing it through the network manager rather than just through the dongle provides the possibility later to do something smarter with changing channels automatically.
 
I don't think this is too hard to implement the scanning - for the Ember at least that you're using, all the code is there in the dongle, it's just a case of hooking it up :) 

Thanks @tejas-n

tejas-n, 2020-08-27:
Thanks, @cdjackson. I'll have a crack at it on the weekend :)