zsmartsystems/com.zsmartsystems.zigbee #967: Exceptions in attribute listeners are not caught by the framework

hsudbrock, 2020-01-15:
The method `ZclCluster.notifyAttributeListener` notifies all registered `ZclAttributeListener`s when an attribute is updated.

If one of the listener throws an exception, the exception is not handled in `ZclCluster.notifyAttributeListener`, with two effects:
1. The attribute listeners following the listener that threw the exception will not be notified about the attribute update.
2. The exception is not logged with the used logging framework, but is logged to the console instead.

My suggestion is to 
* (a) handle exceptions in `ZclCluster.notifyAttributeListener` by simply logging them using the logger of the `ZclCluster` class, and to 
* (b) add an `UncaughtExceptionHandler` to the threads that are created by the executors provided by the `ZigBeeExecutors` class, which also simply logs the exception.

I will provide a PR shortly, as a basis for discussion.

cdjackson, 2020-01-15:
> The attribute listeners following the listener that threw the exception will not be notified about the attribute update.

Why is that? This seems a bit strange to me as the exception is not even in the same thread. It can't stop another thread - is it completely killing the executor (again, I don't think it should do that either).

Note that I'm not disagreeing with the logging - just trying to understand why this is happening as I wouldn't expect threads to interact (unless the executor is somehow dying?).

hsudbrock, 2020-01-15:
You are right and I was wrong. I overlooked that it is not one thread looping over all listeners, but a loop that executes a runnable for each listener. Sorry for this.

cdjackson, 2020-01-15:
Ok, no problem Henning - I was just wondering if you'd found something that I'd not considered :)  Otherwise the PR looks fine - thanks.

PR #968 (2020-01-16): Catch uncaught exceptions in attribute listeners and log them
Also adds an UncaughtExceptionHandler that logs exceptions to the executors
provided by the framework.

Resolves #967
closed 2020-01-16 by cdjackson